nordicsemi/Android-DFU-Library #338: DFU fails, because scanner cannot find the device (MAC address)

Oleksandr1713, 2022-06-13:
**DFU Bootloader version:**
 - SDK version: [SDK 15.3]
 - Bonding used: [no]
 - Library version: [2.0.1, 2.0.3]

**Device information:**
 - Device: [Lenovo TB-X505L]
 - OS: [Android 10]

**Describe the bug**
After enabling DFU mode, the DFU process cannot be completed due to the following reason:

`2022-06-13 11:00:51.500 27088-27515/com.myapp.app.dev I/DfuImpl: Scanning for new address finished with: null`

This issue only occurs with the `Lenovo TB-X505L` device.

**Logs**
```
2022-06-13 11:00:38.622 27088-27482/com.myapp.app.dev V/FA: onActivityCreated
2022-06-13 11:00:38.664 27088-27483/com.myapp.app.dev V/FA: Activity paused, time: 4397817
2022-06-13 11:00:38.675 27088-27483/com.myapp.app.dev V/FA: Activity resumed, time: 4397910
2022-06-13 11:00:38.764 27088-27088/com.myapp.app.dev I/DfuBaseService: DFU service created. Version: debug
2022-06-13 11:00:38.773 27088-27515/com.myapp.app.dev I/DfuBaseService: Starting DFU service in foreground
2022-06-13 11:00:38.799 27088-27515/com.myapp.app.dev I/DfuBaseService: Connecting to the device...
2022-06-13 11:00:38.800 27088-27515/com.myapp.app.dev D/BluetoothGatt: connect() - device: FF:03:A6:82:AB:EC, auto: false
2022-06-13 11:00:38.800 27088-27515/com.myapp.app.dev D/BluetoothGatt: registerApp()
2022-06-13 11:00:38.800 27088-27515/com.myapp.app.dev D/BluetoothGatt: registerApp() - UUID=717aff8c-6523-4a08-999b-935811fe301a
2022-06-13 11:00:38.805 27088-27161/com.myapp.app.dev D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2022-06-13 11:00:38.810 27088-27161/com.myapp.app.dev D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:03:A6:82:AB:EC
2022-06-13 11:00:38.810 27088-27161/com.myapp.app.dev I/DfuBaseService: Connected to GATT server
2022-06-13 11:00:38.813 27088-27161/com.myapp.app.dev D/BluetoothGatt: discoverServices() - device: FF:03:A6:82:AB:EC
2022-06-13 11:00:38.814 27088-27161/com.myapp.app.dev I/DfuBaseService: Attempting to start service discovery... succeed
2022-06-13 11:00:38.815 27088-27214/com.myapp.app.dev D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=FF:03:A6:82:AB:EC
2022-06-13 11:00:38.815 27088-27214/com.myapp.app.dev I/DfuBaseService: Connected to GATT server
2022-06-13 11:00:38.816 27088-27214/com.myapp.app.dev D/BluetoothGatt: discoverServices() - device: FF:03:A6:82:AB:EC
2022-06-13 11:00:38.818 27088-27214/com.myapp.app.dev I/DfuBaseService: Attempting to start service discovery... succeed
2022-06-13 11:00:39.580 27088-27214/com.myapp.app.dev D/BluetoothGatt: onConnectionUpdated() - Device=FF:03:A6:82:AB:EC interval=6 latency=0 timeout=500 status=0
2022-06-13 11:00:40.097 27088-27214/com.myapp.app.dev D/BluetoothGatt: onSearchComplete() = Device=FF:03:A6:82:AB:EC Status=0
2022-06-13 11:00:40.097 27088-27214/com.myapp.app.dev I/DfuBaseService: Services discovered
2022-06-13 11:00:40.099 27088-27515/com.myapp.app.dev I/DfuImpl: Buttonless service without bond sharing found -> SDK 13 or newer
2022-06-13 11:00:40.109 27088-27088/com.myapp.app.dev D/MY_DfuRunningService: Connected
2022-06-13 11:00:40.119 27088-27088/com.myapp.app.dev D/MY_DfuRunningService: ProcessStarting
2022-06-13 11:00:40.164 27088-27214/com.myapp.app.dev D/BluetoothGatt: onConnectionUpdated() - Device=FF:03:A6:82:AB:EC interval=96 latency=0 timeout=400 status=0
2022-06-13 11:00:41.102 27088-27515/com.myapp.app.dev I/DfuImpl: Enabling indications...
2022-06-13 11:00:41.102 27088-27515/com.myapp.app.dev D/BluetoothGatt: setCharacteristicNotification() - uuid: 0b177506-24f2-4343-b580-123c836b01e5 enable: true
2022-06-13 11:00:42.262 27088-27515/com.myapp.app.dev I/DfuImpl: Sending Enter Bootloader (Op Code = 1)
2022-06-13 11:00:42.264 27088-27088/com.myapp.app.dev D/MY_DfuRunningService: EnablingDfuMode
2022-06-13 11:00:43.793 27088-27483/com.myapp.app.dev V/FA: Inactivity, disconnecting from the service
2022-06-13 11:00:46.475 27088-27214/com.myapp.app.dev D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=6 device=FF:03:A6:82:AB:EC
2022-06-13 11:00:46.475 27088-27214/com.myapp.app.dev W/DfuBaseService: Target device disconnected with status: 8
2022-06-13 11:00:46.476 27088-27515/com.myapp.app.dev I/DfuImpl: Device disconnected before receiving notification
2022-06-13 11:00:46.476 27088-27515/com.myapp.app.dev D/BluetoothGatt: refresh() - device: FF:03:A6:82:AB:EC
2022-06-13 11:00:46.478 27088-27515/com.myapp.app.dev I/DfuBaseService: Refreshing result: true
2022-06-13 11:00:46.478 27088-27515/com.myapp.app.dev I/DfuBaseService: Cleaning up...
2022-06-13 11:00:46.478 27088-27515/com.myapp.app.dev D/BluetoothGatt: cancelOpen() - device: FF:03:A6:82:AB:EC
2022-06-13 11:00:46.480 27088-27515/com.myapp.app.dev D/BluetoothGatt: close()
2022-06-13 11:00:46.480 27088-27515/com.myapp.app.dev D/BluetoothGatt: unregisterApp() - mClientIf=6
2022-06-13 11:00:46.482 27088-27515/com.myapp.app.dev I/DfuImpl: Restarting to bootloader mode
2022-06-13 11:00:46.496 27088-27515/com.myapp.app.dev D/BluetoothAdapter: isLeEnabled(): ON
2022-06-13 11:00:46.498 27088-27088/com.myapp.app.dev I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2022-06-13 11:00:46.502 27088-27214/com.myapp.app.dev D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2022-06-13 11:00:51.494 27088-27515/com.myapp.app.dev D/BluetoothAdapter: isLeEnabled(): ON
2022-06-13 11:00:51.500 27088-27515/com.myapp.app.dev I/DfuImpl: Scanning for new address finished with: null
2022-06-13 11:00:51.523 27088-27515/com.myapp.app.dev I/DfuBaseService: Starting DFU service in foreground
2022-06-13 11:00:53.535 27088-27515/com.myapp.app.dev I/DfuBaseService: Connecting to the device...
2022-06-13 11:00:53.537 27088-27515/com.myapp.app.dev D/BluetoothGatt: connect() - device: FF:03:A6:82:AB:EC, auto: false
2022-06-13 11:00:53.537 27088-27515/com.myapp.app.dev D/BluetoothGatt: registerApp()
2022-06-13 11:00:53.537 27088-27515/com.myapp.app.dev D/BluetoothGatt: registerApp() - UUID=72e2ef1f-e6b8-47b1-b81c-9547720c7b67
2022-06-13 11:00:53.541 27088-27214/com.myapp.app.dev D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2022-06-13 11:00:53.542 27088-27088/com.myapp.app.dev D/MY_DfuRunningService: Connecting
[... 34 lines cut ...]
2022-06-13 11:01:57.646 27088-27515/com.myapp.app.dev D/BluetoothGatt: registerApp() - UUID=6afb1bb3-f8e8-432c-ab2b-59602a5b0ef6
2022-06-13 11:01:57.652 27088-27214/com.myapp.app.dev D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2022-06-13 11:02:27.661 27088-27214/com.myapp.app.dev D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=6 device=FF:03:A6:82:AB:EC
2022-06-13 11:02:27.662 27088-27214/com.myapp.app.dev E/DfuBaseService: Connection state change error: 133 newState: 0
2022-06-13 11:02:27.662 27088-27515/com.myapp.app.dev I/DfuBaseService: Connection error after: 30020 ms
2022-06-13 11:02:27.662 27088-27515/com.myapp.app.dev E/DfuBaseService: Device not reachable. Check if the device with address FF:03:A6:82:AB:EC is in range, is advertising and is connectable
2022-06-13 11:02:27.662 27088-27515/com.myapp.app.dev I/DfuBaseService: Attempt: 3
2022-06-13 11:02:27.664 27088-27515/com.myapp.app.dev D/BluetoothGatt: refresh() - device: FF:03:A6:82:AB:EC
2022-06-13 11:02:27.666 27088-27515/com.myapp.app.dev I/DfuBaseService: Refreshing result: true
2022-06-13 11:02:27.666 27088-27515/com.myapp.app.dev I/DfuBaseService: Cleaning up...
2022-06-13 11:02:27.666 27088-27515/com.myapp.app.dev D/BluetoothGatt: cancelOpen() - device: FF:03:A6:82:AB:EC
2022-06-13 11:02:27.668 27088-27515/com.myapp.app.dev D/BluetoothGatt: close()
2022-06-13 11:02:27.668 27088-27515/com.myapp.app.dev D/BluetoothGatt: unregisterApp() - mClientIf=6
2022-06-13 11:02:28.280 27088-27088/com.myapp.app.dev D/MY_DfuRunningService: Error: GATT ERROR
2022-06-13 11:02:28.323 27088-27088/com.myapp.app.dev I/DfuBaseService: DFU service destroyed
2022-06-13 11:02:30.333 27088-27548/com.myapp.app.dev V/FA: Recording user engagement, ms: 111647
2022-06-13 11:02:30.442 27088-27548/com.myapp.app.dev V/FA: Connecting to remote service
2022-06-13 11:02:30.460 27088-27548/com.myapp.app.dev V/FA: Connection attempt already in progress
2022-06-13 11:02:30.460 27088-27548/com.myapp.app.dev V/FA: Activity paused, time: 4509564
2022-06-13 11:02:30.461 27088-27548/com.myapp.app.dev V/FA: Activity resumed, time: 4509599
2022-06-13 11:02:30.578 27088-27548/com.myapp.app.dev V/FA: Connection attempt already in progress
2022-06-13 11:02:30.580 27088-27548/com.myapp.app.dev V/FA: Connection attempt already in progress
2022-06-13 11:02:30.580 27088-27548/com.myapp.app.dev D/FA: Connected to remote service
2022-06-13 11:02:30.582 27088-27548/com.myapp.app.dev V/FA: Processing queued up service tasks: 4
```

connyhald, 2022-07-07:
We are using the `Lenovo TB-X505L` tablet and we could narrow down the issue to the usage of offloaded filtering introduced in this commit: https://github.com/NordicSemiconductor/Android-DFU-Library/pull/275/commits/fa664cb2e75107c9227ab6d28182aa91e62cac9d

We understand what this change does and that it solves a real issue (turning screen off during DFU). On the other hand there seems to be some devices that just won't be able to scan if offloaded filtering is enabled.

Could this be made into a setting?

@josipKra To verify you're having the same issue: Could you please try with version `1.11.0` which does not yet contain this change?

josipKra, 2022-07-08:
@connyhald yes I can confirm that version 1.11.0 works ok on my device

blackcodefr, 2022-07-08:
We have the same problem, our devices can take a few seconds to reboot properly (up to 20 secs or more).

This is what introduced the problem : https://github.com/NordicSemiconductor/Android-DFU-Library/pull/319

This could be ok if it was an option or at least the scan period following the disconnect should be configurable and by default be longer (30 seconds).

@philips77 you should look into this, it is a big regression for industrial devices with reboot constraints like ours. If it stays like this, we would not be able to go higher than version 1.12 or have to fork the library altogether to only change a constant or revert the change.

philips77, 2022-08-01:
> few seconds to reboot properly

Is it intended behavior? Does it mean, that after switching to bootloader mode, the device needs several seconds to reboot in bootloader mode, by design?

In that case, the scan period could be made a setting.

philips77, 2022-08-01:
> we could narrow down the issue to the usage of offloaded filtering introduced in this commit: https://github.com/NordicSemiconductor/Android-DFU-Library/commit/fa664cb2e75107c9227ab6d28182aa91e62cac9d

Yes... But I think I can have a solution for that. Could you try adding an empty filter to `ScanFilter` array, and do the filtering in the app, as it was don't before? That way, as far as I know, it will still work in the background, but we're not limiting DFU to devices with working offloaded filtering.

blackcodefr, 2022-08-01:
> @blackcodefr
> 
> > few seconds to reboot properly
> 
> Is it intended behavior? Does it mean, that after switching to bootloader mode, the device needs several seconds to reboot in bootloader mode, by design?
> 
> In that case, the scan period could be made a setting.

Yes, it is absolutely by design, a shut down sequence must be done before reboot that takes several seconds. The scan period should definitely be a setting. Also, manual scan selection like on iOS would be best on top of that, this way manufacturerData can be used to identify the device.

philips77, 2022-08-02:
OK, will do it. Expect a release in coming days and a PR earlier.

Oleksandr1713, 2022-08-02:
> Yes... But I think I can have a solution for that. Could you try adding an empty filter to ScanFilter array, and do the filtering in the app, as it was don't before? That way, as far as I know, it [...]

@philips77 , unfortunately, it does not resolve the issue with the `Lenovo TB-X505L` tablet. Our app already applies filtering based on manufacture data. 

It is worth mentioning, we have already tried to compile Nordic DFU library ourselves and increase the timeout of bootloader scanner from `5000ms` to `30000ms`.

https://github.com/NordicSemiconductor/Android-DFU-Library/blob/f50ac71026986236f1278225be44ff53a33fba47/lib_dfu/src/main/java/no/nordicsemi/android/dfu/internal/scanner/BootloaderScanner.java#L39-L44

Unfortunately, it still does not resolve the problem and we still can observe the following:

>  I/DfuImpl: Scanning for new address finished with: null

blackcodefr, 2022-08-02:
> OK, will do it. Expect a release in coming days and a PR earlier.

Doesn't change our ask but to be more precise, it is after receiving the switch to Bootloader command, before reboot in Bootloader mode, that the shutdown sequence is done.

philips77, 2022-08-05:
Hi,
Could you please check the PR and verify it works for you? Perhaps it will also work on Leonvo, @connyhald.

Oleksandr1713, 2022-08-05:
Tested on `Lenovo TB-X505L` and now it works. Thank you @philips77!

closed 2022-08-08 by philips77

philips77, 2022-08-08:
Btw, I'll revert https://github.com/NordicSemiconductor/Android-DFU-Library/pull/319 for bonded devices in the next version.

ABouretz, 2022-08-08:
Hi @philips77, I'm working with @blackcodefr.
I can confirm the new scan delay/timeout setup works as expected.
Tested on `Xiaomi Mi 9T Pro`.
Thank you !

philips77, 2022-08-08:
OK, I'm releasing 2.1 and 2.2-beta01 with #350 merged.

PR #345 (2022-08-08): Option to set scan delay, timeout and device selector when looking for Bootloader
This PR is intended to fix #338.

New features:
* [x] Option to set initial delay before scanning for DFU bootloader (some devices may need longer time to reboot) using https://github.com/NordicSemiconductor/Android-DFU-Library/blob/55651d4fbcfaf8df380d2cc02f1f4b0aa749d8b6/lib_dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L300
* [x] Option to set scan timeout using https://github.com/NordicSemiconductor/Android-DFU-Library/blob/55651d4fbcfaf8df380d2cc02f1f4b0aa749d8b6/lib_dfu/src/main/java/ [... cut]