nordicsemi/Android-DFU-Library #339: DFU fails with "Characteristic write error: 133" on Galaxy Tab A8

FluffyBunniesTasteTheBest, 2022-06-23:
**DFU Bootloader version (please complete the following information):**
 - SDK version: nRF5 SDK 17.0.2 with nrf5 SDK for Mesh 5.0.0
 - Bonding used: no
 - Library version: 2.0.3 (through noridc_dfu plugin for Flutter: https://pub.dev/packages/nordic_dfu)

**Device information (please complete the following information):**
 - Device: Samsung Galaxy Tab A8, SM-X200
 - OS: Android 11 (SDK 30), Patch 2022-02-01

**Describe the bug**
DFU reliably fails on Samsung Galaxy Tab A8 with __Characteristic write error: 133__. It (DFU) is, however, reliably working on dozens of other devices from Samsung, Lenovo, Google, Huawai, Vollaphone...

**Logs**
Interesting part starts quite far down after: **Sending Calculate Checksum command (Op Code = 3)**:
```
2022-06-22 22:05:31.599 I/flutter: 2022-06-22 22:05:31.589814: INFO: awesomecompany: Starting awesomecompany 2.0.0 b76
2022-06-22 22:05:31.603 I/ViewRootImpl@f48e10e[MainActivity]: [DP] cancelDraw  isViewVisible: true
2022-06-22 22:05:32.102 E/libc: Access denied finding property "persist.vendor.sf.fbc"
2022-06-22 22:05:32.102 E/libc: Access denied finding property "persist.vendor.gpu.fbc"
2022-06-22 22:05:32.102 E/libc: Access denied finding property "persist.vendor.vsp.fbc"
2022-06-22 22:05:32.120 I/ViewRootImpl@f48e10e[MainActivity]: [DP] cancelDraw  isViewVisible: true
2022-06-22 22:05:32.137 I/ViewRootImpl@f48e10e[MainActivity]: [DP] pd() Asnyc report
2022-06-22 22:05:32.138 I/ViewRootImpl@f48e10e[MainActivity]: mAttachInfo.mThreadedRenderer.draw, mView = DecorView@ab0f1a4[MainActivity] w = 1200 h = 1920
2022-06-22 22:05:32.142 I/SurfaceView: applySurfaceTransforms: t = android.view.SurfaceControl$Transaction@dd25d surfaceControl = Surface(name=SurfaceView - com.awesomecompany.awesome_mesh_apph/com.awesomecompany.awesomecompany_mesh.MainActivity@1fc4a70@0)/@0xce917d2 frame = 1
2022-06-22 22:05:32.166 I/ViewRootImpl@f48e10e[MainActivity]: [DP] pdf(0) 1 android.view.ViewRootImpl.lambda$performDraw$1$ViewRootImpl:4745 android.view.-$$Lambda$ViewRootImpl$DJd0VUYJgsebcnSohO6h8zc_ONI.run:6 android.os.Handler.handleCallback:938 
2022-06-22 22:05:32.166 I/ViewRootImpl@f48e10e[MainActivity]: [DP] rdf()
2022-06-22 22:05:32.188 I/ViewRootImpl@f48e10e[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2022-06-22 22:05:32.189 D/InputMethodManager: prepareNavigationBarInfo() DecorView@ab0f1a4[MainActivity]
2022-06-22 22:05:32.189 D/InputMethodManager: getNavigationBarColor() -855310
2022-06-22 22:05:32.194 D/InputMethodManager: prepareNavigationBarInfo() DecorView@ab0f1a4[MainActivity]
2022-06-22 22:05:32.194 D/InputMethodManager: getNavigationBarColor() -855310
2022-06-22 22:05:32.194 V/InputMethodManager: Starting input: tba=com.awesomecompany.awesome_mesh_apph ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2022-06-22 22:05:32.195 D/InputMethodManager: startInputInner - Id : 0
2022-06-22 22:05:32.195 I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2022-06-22 22:05:32.202 D/InputTransport: Input channel constructed: 'ClientS', fd=121
2022-06-22 22:05:32.203 D/InputMethodManager: prepareNavigationBarInfo() DecorView@ab0f1a4[MainActivity]
2022-06-22 22:05:32.204 D/InputMethodManager: getNavigationBarColor() -855310
2022-06-22 22:05:32.204 V/InputMethodManager: Starting input: tba=com.awesomecompany.awesome_mesh_apph ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2022-06-22 22:05:32.204 D/InputMethodManager: startInputInner - Id : 0
2022-06-22 22:05:32.214 I/SurfaceControl: nativeRelease nativeObject e[-5476376627874255824]
2022-06-22 22:05:32.215 I/SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0x9ba01e9 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1836 android.view.ViewRootImpl.relayoutWindow:9132 android.view.ViewRootImpl.performTraversals:3422 android.view.ViewRootImpl.doTraversal:2671 android.view.ViewRootImpl$TraversalRunnable.run:10094 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
2022-06-22 22:05:32.217 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874256048]
2022-06-22 22:05:32.217 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874255376]
2022-06-22 22:05:32.219 I/ViewRootImpl@f48e10e[MainActivity]: Relayout returned: old=(0,0,1200,1920) new=(0,0,1200,1920) req=(1200,1920)0 dur=9 res=0x1 s={true -5476376623042145776} ch=false fn=2
2022-06-22 22:05:32.221 I/ViewRootImpl@f48e10e[MainActivity]: updateBoundsLayer: shouldReparent = false t = android.view.SurfaceControl$Transaction@2ed50a3 sc = Surface(name=Bounds for - com.awesomecompany.awesome_mesh_apph/com.awesomecompany.awesomecompany_mesh.MainActivity@0)/@0xeaf2fa0 frame = 2
2022-06-22 22:05:32.224 I/ViewRootImpl@f48e10e[MainActivity]: mAttachInfo.mThreadedRenderer.draw, mView = DecorView@ab0f1a4[MainActivity] w = 1200 h = 1920
2022-06-22 22:05:32.227 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874288640]
2022-06-22 22:05:32.227 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874285952]
2022-06-22 22:05:32.227 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874255824]
2022-06-22 22:05:32.533 I/flutter: 2022-06-22 22:05:32.533149: INFO: flutter_i18n: New locale: en
2022-06-22 22:05:32.536 I/flutter: 2022-06-22 22:05:32.536094: INFO: flutter_i18n: The current locale is en
2022-06-22 22:05:32.887 I/flutter: 2022-06-22 22:05:32.887388: INFO: flutter_i18n: YAML file loaded for en/shared
2022-06-22 22:05:32.889 D/PhoneWindow: forceLight changed to false [com.awesomecompany.awesome_mesh_apph/com.awesomecompany.awesomecompany_mesh.MainActivity] from com.android.internal.policy.PhoneWindow.updateForceLightNavigationBar:4239 com.android.internal.policy.PhoneWindow.setNavigationBarColor:4034 io.flutter.plugin.platform.PlatformPlugin.setSystemChromeSystemUIOverlayStyle:431 io.flutter.plugin.platform.PlatformPlugin.access$700:30 io.flutter.plugin.platform.PlatformPlugin$1.setSystemUiOverlayStyle:108 
2022-06-22 22:05:32.912 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874288752]
2022-06-22 22:05:32.912 I/SurfaceControl: nativeRelease nativeObject e[-5476376627874288752]
2022-06-22 22:05:32.913 I/SurfaceControl: assignNativeObject: nativeObject = 0 Surface(name=null)/@0x9ba01e9 / android.view.SurfaceControl.readFromParcel:1117 android.view.IWindowSession$Stub$Proxy.relayout:1836 android.view.ViewRootImpl.relayoutWindow:9132 android.view.ViewRootImpl.performTraversals:3422 android.view.ViewRootImpl.doTraversal:2671 android.view.ViewRootImpl$TraversalRunnable.run:10094 android.view.Choreographer$CallbackRecord.run:1010 android.view.Choreographer.doCallbacks:809 android.view.Choreographer.doFrame:744 android.view.Choreographer$FrameDisplayEventReceiver.run:995 
2022-06-22 22:05:32.914 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874289088]
2022-06-22 22:05:32.914 I/SurfaceControl: nativeRelease nativeObject e[-5476376627874289088]
2022-06-22 22:05:32.914 I/SurfaceControl: nativeRelease nativeObject e[-5476376627874285280]
2022-06-22 22:05:32.914 I/SurfaceControl: nativeRelease nativeObject s[-5476376627874286064]
2022-06-22 22:05:32.914 I/SurfaceControl: nativeRelease nativeObject e[-5476376627874286064]
2022-06-22 22:05:32.914 I/ViewRootImpl@f48e10e[MainActivity]: Relayout returned: old=(0,0,1200,1920) new=(0,0,1200,1920) req=(1200,1920)0 dur=10 res=0x1 s={true -5476376623042145776} ch=false fn=3
2022-06-22 22:05:32.915 I/ViewRootImpl@f48e10e[MainActivity]: updateBoundsLayer: shouldReparent = false t = android.view.SurfaceControl$Transaction@2ed50a3 sc = Surface(name=Bounds for - com.awesomecompany.awesome_mesh_apph/com.awesomecompany.awesomecompany_mesh.MainActivity@0)/@0xeaf2fa0 frame = 3
2022-06-22 22:05:32.916 I/ViewRootImpl@f48e10e[MainActivity]: mAttachInfo.mThreadedRenderer.draw, mView = DecorView@ab0f1a4[MainActivity] w = 1200 h = 1920
2022-06-22 22:05:32.996 I/flutter: 2022-06-22 22:05:32.995846: INFO: flutter_i18n: YAML file loaded for en/screen_appsettings
2022-06-22 22:05:33.006 I/flutter: 2022-06-22 22:05:33.006753: INFO: flutter_i18n: YAML file loaded for en/screen_connecting
2022-06-22 22:05:33.011 I/flutter: 2022-06-22 22:05:33.011041: INFO: flutter_i18n: YAML file loaded for en/screen_nodes
[... cut ...]
2022-06-22 22:20:28.242 I/SurfaceView: updateSurface: releaseSurfaces() #1
2022-06-22 22:20:28.243 I/SurfaceView: updateSurface: releaseSurfaces() #1
2022-06-22 22:20:28.245 I/SurfaceView: updateSurface: releaseSurfaces() #1
2022-06-22 22:20:28.255 I/SurfaceView: updateSurface: releaseSurfaces() #1
2022-06-22 22:20:29.968 I/ViewRootImpl@f48e10e[MainActivity]: stopped(true) old=false
2022-06-22 22:20:29.987 I/SurfaceView: updateSurface: releaseSurfaces() #1
```

Have you ever seen something similar? 

Any ideas how to workaround or fix this issue are welcome.

wodyy666, 2022-06-28:
I also noticed some weird behaviour regarding bluetooth and connectivity in general on our Tab A8.

This issue might be related to this: https://devzone.nordicsemi.com/f/nordic-q-a/88557/samsung-galaxy-tab-a8-does-not-work-with-nrf-mesh-android-app/371422
Here is the issue that is mentioned in the post: https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library/issues/520

FluffyBunniesTasteTheBest, 2022-06-28:
@wodyy666 Thanks for sharing these links! 

There are all kind of issues on the Galaxy Tab A8, and the way it looks they won't be fixed for Android 11, however, the upgrade to Android 12 is scheduled for August, so hopefully the problems disappear with it...

philips77, 2022-08-08:
Hi, we've ordered Samsung TAB A8 and will test it in-house.

philips77, 2022-08-29:
Ok, we have the Tab A8. Let me do some tests.

wodyy666, 2022-09-21:
How are you doing with your tests? Did you manage to reproduce this issue?

philips77, 2022-09-21:
I'm in a process of fixing issues we found in version 2.2.0. I'll run some tests on Tab A8 this week.

RaviKoradiya, 2022-10-07:
@philips77 
I am facing the same issue with the Motorola moto e20.
I have tried all fixes from various GitHub issues and StackOverflow, but no luck ☹️
Even it fails on Nordic's DFU app (downloaded from the play store).

FluffyBunniesTasteTheBest, 2022-10-07:
@RaviKoradiya For now, we've decided to just label the Galaxy Tab A8 as _unsupported_ - well, and in that case the Moto e20 as well. Thank you for joining in and sharing the device details!

siralexeyduke, 2022-12-21:
@philips77 
Hi! We are facing also the same issue with Galaxy Tab A8.
We can reproduce it with following setup:

- Android-DFU-Library, Git commit: 39cd896g (latest)
- Firmware: examples/dfu/secure_bootloader/pca10040_s132_ble_debug + examples/dfu/secure_dfu_test_images/hrs_application_s132.hex
- Hardware: Samsung S7+ Tablet, Samsung A8 Tablet, pca10040 dev board

Steps to reproduce:
1. Flash pca10040 with secure DFU bootloader and softdevice
2. Start Android-DFU-Library App 
3. Select hrs_application_s132.hex DFU image
4. Select DfuTarg device
5. Start firmware update

Expected behaviour:
- Init package after OP_CODE 0x01 is written:
    - Bootloader logs "nrf_dfu_ble: Buffer 0x20008298 acquired, len 144 (244)"
    - Tablet logs "I/DfuImpl: Checksum received (Offset = 144, CRC = 592A3B8C)"

Actual behaviour
- Init package is not written:
    - Bootloader logs nothing after OP_CODE 0x01 confirmation
    - Tablet logs
        - "I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)"
        - "E/DfuImpl: Characteristic write error: 133"

We have also attached the collected logs from tablets and from bootloader for these scenarios. [DFU Samsung A8 bug.zip](https://github.com/NordicSemiconductor/Android-DFU-Library/files/10276502/DFU.Samsung.A8.bug.zip)

Some observations:

Comparing logs from bootloader connected to A8 and to S7+ we have noticed, that after MTU exchange there is slightly different behaviour. In S7+ case bootloader receives BLE_GAP_EVT_DATA_LENGTH_UPDATE (251, max_rx_time 2120) and in A8 case it receives BLE_GAP_EVT_DATA_LENGTH_UPDATE (27, max_rx_time 328).

philips77, 2022-12-22:
Thank you for the analysis and steps to reproduce! This indeed seems to be an issue with that tablet. I'll try to find a workaround, if possible.

I didn't find time to test it on A8 this year, sorry for that. It will have to wait until 2023 😞

FluffyBunniesTasteTheBest, 2022-12-22:
We had a similar report on a Galaxy S8 which refused to communicate with the mesh after it was upgraded to Android 13. Our customer fixed the issue by downgrading back to Android 12.1. Unfortunately the device logs went lost during the downgrade, leaving me clueless about what happened... Sorry for the vague report, but it seems like Samsung devices are about to make 2023 an _interesting_ year for us. 

Btw., thanks for providing the libraries! I wish you all the best for 2023.

siralexeyduke, 2023-02-17:
Hi @philips77 !
Are there eventually any updates on this issue?

abhriyaroy, 2023-06-19:
any updates on this?

philips77, 2023-06-21:
Hi, sorry, I'm still busy with other tasks :(

philips77, 2023-10-03:
OK, I finally resumed working on DFU. I'm testing on Samsung Tab A8.

philips77, 2023-10-03:
> Actual behaviour
> 
> Init package is not written:
   Bootloader logs nothing after OP_CODE 0x01 confirmation
   Tablet logs
>  ```
>   I/DfuImpl: Sending Calculate Checksum command (Op Code = 3)
>   E/DfuImpl: Characteristic write error: 133
>   ```

I am getting this result. Which is great, I managed to replicate the issue.

philips77, 2023-10-03:
I am able to replicate this issue by writing data manually using nRF Connect. Something is very bad with this Samsung Tab A8.

Tomorrow I'll check with a sniffer. Perhaps it has issues with MTU > 23.

philips77, 2023-10-04:
I'm attaching the sniffer logs. You need Ellisys Bluetooth Analyzer to open (Windows only).
[Sniffer logs.zip](https://github.com/NordicSemiconductor/Android-DFU-Library/files/12802518/Sniffer.logs.zip)

There are 2 logs, one from a PIxel 5 with Android 14, which works, and the other from Samsung Tab A8, which fails.

On Samsung everything works until it tries to send long packet (251 bytes using Data Length Extension). The same packet goes through from Pixel without problems, but when it's sent from the tablet the DK does not reply. I suspect a timing issue. Both devices use slightly different Connection Parameters. Perhaps Samsung does not send packets according to its own parameters and the DK ignores what is received. The tablet tries to send the 1st packet 397 times before disconnecting due to Supervision Timeout (no response from Slave, which is 6 seconds).

It's very hard to say what the timing issue is, but some replies from the DK are overlapping the transmitted packets by few hundreds us.

## Solution

I tried disabling MTU request when doing DFU from the Samsung Tab A8 and it seems to be working. So the issue is only when the tablet is trying to use Data Length Extension, short packets work fine.

You may try using this method to set the MTU to 23:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/53ae6f8ed7f0f14f66aea1f9922fbe186ce16f33/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L399-L402

philips77, 2023-10-04:
A helpful link to a related DevZone case: https://devzone.nordicsemi.com/f/nordic-q-a/88557/samsung-galaxy-tab-a8-does-not-work-with-nrf-mesh-android-app/387530

And a quotation:
> We found an issue in the Samsung Galaxy A8 Bluetooth capabilities this summer, where it seems to send packets with a longer payload than what has been negotiated, which caused these issues with our devices, as they will terminate the connection if this happens. Our conclusion was that there wasn't anything to be done on the nRF side of the application to remedy this.

From [this](https://devzone.nordicsemi.com/f/nordic-q-a/88557/samsung-galaxy-tab-a8-does-not-work-with-nrf-mesh-android-app/386873) comment.

philips77, 2023-10-04:
Here's the screenshot from the sniffer log:

[image]

As you can see, the Master claims maxRx = 27 bytes, but later sends 251.

philips77, 2023-10-04:
I added an option in nRF DFU app to disable MTU request.
If you're using the library, use `disableMtuRequest()` in `DfuServiceInitiator`. This should do the trick.

closed 2023-10-04 by philips77

QuentinFarizon, 2023-10-04:
@philips77 Shouldn't the library itself disable MTU request for devices with this issue ?
As library users, should we disable mtu for every phones ? Or testing if we are on a Galaxy Tab
 ? 🤔🙁

QuentinFarizon, 2023-10-05:
@philips77 I think it's better integrated in the lib than as an option, yes. Thanks !