nordicsemi/Android-DFU-Library #353: setRebootTime

darren1713, 2022-09-08:
### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: SDK 15
 - Are you using bonding: no
 - DFU Library version: 2.1.0

### Device information (please complete the following information):
 - Device: Pixel 4 and Pixel 5
 - OS: Android 13

### Your question
Default value for setRebootTime is zero, causing the main nRF DFU app to fail, as well as the library implementation.

Some not so detailed background...this worked at some point and then we updated libraries, Android version (API levels), new Android phones, and not sure where this stopped working, but that's ok, it's fixed!

The specific class that is used is SecureDfuImpl

We now HAVE to setRebootTime to give the library/phone/device some time to swap to DFU mode and start advertising.

For the nRF DFU app to work, we added "setRebootTime(5000)" here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/profile_dfu/src/main/java/no/nordicsemi/android/dfu/profile/main/data/DFUManager.kt#L67

Which sets some delay before rescanning here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/lib_dfu/src/main/java/no/nordicsemi/android/dfu/BaseDfuImpl.java#L759

5000 (5 seconds) is probably too much, but haven't tested a range to see exactly how much works every time. At minimum, this could be a configuration option in the nRF DFU app, but it might be a new issue with higher API versions, Android 13, etc, where some delay is necessary.

philips77, 2022-09-08:
Thank you for the issue. This also may be related to how the Android detects disconnected device (after switching from bootloader to app mode). I'm working on a PR that improves that, so perhaps this 2 sec delay won't be needed after all.
I'm working on an update of the DFU library and the app, and will add the settings to customize timings in Settings. I hope to release it tomorrow. please test. Version will be 2.2.

philips77, 2022-09-09:
Fixed in 2.2. It is now possible to set the timings in the app's setting, but should also work with default values.

PR #356 (2022-09-09): Improvement: disconnecting device when switching to dfu bootloader
This PR improves behavior with switching to DFU bootloader. Before, the library was either waiting until the connection is terminated or just starting scanning for the bootloader, hoping that the connection will terminate at some point. This, however, seemed to have issues on some devices, which were unable to connect to the device in bootloader mode for a long time.
Now, when the new address is expected, the device disconnects the connection to app mode before starting scanning for the bootloader.

Fixes #353 and #328.