nordicsemi/Android-DFU-Library #358: Unable to write Op Code 3 (error 133), and dfu library is stuck in the disconnecting status

wizardmly, 2022-09-13:
### Information
This GitHub Issue page is for reporting issues or asking questions regarding the Android DFU library. **For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).**

Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)

Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: [Unknown]
 - Are you using bonding: [no]
 - DFU Library version: [2.1.0]

### Device information (please complete the following information):
 - Device: [Samsung A136B]
 - OS: [Android 12]

### Your question
Step2 by Steps:
1. Use the dfu library perform dfu
2. Before the dfu progress finish, move the sensor device away to let the gatt connection disconnect from the phone.
3. Then, check the phone dfu notification, I found that, the dfu notification is stuck, and show the "Disconnecting...".
4. Check the logcat, the log "Disconnected from GATT server" was triggered before "Disconnecting from the device..."

And after dived into the dfu library source code, I finnally found the root cause is that:  the function com.vivalnk.no.nordicsemi.android.dfu.DfuProgressInfo.ProgressListener#updateProgressNotification is taken too much times(about 30~40ms), so before the library finish setProgress:
6. 
[image]

the library had trigger the connection state change callback function:
[image]

### Logs

2022-09-10 00:27:26.439 14101-24017/?  E/DfuBaseService: Unable to write Op Code 3 (error 133)
2022-09-10 00:27:26.464 14101-14101/?  I/DfuBaseService: Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2022-09-10 00:27:26.479 14101-16657/?  I/DfuBaseService: Disconnected from GATT server
2022-09-10 00:27:26.481 14101-24017/?  I/DfuBaseService: Disconnecting from the device...

### Current My Workaround
move the function uupdateProgressNotification to main thread:
[image]

I konw it's a tempoary cheap way to avoid this issue, may be you guys have another better way.

zippert, 2022-10-14:
We're experiencing the same issue. Without actually removing the device from the phone, the connection fails, and the notification is stuck at "Disconnecting...".

The call to "waitUntilDisconnected" then waits indefinitely for a lock:

[image]

Presumably this is some kind of runtime issue that causes our app to hang indefinitely during update (when we manage to reproduce the error, we don't even seem to be able to start the update).

brianherold, 2023-01-27:
We also have the same issue.

**DFU Bootloader version (please complete the following information):**
nRF5 SDK version: [15]
Are you using bonding: [no]
DFU Library version: [2.3.0]

**Device information (please complete the following information):**
Device: [Samsung S22]
OS: [Android 12]

If the device disconnects during an update, most of the time we will get "Disconnected" and then "Disconnecting". When that happens, we are unable to restart the update. Presumably that is because the DFU service is stuck in an endless loop.

Our listener does see a call to `onDeviceDisconnecting` when that happens. My current intention is to try to create a workaround in our app based on that call. If we see `onDeviceDisconnecting` and the update hangs without us seeing an error or completion event, we kill the DFU service and reinitialize it from scratch.

If I wanted to mess with the Android-DFU-Library to fix the issue, I would probably make a `waitUntilDisconnecting` method and call that before `gatt.disconnect()` to ensure we don't hit this race condition.

philips77, 2023-01-31:
Hi all,
Thank you for reporting the issue. Please, use @brianherold's solution until I release a new version. I'm busy now with other tasks so it may be few weeks. Sorry for the delay.

Rem193, 2023-02-08:
Hi,
@brianherold's solution is working on my project. 👍

hchoang, 2023-03-31:
@Rem193  Hi Remy, how do you apply the fix to your project?

PavlosTze, 2023-06-28:
@philips77 Any news on that?

philips77, 2023-11-20:
Closed with #374.

closed 2023-11-20 by philips77