nordicsemi/Android-DFU-Library #359: Using Legacy files .hex and .bin to update firmware

rashmimatur, 2022-09-15:
### Information
This GitHub Issue page is for reporting issues or asking questions regarding the Android DFU library. **For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).**

Make sure you are using the latest version of the library: [ [image] ](https://bintray.com/nordic/android/no.nordicsemi.android%3Adfu/_latestVersion)

Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/Android-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/Android-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: [e.g. SDK 16]
 - Are you using bonding: [yes/no]
 - DFU Library version: [e.g. 2.1]

### Device information (please complete the following information):
 - Device: [e.g. Samsung A20]
 - OS: [e.g. Android 6]

### Your question
I am trying to update the firmware using the legacy .Hex and .Bin files, Where as the source code supports firmware update using a .zip file. Please guide me on the changes that are required to get the application working for .BIN/.Hex file

### Logs
Add any other context or logs describing the problem here.

philips77, 2022-09-15:
If you want to update your device, you can use nRF Connect for Mobile for Android. That app still supports separate hex and bin files. However, by using [nrfutil](https://github.com/NordicSemiconductor/pc-nrfutil) you may create a zip file which will contain the package. A single zip file is easier to distribute and allows additional features, like updating both soft device and the app in a single file.

If you want to add this feature to your app, there's a `setHexAndBin` method in `DfuServiceInitiator` which should do the trick.

philips77, 2022-09-15:
For legacy dfu use branch 0.5.x of nrfutil.

rashmimatur, 2022-09-15:
Thank you for Quick response. The info provided by you really helps. If I may ask what goes inside the Zip file? what files are required by the nrfutil to generate the .zip file? I have only the .bin and .hex file with me

philips77, 2022-09-19:
The nRF Util generates the ZIP file from the HEX file only. It creates the BIN file automatically, depending on the parameters you give to the script.
If you have already a valid BIN file, you may create a dummy ZIP file with nRF Util giving incorrect params and then replace the BIN file inZIP with your copy.

The content of the ZIP file contains a *manifest.json* file, which must have the name exactly *manifest.json*. It then describes the content and names of other files in the ZIP. A ZIP file may contain both app firmware and soft device + bootloader firmware in a single file.

closed 2022-09-19 by philips77