nordicsemi/Android-DFU-Library #366: Notification Abort button

eyrdor, 2022-09-23:
### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: N/A
 - Are you using bonding: no
 - DFU Library version: 1.11.1

### Device information (please complete the following information):
 - Device: Samsung SM-T307U
 - OS: Android 10

### Your question
Can the "Abort" button in the update notification be disabled or removed and if it can, how?

### Logs
N/A

philips77, 2022-09-24:
Let me check and come back to you after the weekend.

philips77, 2022-10-07:
In your `DfuService` class, which you use to start the DFU process there are 3 methods that you may override:
* By overriding this method, and not calling `super`, you will remove the **Abort** button.
   https://github.com/NordicSemiconductor/Android-DFU-Library/blob/38dad6229fa949dca090cfc8a65aa26f0e4ea7f9/lib_dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1789

* By overriding this method, you may customize the error notification.
   https://github.com/NordicSemiconductor/Android-DFU-Library/blob/38dad6229fa949dca090cfc8a65aa26f0e4ea7f9/lib_dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1860

* And by overriding this method you can customize the notification that comes when the DFU starts, when the service is started as *foreground service*.
   https://github.com/NordicSemiconductor/Android-DFU-Library/blob/38dad6229fa949dca090cfc8a65aa26f0e4ea7f9/lib_dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1903

For example, this will make the **Abort** button gone:
```kotlin
override fun updateProgressNotification(builder: NotificationCompat.Builder, progress: Int) {
   // don't do a thing.
}
```

philips77, 2022-10-07:
For reference, this is the `DfuService` class in the nRF DFU app:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/38dad6229fa949dca090cfc8a65aa26f0e4ea7f9/profile_dfu/src/main/java/no/nordicsemi/android/dfu/profile/main/repository/DFUService.kt#L45

and here's where it is used to start DFU:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/38dad6229fa949dca090cfc8a65aa26f0e4ea7f9/profile_dfu/src/main/java/no/nordicsemi/android/dfu/profile/main/data/DFUManager.kt#L65-L87

closed 2022-10-07 by philips77

eyrdor, 2022-10-08:
Thanks for the assistance @philips77 !