nordicsemi/Android-DFU-Library #369: Update firmware process never completed

dan-developer, 2022-11-22:
### DFU Bootloader version
 - Are you using bonding: no
 - DFU Library version: 2.0.1

### Device information:
 - Device: Samsung A13
 - OS: Android 12

### My question
When I start a firmware upgrade, it always reaches 100%, but it never completes, it runs forever and the device is inaccessible.

#### Code

```java
        mPromise = promise;
        final DfuServiceInitiator starter = new DfuServiceInitiator(address)
                .setKeepBond(false);
        if (name != null) {
            starter.setDeviceName(name);
        }
        starter.setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true);
        starter.setZip(filePath);
        final DfuServiceController controller = starter.start(this.reactContext, DfuService.class);
```
### Logs
```
11-22 01:43:08.589   583   583 I SehLight: lcd : 128+
11-22 01:43:08.589   568   599 D BatteryDump: wait_for_battery_event : occurred : change@/devices/platform/panel_drv@0/backlight/panel 
11-22 01:43:08.590   583   583 I SehLight: lcd : 128-
11-22 01:43:08.605   912   965 D DisplayPowerController[0]: animateScreenStateChange: target=2, mLogicalDisplay.isEnabled()=true
11-22 01:43:08.606   912   965 D DisplayPowerController[0]: getFinalBrightness : Summary is 128 (0.50) -> 128 (0.50)
11-22 01:43:08.704 27146 27156 I .gms.persisten: Background concurrent copying GC freed 183043(12MB) AllocSpace objects, 0(0B) LOS objects, 38% free, 18MB/30MB, paused 154us,94us total 160.495ms
11-22 01:43:08.720 12748 12780 D BtGatt.GattService: onWriteCharacteristic() - address=FEEBA9_8, status=0
11-22 01:43:08.721 12748 12780 D BtGatt.GattService: onWriteCharacteristic : isCongested == false
11-22 01:43:08.726 12748 12780 D BtGatt.GattService: onWriteCharacteristic() - address=FEEBA9_8, status=0
11-22 01:43:08.726 12748 12780 D BtGatt.GattService: onWriteCharacteristic : isCongested == false
11-22 01:43:08.730 12748 12780 D BtGatt.GattService: onWriteCharacteristic() - address=FEEBA9_8, status=0
11-22 01:43:08.730 12748 12780 D BtGatt.GattService: onWriteCharacteristic : isCongested == false
11-22 01:43:08.738 28030 28181 I ReactNativeJS: DFU progress: 100%
11-22 01:43:08.743   912   929 D ApplicationPolicy: isStatusBarNotificationAllowedAsUser: packageName = com.cleanproject,userId = 0
11-22 01:43:08.744   912   929 D ApplicationPolicy: isStatusBarNotificationAllowedAsUser: packageName = com.cleanproject,userId = 0
11-22 01:43:08.951  1724  1724 I Launcher.NotificationListener: onNotificationPosted : com.cleanproject number : 0
11-22 01:43:08.953  1240  1240 D InterruptionStateProvider: No bubble up: not allowed to bubble: 0|com.cleanproject|283|null|10282
11-22 01:43:08.954  1240  1240 D InterruptionStateProvider:  no Heads up : edgelighting enabled app. 0|com.cleanproject|283|null|10282
11-22 01:43:08.960  1240  1456 D PeopleSpaceWidgetMgr: Sbn doesn't contain valid PeopleTileKey: null/0/com.cleanproject
11-22 01:43:08.974  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:08.974  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:08.993  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:08.993  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:09.010  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:09.010  1240  1240 D QS      : setQSExpansion 0.0 -118.125
11-22 01:43:09.105   912   924 I system_server: Background concurrent copying GC freed 605213(12MB) AllocSpace objects, 11(284KB) LOS objects, 17% free, 56MB/68MB, paused 592us,530us total 414.360ms
11-22 01:43:09.467 27146 27226 W SQLiteConnectionPool: The connection pool for database '+data+user+0+com_google_android_gms+databases+phenotype_db' has been unable to grant a connection to thread 61 (highpool[3]) with flags 0x2 for 4.0 seconds.
11-22 01:43:09.467 27146 27226 W SQLiteConnectionPool: Connections: 0 active, 1 idle, 1 available.
11-22 01:43:10.102   912  1172 D NetdEventListenerService: DNS Requested by : 102, 10282, 0
11-22 01:43:10.730 27146 27156 I .gms.persisten: Background concurrent copying GC freed 183158(12MB) AllocSpace objects, 0(0B) LOS objects, 38% free, 18MB/30MB, paused 204us,82us total 163.437ms
11-22 01:43:10.915   912   965 D PowerManagerService: UserActivityStateListenerState: 0
11-22 01:43:11.331   364   364 D io_stats: !@ 179,0 r 641083 23665591 w 519924 17877876 d 120413 15320788 f 0 0 iot 479940 0 th 0 0 0 pt 0 inp 0 0 34239.786
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: Phenotype bulk platform registration failed. [CONTEXT service_id=204 ]
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: java.util.concurrent.TimeoutException: Timed out waiting for Task
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at bxxi.m(:com.google.android.gms@224417042@22.44.17 (190300-488410709):6)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at bpkg.f(:com.google.android.gms@224417042@22.44.17 (190300-488410709):9)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at com.google.android.gms.platformconfigurator.PhenotypeConfigurationUpdateListener.b(:com.google.android.gms@224417042@22.44.17 (190300-488410709):10)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at com.google.android.gms.platformconfigurator.PhenotypeConfigurationUpdateListener.c(:com.google.android.gms@224417042@22.44.17 (190300-488410709):0)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at com.google.android.gms.platformconfigurator.PhenotypeConfigurationUpdateListener.onHandleIntent(:com.google.android.gms@224417042@22.44.17 (190300-488410709):5)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at com.google.android.chimera.IntentOperation.onHandleIntent(:com.google.android.gms@224417042@22.44.17 (190300-488410709):2)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at xla.onHandleIntent(:com.google.android.gms@224417042@22.44.17 (190300-488410709):4)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at gug.run(:com.google.android.gms@224417042@22.44.17 (190300-488410709):3)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at guf.run(:com.google.android.gms@224417042@22.44.17 (190300-488410709):11)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at cvgw.run(:com.google.android.gms@224417042@22.44.17 (190300-488410709):2)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
11-22 01:43:11.466 27146 28951 W ConfiguratorPhenotypeUt: 	at java.lang.Thread.run(Thread.java:920)
11-22 01:43:11.467 27146 28951 I ExperimentPackageManage: Package com.google.android.as name not found! Using module version.
11-22 01:43:11.468 27146 28951 I ExperimentPackageManage: Package com.android.refreshratecontrol name not found! Using module version.
11-22 01:43:11.469 27146 28951 I ExperimentPackageManage: Package com.google.android.as name not found! Using module version.
11-22 01:43:11.470 27146 28951 I ExperimentPackageManage: Package com.google.android.apps.nexuslauncher name not found! Using module version.
```

philips77, 2022-11-23:
Hi, your logs are irrelevant. Do you have logs from the dfu service?
If the upload reaches 100% you should get a competition callback. Aren't you staying dfu again then?

dan-developer, 2022-11-23:
> Hi, your logs are irrelevant. Do you have logs from the dfu service? If the upload reaches 100% you should get a competition callback. Aren't you staying dfu again then?

After updated to 2.2.0, set MTU to lower value (23) and setForceScanningForNewAddressInLegacyDfu to true, the process is successfully finished.

closed 2022-11-23 by dan-developer