nordicsemi/Android-DFU-Library #372: 2.3.0 Build failed on Jitpack

QuentinFarizon, 2023-01-03:
Build failed for 2.3.0 on Jitpack , here is the build log : https://www.jitpack.io/com/github/NordicSemiconductor/Android-DFU-Library/2.3.0/build.log

The last successful version building on Jitpack is 1.11.0 : https://www.jitpack.io/com/github/NordicSemiconductor/Android-DFU-Library/1.11.0/build.log

Could you look into it ? This is a very useful way to retrieve dependencies.

QuentinFarizon, 2023-01-03:
From what I can see there are two separate issues :

- you need to add a jitpack.xml to indicate it requires Java 11 : https://jitpack.io/docs/BUILDING/#java-version

- There is an issue with signing a lib:dfu :
```
[$ Android-DFU-Library (main)]$ ./gradlew publishToMavenLocal
Checking the license for package Android SDK Platform 33 in /home/quentin/Android/Sdk/licenses
License for package Android SDK Platform 33 accepted.
Preparing "Install Android SDK Platform 33 (revision: 2)".
"Install Android SDK Platform 33 (revision: 2)" ready.
Installing Android SDK Platform 33 in /home/quentin/Android/Sdk/platforms/android-33
"Install Android SDK Platform 33 (revision: 2)" complete.
"Install Android SDK Platform 33 (revision: 2)" finished.

> Task :lib:dfu:compileReleaseJavaWithJavac
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.

> Task :lib:dfu:signMavenPublicationPublication FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':lib:dfu:signMavenPublicationPublication'.
> Cannot perform signing task ':lib:dfu:signMavenPublicationPublication' because it has no configured signatory

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 23s
26 actionable tasks: 26 executed
```

Thanks for looking into it !

philips77, 2023-01-09:
Why are you running `publishToMavenLocal` task? This is only used for releasing the library on Local Maven. You need to set up the PGP key, etc to make it work. The library is published on Maven Central, so you can just add dependency and that should work.

QuentinFarizon, 2023-01-09:
I was just testing locally to see if I could give more info on why the build was failing. Indeed it's not possible without the GP key.
It would still be great if this could work with Jitpack

QuentinFarizon, 2023-03-21:
@philips77 Up, it's still not working

sylwester-zielinski, 2023-03-21:
Hi, I copied to Android Studio command from Jitpack log: `./gradlew clean -Pgroup=com.github.NordicSemiconductor -Pversion=2.3.0 -xtest -xlint assemble publishToMavenLocal` and in my opinion argument `-Pgroup=com.github.NordicSemiconductor` is causing the problem. This can be the result of the project's module structure and overriding `group` may be causing this circular dependency problem to AS. You can try to build without it. Anyway, our release build is configured to work with our GithhubAction's configuration so you need to configure signing in the same way to make it working. Eventually you can try modifying Gradle scripts for your use case.

philips77, 2023-10-05:
I'm closing the issue as it became stale.

philips77, 2023-10-05:
Btw, the version 1.11 didn't have release scripts, as we were releasing manually back then.
Now there is a gradle task signing the release, which fails if keys are not set up.

closed 2023-10-05 by philips77