nordicsemi/Android-DFU-Library #373: context.startForegroundService(intent) from DfuServiceInitiator generates a runtime exception

mmontanes, 2023-01-25:
### Information
I develop a feature for a legacy app written in Java:
- Android compileSdkVersion  = 33
- Android buildToolsVersion = '33.0.0'
- Android minSdkVersion  = 23      Version code: M             // Android 6.0 (Marshmallow)
- Android targetSdkVersion  = 33  Version code: Tiramisu   // Android 13  (Tiramisu)

### DFU Bootloader version:
 - nRF5 SDK version: 17.1.0
 - Are you using bonding: no
 - DFU Library version: 2.3.0

### Device information:
 - Devices: Huawei Y5 2019 and Samsung Galaxy A13
 - OS: Android 9 (Huawei) and Adroid 13 (Samsung)

### Your question
The app crashes when I execute the following:
```java
if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
    DfuServiceInitiator.createDfuNotificationChannel(getActivity());
        final DfuServiceInitiator starter = new DfuServiceInitiator(incrementedDeviceAddress)
            .setDeviceName(incrementedDeviceAddress)
            .setKeepBond(false)
//            .setForceScanningForNewAddressInLegacyDfu(true)
            .setUnsafeExperimentalButtonlessServiceInSecureDfuEnabled(true)
            .setPrepareDataObjectDelay(1000L)
            .setZip(packageToInstallId);
        starter.start(getActivity(), DfuService.class);
}
```

### Logs
The logs that I get for the Huawei device is:

```
2023-01-25 10:52:25.100 27017-27083 ion                     com.xxxxxxxxx.xxxxxxxxx              E  ioctl c0044901 failed with code -1: Invalid argument
2023-01-25 10:52:25.101 27017-27083 ion                     com.xxxxxxxxx.xxxxxxxxx              E  ioctl c0044901 failed with code -1: Invalid argument
2023-01-25 10:52:25.108  1318-2038  <no-tag>                pid-1318                             E  [ZeroHung]zrhung_get_config: Get config failed for wp[0x0111]
2023-01-25 10:52:25.139  1318-1501  WifiVendorHal           pid-1318                             E  getWifiLinkLayerStats(l.939) failed {.code = ERROR_NOT_SUPPORTED, .description = }
2023-01-25 10:52:25.146  1318-1501  WificondControl         pid-1318                             E  Noise: 0, Snr: -1, Chload: -1
2023-01-25 10:52:25.214  1318-2961  <no-tag>                pid-1318                             E  [ZeroHung]zrhung_get_config: Get config failed for wp[0x0111]
2023-01-25 10:52:25.276   506-18025 ion                     pid-506                              E  ioctl c0044901 failed with code -1: Invalid argument
2023-01-25 10:52:25.276  1318-1453  WindowManager           pid-1318                             E  win=Window{71a9e4d u0 com.xxxxxxxxx.xxxxxxxxx/com.xxxxxxxxx.xxxxxxxxx.MainActivity EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true
2023-01-25 10:52:25.276   506-18025 ion                     pid-506                              E  ioctl c0044901 failed with code -1: Invalid argument
2023-01-25 10:52:25.328   543-1014  BufferQueueProducer     pid-543                              E  [] mConsumerName == NULL!!!!!!
2023-01-25 10:52:25.453 27017-27153 DfuBaseService          com.xxxxxxxxx.xxxxxxxxx              E  Device Address of firmware location are empty. Hint: use DfuServiceInitiator to start DFU
2023-01-25 10:52:25.479 27017-27017 aiser.bigkaise          com.xxxxxxxxx.xxxxxxxxx              E  [qarth_debug:]  get PatchStore::createDisableExceptionQarthFile method fail.
2023-01-25 10:52:25.484 27017-27017 AndroidRuntime          com.xxxxxxxxx.xxxxxxxxx              E  FATAL EXCEPTION: main
                                                                                                    Process: com.xxxxxxxxx.xxxxxxxxx, PID: 27017
                                                                                                    android.app.RemoteServiceException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{233cffe u0 com.xxxxxxxxx.xxxxxxxxx/.DfuService}
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2126)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:112)
                                                                                                    	at android.os.Looper.loop(Looper.java:216)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:7625)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
2023-01-25 10:52:25.493   662-2775  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.496   662-2775  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.497   662-4917  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.506   662-4917  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.508   662-4984  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.509   662-4984  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.511   662-4984  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.517   662-4984  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
2023-01-25 10:52:25.525   616-819   dubaid                  pid-616                              E  [DubaiUtils.h] parseMessage# Failed to find second tag: uid=
2023-01-25 10:52:25.525   616-819   dubaid                  pid-616                              E  [SystemHandler.cpp] parsePartialWakelockAcquiredMessage# Failed to parse count: lock=196652697 tag=IntentOp:.common.broadcast.BackgroundBroadcastReceiverSupport$PersistentReceiverIntentOperation count=0
2023-01-25 10:52:25.526   616-819   dubaid                  pid-616                              E  [SystemHandler.cpp] onPartialWakelockAcquireMessage# Failed to parse message
2023-01-25 10:52:25.531   662-4984  iMonitor                pid-662                              E  FaultDetect: DUMPTOOL_PRINTF return.
[... 67 lines cut ...]
}
```
So far so good, so I clicked on the previous line: `android.content.ContextWrapper.startForegroundService(ContextWrapper.java:839)`. Here rises the problem, I found the following code:

```java
@Override
    @UnsupportedAppUsage(maxTargetSdk = Build.VERSION_CODES.R, trackingBug = 170729553)
    public @Nullable ComponentName startForegroundServiceAsUser(Intent service, UserHandle user) {
        return mBase.startForegroundServiceAsUser(service, user);
    }
```

I don't know how this method was reached though I tried to follow the execution step by step with the debugger. The annotation rises the exception.

I decided to read the documentation of the `startForegroundService` method of the [`Context`] (https://developer.android.com/reference/android/content/Context#startForegroundService(android.content.Intent)) class. It shows the following message:

> 🔴 **Note:** Beginning with SDK Version [Build.VERSION_CODES.S](https://developer.android.com/reference/android/os/Build.VERSION_CODES#S), apps targeting SDK Version [Build.VERSION_CODES.S](https://developer.android.com/reference/android/os/Build.VERSION_CODES#S) or higher are not allowed to start foreground services from the background. See[ Behavior changes: Apps targeting Android 12 ](https://developer.android.com/about/versions/12/behavior-changes-12)for more details.

Then in the [behaviour changes for foreground services] (https://developer.android.com/about/versions/12/behavior-changes-12#foreground-service-launch-restrictions) I found the following:

[... cut ...]
> Consider using WorkManager to schedule and start [expedited work](https://developer.android.com/topic/libraries/architecture/workmanager/how-to/define-work#expedited) while your app runs in the background. To complete time-sensitive actions that the user requests, start foreground services within an [exact alarm](https://developer.android.com/training/scheduling/alarms#exact).

It might be possible that I did something wrong but me research makes me thing that your library doesn't support Android 12.

philips77, 2023-01-25:
If the service was started from background the log would say so.

The root cause of the issue is here:
> Device Address or firmware location are empty. Hint: use DfuServiceInitiator to start DFU

Seems like your `packageToInstallId` or `incrementedDeviceAddress ` is null.
The service starts, checks input data and returns immediately with that error. Then, as it didn't call `startService()`, it sends the unrelated crash.

philips77, 2023-01-25:
The service quits here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/39cd89639ec5080e298a3f5a46355b19fbcce387/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1120
but `Service.startForeground()` is called here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/39cd89639ec5080e298a3f5a46355b19fbcce387/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1148

Indeed, that call should be made earlier, but by ensuring your input parameters you won't have this issue.

mmontanes, 2023-01-26:
> If the service was started from background the log would say so.
>
> The root cause of the issue is here:
>
>
> Seems like your `packageToInstallId` or `incrementedDeviceAddress ` is null. The service [...]

It was clearly the problem. The `packageToInstallId` was equal to 0. The system has a problem reaching the file but never reported it as an error. Thank you very much.

closed 2023-01-26 by philips77