nordicsemi/Android-DFU-Library #375: dfu update  fail 133 error（Individual mobile phones）

yinxiaojun, 2023-02-02:
### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: [e.g. SDK 16]
 - Are you using bonding: [no]
 - DFU Library version: [e.g. 1.7.0]

### Device information (please complete the following information):
 - Device: []
 - OS: [e.g. Android 11]

### Your question
A clear and concise description of what the bug is.

### Logs
Add any other context or logs describing the problem here.
行 4859: P    [2023-01-10 23:28:48.992-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] dfuConfig is {"PRN":0,"deviceId":"390","filePath":"/data/user/0/com.watch.life/files/Veryfit/ota/390/14.zip","isNeedAuth":false,"isNeedReOpenBluetoothSwitchIfFailed":true,"macAddress":"D8:58:E7:2E:A3:DA","maxRetryTime":0,"otaWorkMode":0,"platform":-1}
	行 4863: P    [2023-01-10 23:28:48.995-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] to enter dfu mode
	行 4864: P    [2023-01-10 23:28:48.996-0300]        [IDO_BLE_DFU_NODIC   ]    [EnterDFUModeTask] start...
	行 4876: P    [2023-01-10 23:28:49.227-0300]        [IDO_BLE_DFU_NODIC   ]    [EnterDFUModeTask] enter dfu mode success!
	行 4877: P    [2023-01-10 23:28:49.227-0300]        [IDO_BLE_DFU_NODIC   ]    [EnterDFUModeTask] finished!
	行 4912: P    [2023-01-10 23:28:50.420-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] to scan target dfu Device.
	行 4913: P    [2023-01-10 23:28:50.421-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] start
	行 4915: P    [2023-01-10 23:28:50.427-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] startScanDevices()
	行 4916: P    [2023-01-10 23:28:50.637-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] -------onLeScan :D8:58:E7:2E:A3:DA
	行 4917: P    [2023-01-10 23:28:50.650-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] has find target device, is not in dfu mode
	行 4918: P    [2023-01-10 23:28:50.657-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] task finished.
	行 4919: P    [2023-01-10 23:28:50.665-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] (hasFindDeviceAndToConnectDevice) to connect Device
	行 4941: P    [2023-01-10 23:28:55.716-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...mac:D8:58:E7:2E:A3:DA
	行 4942: P    [2023-01-10 23:28:55.717-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...findDecive:false
	行 4943: P    [2023-01-10 23:28:55.724-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade... setForceSendInitFilefalse
	行 4945: P    [2023-01-10 23:28:56.285-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceConnected
	行 4946: P    [2023-01-10 23:28:56.286-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDfuProcessStarting
	行 4957: P    [2023-01-10 23:29:05.235-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceDisconnected
	行 4958: E    [2023-01-10 23:29:05.244-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] error=133, errorType=2,GATT ERROR
	行 4960: E    [2023-01-10 23:29:05.251-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] wait for restart ...1
	行 4962: P    [2023-01-10 23:29:20.255-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] restart, times is 1
	行 4963: P    [2023-01-10 23:29:20.257-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...mac:D8:58:E7:2E:A3:DA
	行 4964: P    [2023-01-10 23:29:20.257-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...findDecive:false
	行 4965: P    [2023-01-10 23:29:20.269-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade... setForceSendInitFilefalse
	行 4970: P    [2023-01-10 23:29:21.848-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceConnected
	行 4971: P    [2023-01-10 23:29:21.848-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDfuProcessStarting
	行 4981: P    [2023-01-10 23:29:30.283-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceDisconnecting
	行 4983: P    [2023-01-10 23:29:30.888-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceDisconnected
	行 4984: E    [2023-01-10 23:29:30.889-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] error=133, errorType=2,GATT ERROR
	行 4986: E    [2023-01-10 23:29:30.898-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] wait for restart ...2
	行 4988: P    [2023-01-10 23:29:45.903-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] restart, times is 2
	行 4989: P    [2023-01-10 23:29:45.905-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] to scan target dfu Device.
	行 4990: P    [2023-01-10 23:29:45.905-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] start
	行 4992: P    [2023-01-10 23:29:45.909-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] startScanDevices()
	行 4993: P    [2023-01-10 23:29:46.143-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] -------onLeScan :D8:58:E7:2E:A3:DA
	行 4994: P    [2023-01-10 23:29:46.144-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] has find target device, is in dfu mode:D8:58:E7:2E:A3:DA
	行 4995: P    [2023-01-10 23:29:46.144-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] task finished.
	行 4996: P    [2023-01-10 23:29:46.151-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...mac:D8:58:E7:2E:A3:DA
	行 4997: P    [2023-01-10 23:29:46.152-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...findDecive:true
	行 4998: P    [2023-01-10 23:29:46.154-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] removeBondState D8:58:E7:2E:A3:DA
	行 4999: P    [2023-01-10 23:29:46.164-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade... setForceSendInitFilefalse
	行 5002: P    [2023-01-10 23:29:47.973-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceConnected
	行 5003: P    [2023-01-10 23:29:47.974-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDfuProcessStarting
	行 5013: P    [2023-01-10 23:29:56.943-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceDisconnected
	行 5014: E    [2023-01-10 23:29:56.948-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] error=133, errorType=2,GATT ERROR
	行 5016: E    [2023-01-10 23:29:56.957-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] wait for restart ...3
	行 5018: P    [2023-01-10 23:30:11.962-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] restart, times is 3
	行 5019: P    [2023-01-10 23:30:11.964-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] findDecive trueupdatemac:D8:58:E7:2E:A3:DA
	行 5020: P    [2023-01-10 23:30:11.965-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] removeBondState D8:58:E7:2E:A3:DA
	行 5049: P    [2023-01-10 23:30:19.484-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] to scan target dfu Device.
	行 5050: P    [2023-01-10 23:30:19.485-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] start
	行 5052: P    [2023-01-10 23:30:19.528-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] startScanDevices()
	行 5063: P    [2023-01-10 23:30:19.702-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] -------onLeScan :D8:58:E7:2E:A3:DA
	行 5064: P    [2023-01-10 23:30:19.704-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] has find target device, is in dfu mode:D8:58:E7:2E:A3:DA
	行 5065: P    [2023-01-10 23:30:19.704-0300]        [IDO_BLE_DFU_NODIC   ]    [ScanTargetDFUDeviceTask] task finished.
	行 5066: P    [2023-01-10 23:30:19.719-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade...mac:D8:58:E7:2E:A3:DA
[... 28 lines cut ...]
	行 5133: P    [2023-01-10 23:31:11.498-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade... setForceSendInitFilefalse
	行 5137: P    [2023-01-10 23:31:13.277-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceConnected
	行 5138: P    [2023-01-10 23:31:13.279-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDfuProcessStarting
	行 5148: P    [2023-01-10 23:31:22.271-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] onDeviceDisconnected
[... cut ...]
	行 5179: E    [2023-01-10 23:31:48.301-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager:DfuProgressListener] error=133, errorType=2,GATT ERROR
	行 5181: E    [2023-01-10 23:31:48.311-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] wait for restart ...7
	行 5183: E    [2023-01-10 23:32:03.316-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] out of max retry times, upgrade failed, exit!
	行 5184: E    [2023-01-10 23:32:03.317-0300]        [IDO_BLE_DFU_NODIC   ]    [NodicDFUManager] upgrade failed, exit!

joernalraun, 2023-07-20:
I would like to additionally point out that the "133 Gatt error" appears in our tests on all Samsung S23 phones running Android 13 (so this might help in getting that error),  when trying to flash a Calliope mini or microbit V1. Is there any advice how to get around this?

MarioTheGreatest, 2024-04-08:
Did you find a solution?

philips77, 2024-10-25:
Hi,
Are you using this DFU Library? I don't recognize these logs. Could you paste here what you're getting from LogCat?

philips77, 2024-10-25:
And I'm also sorry for veeeeery long delay. I just came back to this library after a long time and trying to fix issues.

dinesharjani, 2026-08-21:
Closing due to lack of feedback. Please reopen if necessary.

closed 2026-08-21 by dinesharjani