nordicsemi/Android-DFU-Library #376: dfu update failed (device disconnected)

Rem193, 2023-02-08:
### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: 8.1
 - Are you using bonding: No
 - DFU Library version: 2.3.0

### Device information (please complete the following information):
 - Samsung S20 FE (android 13)
 - Redmi 9C (android 10)

### Your question
DFU doesn't work , I get errors:

[DFU] Device has disconnected , on Redmi 9C
[DFU] Connection failed (0x85): GATT ERROR,  on Samsung S20 FE

I notice when my phones are bonded to the nRF device the DFU is working.

With DFU library 1.11.0 is working on Redmi 9C with or without bonding.

### Logs

On Redmi 9C
```
V	14:32:01.612	[DFU] DFU service started
V	14:32:01.627	[DFU] Opening file...
I	14:32:01.632	[DFU] Firmware file opened successfully
V	14:32:01.640	[DFU] Connecting to DFU target...
D	14:32:01.647	[DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
D	14:32:01.915	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	14:32:01.930	[DFU] Connected to DF:F1:8D:6E:05:FD
V	14:32:01.941	[DFU] Discovering services...
D	14:32:01.958	[DFU] gatt.discoverServices()
I	14:32:02.929	[DFU] Services discovered
V	14:32:02.941	[DFU] Reading DFU version number...
D	14:32:02.951	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I	14:32:02.960	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A	14:32:02.969	[DFU] Version number read: 0.1
W	14:32:02.980	[DFU] Application with buttonless update found
V	14:32:02.991	[DFU] Jumping to the DFU Bootloader...
V	14:32:03.001	[DFU] Requesting new MTU...
D	14:32:03.011	[DFU] gatt.requestMtu(517)
I	14:32:03.021	[DFU] MTU changed to: 23
V	14:32:03.029	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	14:32:03.036	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D	14:32:03.045	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	14:32:03.086	[DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V	14:32:03.096	[DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A	14:32:03.105	[DFU] Notifications enabled
V	14:32:03.115	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D	14:32:03.125	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
A	14:32:03.187	[DFU] Jump to bootloader sent (Op Code = 1, Upload Mode = 4)
I	14:32:03.284	[DFU] Disconnected by the remote device
D	14:32:03.308	[DFU] gatt.refresh() (hidden)
D	14:32:03.335	[DFU] gatt.disconnect()
D	14:32:03.351	[DFU] gatt.close()
D	14:32:03.369	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
V	14:32:03.386	[DFU] DFU service started
I	14:32:03.394	[DFU] Firmware file opened successfully
V	14:32:03.400	[DFU] Connecting to DFU target...
D	14:32:03.405	[DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
D	14:32:03.871	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	14:32:03.898	[DFU] Connected to DF:F1:8D:6E:05:FD
V	14:32:03.922	[DFU] Discovering services...
D	14:32:03.958	[DFU] gatt.discoverServices()
I	14:32:04.696	[DFU] Services discovered
V	14:32:04.713	[DFU] Reading DFU version number...
D	14:32:04.727	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I	14:32:04.742	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 06-00
A	14:32:04.754	[DFU] Version number read: 0.6
V	14:32:04.766	[DFU] Requesting new MTU...
[... 67 lines cut ...]
D 14:32:04.727 [DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
I 14:32:04.742 [DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 06-00
A 14:32:04.754 [DFU] Version number read: 0.6
V 14:32:04.766 [DFU] Requesting new MTU...
D 14:32:04.779 [DFU] gatt.requestMtu(517)
I 14:32:04.799 [DFU] MTU changed to: 23
V 14:32:04.812 [DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D 14:32:04.825 [DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D 14:32:04.835 [DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I 14:32:04.866 [DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123, value (0x): 01-00
V 14:32:04.877 [DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A 14:32:04.885 [DFU] Notifications enabled
V 14:32:04.895 [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123
D 14:32:04.905 [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
I 14:32:04.913 [DFU] Data written to 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
A 14:32:04.922 [DFU] DFU Start sent (Op Code = 1, Upload Mode = 4)
V 14:32:04.928 [DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123
D 14:32:04.935 [DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
I 14:32:04.942 [DFU] Data written to 00001532-1212-efde-1523-785feabcd123, value (0x): 00-00-00-00-00-00-00-00-90-C3-00-00
A 14:32:04.947 [DFU] Firmware image size sent (0b, 0b, 50064b)
E 14:32:09.287 [DFU] Device has disconnected
D 14:32:09.314 [DFU] gatt.disconnect()
D 14:32:09.335 [DFU] gatt.close()
D 14:32:09.375 [DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

philips77, 2023-10-03:
Hello,
First of all, sorry for super late response. I was buried with other tasks.

> DFU library 1.11.0 is working on Redmi 9C with or without bonding.

I wonder why. In Legacy DFU, unless you use `forceScan` flag, the bootloader is assumed to use the same MAC address then the app. In both logs above it seems like the reconnection to the bootlaoder succeeded, the correct Version number is read and only after sending the sizes the device disconnects. 

Are you sending the same fw using 2.3.0 and 1.11.0? Could you share log from 1.11.0?

Rem193, 2023-10-03:
Hello, yes same FW.

I don't call setForceScanningForNewAddressInLegacyDfu() in my code.

I used the same FW and same phone Redmi 9C.

> Are you sending the same fw using 2.3.0 and 1.11.0? Could you share log from 1.11.0?

Yes same fw, I bumped on my project the DFU library from 1.11.0 to 1.12.0 and it work well too

**How I initialize the DfuServiceInitiator**
```
// Kick off DFU
final DfuServiceInitiator starter = new DfuServiceInitiator(bleAddress)
	.setDeviceName(bluetoothManager.getAdapter().getRemoteDevice(bleAddress).getName())
				.setForceDfu(true)
				.setNumberOfRetries(DFU_NUMBER_OF_RETRIES)
				.setPacketsReceiptNotificationsEnabled(true)
				.setPacketsReceiptNotificationsValue(PACKETS_RECEIPT_NOTIFICATIONS_VALUE)
				.setKeepBond(keepBond);
```

**Logs**

**DFU lib 2.3.0**
```2023-10-03 13:58:39.157 16070-16396 DfuImpl              I  Scanning for new address finished with: null
2023-10-03 13:58:39.180 16070-16396 DfuBaseService       I  Starting DFU service in foreground
2023-10-03 13:58:39.223 16070-16396 DfuBaseService       I  Connecting to the device...
2023-10-03 13:58:39.226 16070-16396 BluetoothGatt        D  connect() - device: EA:47:BA:46:6D:AB, auto: false
2023-10-03 13:58:39.226 16070-16396 BluetoothGatt        D  registerApp()
2023-10-03 13:58:39.226 16070-16396 BluetoothGatt        D  registerApp() - UUID=8f634e8f-bacb-4580-8d66-8a9cbe911629
2023-10-03 13:58:39.231 16070-16213 BluetoothGatt        D  onClientRegistered() - status=0 clientIf=8
2023-10-03 13:59:09.256 16070-16213 BluetoothGatt        D  onClientConnectionState() - status=133 clientIf=8 device=EA:47:BA:46:6D:AB
2023-10-03 13:59:09.257 16070-16213 DfuBaseService       E  Connection state change error: 133 newState: 0
2023-10-03 13:59:09.257 16070-16396 DfuBaseService       I  Connection error after: 30036 ms
2023-10-03 13:59:09.258 16070-16396 DfuBaseService       E  Device not reachable. Check if the device with address EA:47:BA:46:6D:AB is in range, is advertising and is connectable
2023-10-03 13:59:09.259 16070-16396 DfuBaseService       I  Attempt: 1
2023-10-03 13:59:09.260 16070-16396 BluetoothGatt        D  refresh() - device: EA:47:BA:46:6D:AB
2023-10-03 13:59:09.263 16070-16396 DfuBaseService       I  Refreshing result: true
2023-10-03 13:59:09.263 16070-16396 DfuBaseService       I  Cleaning up...
2023-10-03 13:59:09.263 16070-16396 BluetoothGatt        D  cancelOpen() - device: EA:47:BA:46:6D:AB
2023-10-03 13:59:09.266 16070-16396 BluetoothGatt        D  close()
2023-10-03 13:59:09.266 16070-16396 BluetoothGatt        D  unregisterApp() - mClientIf=8
2023-10-03 13:59:09.268 16070-16396 DfuBaseService       I  Restarting the service
2023-10-03 13:59:09.289 16070-16396 DfuBaseService       I  Starting DFU service in foreground
2023-10-03 13:59:09.316 16070-16396 DfuBaseService       I  Connecting to the device...
2023-10-03 13:59:09.319 16070-16396 BluetoothGatt        D  connect() - device: EA:47:BA:46:6D:AB, auto: false
2023-10-03 13:59:09.319 16070-16396 BluetoothGatt        D  registerApp()
2023-10-03 13:59:09.319 16070-16396 BluetoothGatt        D  registerApp() - UUID=fd1557c6-0b8c-405e-85a3-af2b8b335726
2023-10-03 13:59:09.326 16070-16213 BluetoothGatt        D  onClientRegistered() - status=0 clientIf=8
2023-10-03 13:59:39.346 16070-16213 BluetoothGatt        D  onClientConnectionState() - status=133 clientIf=8 device=EA:47:BA:46:6D:AB
2023-10-03 13:59:39.346 16070-16213 DfuBaseService       E  Connection state change error: 133 newState: 0
2023-10-03 13:59:39.347 16070-16396 DfuBaseService       I  Connection error after: 30033 ms
2023-10-03 13:59:39.347 16070-16396 DfuBaseService       E  Device not reachable. Check if the device with address EA:47:BA:46:6D:AB is in range, is advertising and is connectable
2023-10-03 13:59:39.348 16070-16396 DfuBaseService       I  Attempt: 2
2023-10-03 13:59:39.349 16070-16396 BluetoothGatt        D  refresh() - device: EA:47:BA:46:6D:AB
2023-10-03 13:59:39.352 16070-16396 DfuBaseService       I  Refreshing result: true
2023-10-03 13:59:39.352 16070-16396 DfuBaseService       I  Cleaning up...
2023-10-03 13:59:39.353 16070-16396 BluetoothGatt        D  cancelOpen() - device: EA:47:BA:46:6D:AB
2023-10-03 13:59:39.358 16070-16396 BluetoothGatt        D  close()
2023-10-03 13:59:39.358 16070-16396 BluetoothGatt        D  unregisterApp() - mClientIf=8
2023-10-03 13:59:39.361 16070-16396 DfuBaseService       I  Restarting the service
2023-10-03 13:59:39.380 16070-16396 DfuBaseService       I  Starting DFU service in foreground
2023-10-03 13:59:39.405 16070-16396 DfuBaseService       I  Connecting to the device...
2023-10-03 13:59:39.410 16070-16396 BluetoothGatt        D  connect() - device: EA:47:BA:46:6D:AB, auto: false
2023-10-03 13:59:39.410 16070-16396 BluetoothGatt        D  registerApp()
2023-10-03 13:59:39.410 16070-16396 BluetoothGatt        D  registerApp() - UUID=eabd4319-5649-4408-96fc-77c422e97aca
2023-10-03 13:59:39.417 16070-16213 BluetoothGatt        D  onClientRegistered() - status=0 clientIf=8
2023-10-03 13:59:40.088 16070-16070 DfuBaseService       I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2023-10-03 13:59:40.095 16070-16213 BluetoothGatt        D  onClientConnectionState() - status=0 clientIf=8 device=EA:47:BA:46:6D:AB
[... 119 lines cut ...]
2023-10-03 14:12:47.246 18024-18454 BluetoothGatt     D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
[... cut ...]
2023-10-03 14:12:57.373 18024-18554 DfuImpl           I  Sending init packet (Value = FF-FF-DD-DD-FF-FF-FF-FF-01-00-64-00-DF-26)
2023-10-03 14:12:57.382 18024-18554 DfuImpl           I  Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
2023-10-03 14:12:57.462 18024-18554 DfuImpl           I  Sending the number of packets before notifications (Op Code = 8, Value = 12)
2023-10-03 14:12:57.508 18024-18554 DfuImpl           I  Sending Receive Firmware Image request (Op Code = 3)
2023-10-03 14:12:57.577 18024-18554 DfuImpl           I  Uploading firmware...
```

philips77, 2023-10-03:
Did you try disabling MTU request when using 2.3.0?
It can be fine using `DfuServiceInitiator`.
I'll go though the difference between those 2 tomorrow.

Rem193, 2023-10-04:
I disabled the MTU request with 2.3.0 but I got the same behavior.

**DfuServiceInitiator**
```
// Kick off DFU
final DfuServiceInitiator starter = new DfuServiceInitiator(bleAddress)
		.setDeviceName(bluetoothManager.getAdapter().getRemoteDevice(bleAddress).getName())
		.setForceDfu(true)
		.disableMtuRequest()
		.setNumberOfRetries(DFU_NUMBER_OF_RETRIES)
		.setPacketsReceiptNotificationsEnabled(true)
		.setPacketsReceiptNotificationsValue(PACKETS_RECEIPT_NOTIFICATIONS_VALUE)
		.setKeepBond(keepBond);
```

**Logs 2.3.0**
```
2023-10-04 09:42:19.854  7438-7951  DfuBaseService       I  Starting DFU service in foreground
2023-10-04 09:42:19.895  7438-7951  DfuBaseService       I  Connecting to the device...
2023-10-04 09:42:19.898  7438-7951  BluetoothGatt        D  connect() - device: EA:47:BA:46:6D:AB, auto: false
2023-10-04 09:42:19.899  7438-7951  BluetoothGatt        D  registerApp()
2023-10-04 09:42:19.899  7438-7951  BluetoothGatt        D  registerApp() - UUID=803c2bb5-b4a9-4d68-ac5d-021cb39482e0
2023-10-04 09:42:19.902  7438-7456  BluetoothGatt        D  onClientRegistered() - status=0 clientIf=8
2023-10-04 09:42:20.345  7438-7438  DfuBaseService       I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2023-10-04 09:42:20.346  7438-7456  BluetoothGatt        D  onClientConnectionState() - status=0 clientIf=8 device=EA:47:BA:46:6D:AB
2023-10-04 09:42:20.346  7438-7456  DfuBaseService       I  Connected to GATT server
2023-10-04 09:42:20.350  7438-7456  BluetoothGatt        D  discoverServices() - device: EA:47:BA:46:6D:AB
2023-10-04 09:42:20.352  7438-7456  DfuBaseService       I  Attempting to start service discovery... succeed
2023-10-04 09:42:20.844  7438-7456  BluetoothGatt        D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
2023-10-04 09:42:21.049  7438-7456  BluetoothGatt        D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=6 latency=0 timeout=500 status=0
2023-10-04 09:42:21.193  7438-7456  BluetoothGatt        D  onSearchComplete() = Device=EA:47:BA:46:6D:AB Status=0
2023-10-04 09:42:21.193  7438-7456  DfuBaseService       I  Services discovered
2023-10-04 09:42:21.214  7438-7951  DfuImpl              I  Reading DFU version number...
2023-10-04 09:42:21.237  7438-7951  DfuImpl              I  Version number read: 0.6 -> Bootloader from SDK 8.0 or newer. Bond sharing supported
2023-10-04 09:42:21.241  7438-7951  DfuImpl              W  Legacy DFU bootloader found
2023-10-04 09:42:21.242  7438-7951  DfuImpl              I  Enabling notifications...
2023-10-04 09:42:21.243  7438-7951  BluetoothGatt        D  setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
2023-10-04 09:42:21.253  7438-7456  BluetoothGatt        D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
2023-10-04 09:42:21.300  7438-7951  DfuImpl              I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2023-10-04 09:42:21.344  7438-7951  DfuImpl              I  Sending image size array to DFU Packet (0b, 0b, 57684b)
2023-10-04 09:42:21.781  7438-7456  BluetoothGatt        D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
2023-10-04 09:42:25.786  7438-7456  BluetoothGatt        D  onClientConnectionState() - status=8 clientIf=8 device=EA:47:BA:46:6D:AB
2023-10-04 09:42:25.786  7438-7456  DfuBaseService       W  Target device disconnected with status: 8
2023-10-04 09:42:25.787  7438-7951  DfuBaseService       E  Unable to write Op Code: device disconnected
2023-10-04 09:42:25.787  7438-7951  DfuBaseService       I  Cleaning up...
2023-10-04 09:42:25.788  7438-7951  BluetoothGatt        D  cancelOpen() - device: EA:47:BA:46:6D:AB
2023-10-04 09:42:25.791  7438-7951  BluetoothGatt        D  close()
2023-10-04 09:42:25.792  7438-7951  BluetoothGatt        D  unregisterApp() - mClientIf=8
2023-10-04 09:42:25.794  7438-7951  DfuBaseService       I  Restarting the service (1 /3)
2023-10-04 09:42:25.798  7438-7438  DfuBaseService       I  Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2023-10-04 09:42:25.814  7438-7951  DfuBaseService       I  Starting DFU service in foreground
2023-10-04 09:42:25.872  7438-7951  DfuBaseService       I  Connecting to the device...
2023-10-04 09:42:25.874  7438-7951  BluetoothGatt        D  connect() - device: EA:47:BA:46:6D:AB, auto: false
2023-10-04 09:42:25.875  7438-7951  BluetoothGatt        D  registerApp()
2023-10-04 09:42:25.875  7438-7951  BluetoothGatt        D  registerApp() - UUID=b9ca2e06-c459-4a90-a26f-010691bab74d
2023-10-04 09:42:25.883  7438-7456  BluetoothGatt        D  onClientRegistered() - status=0 clientIf=8
2023-10-04 09:42:26.249  7438-7438  DfuBaseService       I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2023-10-04 09:42:26.253  7438-7909  BluetoothGatt        D  onClientConnectionState() - status=0 clientIf=8 device=EA:47:BA:46:6D:AB
2023-10-04 09:42:26.254  7438-7909  DfuBaseService       I  Connected to GATT server
2023-10-04 09:42:26.258  7438-7909  BluetoothGatt        D  discoverServices() - device: EA:47:BA:46:6D:AB
2023-10-04 09:42:26.261  7438-7909  DfuBaseService       I  Attempting to start service discovery... succeed
2023-10-04 09:42:26.679  7438-7909  BluetoothGatt        D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=6 latency=0 timeout=500 status=0
[... cut ...]
2023-10-04 09:42:36.695  7438-7951  DfuBaseService       I  Cleaning up...
2023-10-04 09:42:36.695  7438-7951  BluetoothGatt        D  cancelOpen() - device: EA:47:BA:46:6D:AB
2023-10-04 09:42:36.696  7438-7951  BluetoothGatt        D  close()
2023-10-04 09:42:36.697  7438-7951  BluetoothGatt        D  unregisterApp() - mClientIf=8
2023-10-04 09:42:36.698  7438-7951  DfuImpl              I  Restarting to bootloader mode
2023-10-04 09:42:36.703  7438-7438  DfuBaseService       I  Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

philips77, 2023-10-04:
Hmm.. I see that 1.11 spends almost 7 seconds erasing the secondary slot:
```
2023-10-03 14:12:50.746 I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2023-10-03 14:12:50.804 I  Sending image size array to DFU Packet (0b, 0b, 57684b)
// 6.5 seconds pass
2023-10-03 14:12:57.329 I  Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
2023-10-03 14:12:57.373 I  Sending 14 bytes of init packet
2023-10-03 14:12:57.373 I  Sending init packet (Value = FF-FF-DD-DD-FF-FF-FF-FF-01-00-64-00-DF-26)
2023-10-03 14:12:57.382 I  Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)

```
but 2.3.0 gets a disconnection after just 4:
```
2023-10-04 09:42:21.344  I  Sending image size array to DFU Packet (0b, 0b, 57684b)
2023-10-04 09:42:21.781  D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
// disconnection after 4 seconds
2023-10-04 09:42:25.786  D  onClientConnectionState() - status=8 clientIf=8 device=EA:47:BA:46:6D:AB

```

Rem193, 2023-10-25:
Still have the issue with some mobiles with the latest app DFU 2.3.1

* nRF5 SDK version: 8.1
* Are you using bonding: No
* DFU Library version: 2.3.1

```
DFU, 2023-10-25
phone: Redmi 9C
os version: Android 10
bluetooth version: Bluetooth 5.0

Device (XX:XX:XX:XX:FB:B0)
V	13:56:31.834	[DFU] DFU service started
V	13:56:31.848	[DFU] Opening file...
W	13:56:32.167	[DFU] Firmware file opened successfully
V	13:56:32.176	[DFU] Connecting to DFU target...
D	13:56:32.184	[DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
D	13:56:32.820	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
W	13:56:32.832	[DFU] Connected to XX:XX:XX:XX:FB:B0
V	13:56:32.841	[DFU] Discovering services...
D	13:56:32.849	[DFU] gatt.discoverServices()
W	13:56:33.805	[DFU] Services discovered
V	13:56:33.814	[DFU] Reading DFU version number...
D	13:56:33.823	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
W	13:56:33.831	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 01-00
A	13:56:33.842	[DFU] Version number read: 0.1
W	13:56:33.852	[DFU] Application with buttonless update found
V	13:56:33.860	[DFU] Jumping to the DFU Bootloader...
V	13:56:33.870	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	13:56:33.880	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D	13:56:33.889	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
W	13:56:33.904	[DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123
V	13:56:33.919	[DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A	13:56:33.936	[DFU] Notifications enabled
V	13:56:33.952	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
D	13:56:33.957	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
A	13:56:33.963	[DFU] Jump to bootloader sent (Op Code = 1, Upload Mode = 4)
V	13:56:33.968	[DFU] Disconnecting...
D	13:56:33.974	[DFU] gatt.disconnect()
W	13:56:33.979	[DFU] Disconnected
D	13:56:33.984	[DFU] gatt.refresh() (hidden)
D	13:56:33.996	[DFU] gatt.disconnect()
D	13:56:34.004	[DFU] gatt.close()
V	13:56:34.013	[DFU] Scanning for the DFU Bootloader... (timeout 2000 ms)
D	13:56:34.025	[DFU] wait(1000)
D	13:56:34.031	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
W	13:56:35.094	[DFU] DFU Bootloader found with address XX:XX:XX:XX:FB:B0
V	13:56:35.130	[DFU] DFU service started
W	13:56:35.142	[DFU] Firmware file opened successfully
V	13:56:35.151	[DFU] Connecting to DFU target...
D	13:56:35.157	[DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
D	13:56:35.396	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
W	13:56:35.428	[DFU] Connected to XX:XX:XX:XX:FB:B0
V	13:56:35.450	[DFU] Discovering services...
D	13:56:35.464	[DFU] gatt.discoverServices()
W	13:56:36.220	[DFU] Services discovered
V	13:56:36.237	[DFU] Reading DFU version number...
D	13:56:36.250	[DFU] gatt.readCharacteristic(00001534-1212-efde-1523-785feabcd123)
W	13:56:36.263	[DFU] Read Response received from 00001534-1212-efde-1523-785feabcd123, value (0x): 06-00
A	13:56:36.274	[DFU] Version number read: 0.6
V	13:56:36.285	[DFU] Enabling notifications for 00001531-1212-efde-1523-785feabcd123
D	13:56:36.298	[DFU] gatt.setCharacteristicNotification(00001531-1212-efde-1523-785feabcd123, true)
D	13:56:36.310	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
W	13:56:36.325	[DFU] Data written to descr.00001531-1212-efde-1523-785feabcd123
V	13:56:36.338	[DFU] Notifications enabled for 00001531-1212-efde-1523-785feabcd123
A	13:56:36.351	[DFU] Notifications enabled
V	13:56:36.359	[DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123, value (0x): 01-04
D	13:56:36.368	[DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123)
W	13:56:36.375	[DFU] Data written to 00001531-1212-efde-1523-785feabcd123
A	13:56:36.384	[DFU] DFU Start sent (Op Code = 1, Upload Mode = 4)
V	13:56:36.397	[DFU] Writing to characteristic 00001532-1212-efde-1523-785feabcd123, value (0x): 00-00-00-00-00-00-00-00-54-E1-00-00
D	13:56:36.405	[DFU] gatt.writeCharacteristic(00001532-1212-efde-1523-785feabcd123)
W	13:56:36.415	[DFU] Data written to 00001532-1212-efde-1523-785feabcd123
A	13:56:36.425	[DFU] Firmware image size sent (0b, 0b, 57684b)
E	13:56:40.830	[DFU] Device has disconnected
D	13:56:40.849	[DFU] gatt.disconnect()
D	13:56:40.863	[DFU] gatt.close()
D	13:56:40.897	[DFU] [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED

```

philips77, 2023-10-25:
This, actually, indeed may be due to the remote not sending any packets while it erases the slot.
Depending on Supervision Timeout in Connection Parameters the Central will disconnect if nothing is received. ST used to be very long, like 20 sec, not it's around 4-5 seconds, but may vary on different phones.

Have a look at the log I pasted before:
```
2023-10-04 09:42:21.344  I  Sending image size array to DFU Packet (0b, 0b, 57684b)
2023-10-04 09:42:21.781  D  onConnectionUpdated() - Device=EA:47:BA:46:6D:AB interval=18 latency=0 timeout=400 status=0
// disconnection after 4 seconds
2023-10-04 09:42:25.786  D  onClientConnectionState() - status=8 clientIf=8 device=EA:47:BA:46:6D:AB
```
The Supervision Timeout was set to 400, which translates to 4 seconds. And exactly after 4 seconds the connection is terminated.

If it is possible to modify the fw to request longer connection interval, that would help. Or, we can add a Connection Priority request from Android just before sending the sizes. Starting from Android 8+ Android requests 5 seconds Supervision Timeout, but seems like you need almost 7.

I wonder why 1.11 works...

philips77, 2023-10-25:
The logs from 1.11 you pasted above also have 4 seconds Supervision Timeout:
```
2023-10-03 14:12:48.483 18024-18454 BluetoothGatt     D  onConnectionUpdated() - Device=XX:XX:XX:XX:XX:XX interval=18 latency=0 timeout=400 status=0
[...]
2023-10-03 14:12:50.804 18024-18554 DfuImpl           I  Sending image size array to DFU Packet (0b, 0b, 57684b)
// 6.5 seconds
2023-10-03 14:12:57.329 18024-18554 DfuImpl           I  Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
```

philips77, 2023-10-25:
Perhaps it's not the Supervision Timeout, after all.

Rem193, 2024-08-19:
I tried on an other phone (samsung a15) with the latest version (2.5.0) of DFU lib but I still have an issue with the latest one and I notice that with an old version (2.1.0) the update works

I tested both versions on the same phone, same application I only changed the used version of the dfu library.
I also tested with the latest nRF Device Firmware Update app from the play store, the FW update doesn't work too.

### DFU Bootloader version:
 - nRF5 SDK version: 8.1
 - Are you using bonding: No
 - DFU Library version: 2.5.0

### Device information:
 - Samsung Galaxy A15 (Android 14)

#### With Android DFU lib 2.5.0 (not working)
`no.nordicsemi.android:dfu:2.5.0`
```
2024-08-19 14:49:54.071  DfuBaseService  I  DFU service created. Version: 2.5.0
2024-08-19 14:49:54.100  DfuBaseService  I  Starting DFU service in foreground
2024-08-19 14:49:54.121  DfuBaseService  I  Connecting to the device...
2024-08-19 14:49:54.409  DfuBaseService  I  Connected to GATT server
2024-08-19 14:49:54.415  DfuBaseService  I  Attempting to start service discovery... succeed
2024-08-19 14:49:54.419  DfuBaseService  I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2024-08-19 14:49:55.182  DfuBaseService  I  Services discovered
2024-08-19 14:49:55.190  DfuImpl         I  Reading DFU version number...
2024-08-19 14:49:55.214  DfuImpl         I  Version number read: 0.6 -> Bootloader from SDK 8.0 or newer. Bond sharing supported
2024-08-19 14:49:55.215  DfuImpl         W  Legacy DFU bootloader found
2024-08-19 14:49:55.215  DfuImpl         I  Requesting MTU = 517
2024-08-19 14:49:55.260  DfuImpl         I  MTU changed to: 23
2024-08-19 14:49:55.261  DfuImpl         I  Enabling notifications...
2024-08-19 14:49:55.319  DfuImpl         I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2024-08-19 14:49:55.365  DfuImpl         I  Sending image size array to DFU Packet (0b, 0b, 58628b)
2024-08-19 14:50:00.133  DfuBaseService  W  Target device disconnected with status: 8
2024-08-19 14:50:00.134  DfuBaseService  E  Unable to write Op Code: device disconnected
2024-08-19 14:50:00.135  DfuBaseService  I  Cleaning up...
2024-08-19 14:50:00.145  DfuBaseService  I  Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

#### With Android DFU lib 2.1.0 (working)
`no.nordicsemi.android:dfu:2.1.0`
```
2024-08-19 14:52:50.627  DfuBaseService  I  Starting DFU service in foreground
2024-08-19 14:52:52.638  DfuBaseService  I  Connecting to the device...
2024-08-19 14:52:52.901  DfuBaseService  I  Connected to GATT server
2024-08-19 14:52:52.906  DfuBaseService  I  Attempting to start service discovery... succeed
2024-08-19 14:52:53.291  DfuBaseService  I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2024-08-19 14:52:53.642  DfuBaseService  I  Services discovered
2024-08-19 14:52:54.670  DfuImpl         I  Reading DFU version number...
2024-08-19 14:52:54.739  DfuImpl         I  Version number read: 0.6 -> Bootloader from SDK 8.0 or newer. Bond sharing supported
2024-08-19 14:52:54.742  DfuImpl         W  Legacy DFU bootloader found
2024-08-19 14:52:54.742  DfuImpl         I  Requesting MTU = 517
2024-08-19 14:52:54.788  DfuImpl         I  MTU changed to: 23
2024-08-19 14:52:55.791  DfuImpl         I  Enabling notifications...
2024-08-19 14:52:56.840  DfuImpl         I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2024-08-19 14:52:56.901  DfuImpl         I  Sending image size array to DFU Packet (0b, 0b, 58628b)
2024-08-19 14:53:03.427  DfuImpl         I  Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
2024-08-19 14:53:03.467  DfuImpl         I  Sending 14 bytes of init packet
2024-08-19 14:53:03.467  DfuImpl         I  Sending init packet (Value = FF-FF-DD-DD-FF-FF-FF-FF-01-00-64-00-8F-27)
2024-08-19 14:53:03.477  DfuImpl         I  Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
2024-08-19 14:53:03.540  DfuImpl         I  Sending the number of packets before notifications (Op Code = 8, Value = 12)
2024-08-19 14:53:03.602  DfuImpl         I  Sending Receive Firmware Image request (Op Code = 3)
2024-08-19 14:53:03.671  DfuImpl         I  Uploading firmware...
```

#### How I initialize the DfuServiceInitiator
```
		// Kick off DFU
		final DfuServiceInitiator starter = new DfuServiceInitiator(targetedBleAddress)
				.setDeviceName(bluetoothManager.getAdapter().getRemoteDevice(name)
				.setNumberOfRetries(DFU_NUMBER_OF_RETRIES)
				.setForceDfu(true)
				.setPacketsReceiptNotificationsEnabled(true)
				.setPacketsReceiptNotificationsValue(PACKETS_RECEIPT_NOTIFICATIONS_VALUE)
				.setKeepBond(keepBond);
```

philips77, 2024-10-25:
Hi,
I don't know how library version may cause a timeout in BLE communication. I'm about to release a new version (2.7.0), please give it a spin.

tudang88, 2025-01-22:
Hi everyone,
Do you know the cause and solution for this issue?
The same issue occurs with library version 2.9.0. 
Confirm with Samsung Galaxy s23/ android 14 and 15 also.
Thank you very much.

Rem193, 2025-03-11:
Hi, @tudang88 

I received a customer message this morning he has a Samsung S23 and the DFU does not work either...

@philips77 I tested the 2.9.0, on some phones the DFU work well while on others the update systematically fails. I will dump the logs soon on this thread.

Rem193, 2025-03-11:
DFU Bootloader version (please complete the following information):
nRF5 SDK version: 8.1
Are you using bonding: Yes
DFU Library version: 2.9.0
Device information (please complete the following information):
Samsung Galaxy A145R/DSN (Android 14)

```
2025-03-11 16:26:59.700 22627-23765 DfuBaseService                   I  Connecting to the device...
2025-03-11 16:27:00.249 22627-23535 DfuBaseService                   I  Connected to GATT server
2025-03-11 16:27:00.249 22627-23535 DfuBaseService                   I  Waiting 4000 ms for a possible Service Changed indication...
2025-03-11 16:27:00.251 22627-22627 DfuBaseService                   I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2025-03-11 16:27:00.931 22627-23535 DfuBaseService                   I  Service Changed indication received
2025-03-11 16:27:00.936 22627-23535 DfuBaseService                   I  Attempting to start service discovery... succeed
2025-03-11 16:27:01.331 22627-22669 DfuBaseService                   I  Services discovered
2025-03-11 16:27:01.333 22627-22627 GollumBaseFrag                   D  cbOnProgress: title: Démarrage DFU…, message: Initialisation du processus DFU…, progress: 0, exception: null
2025-03-11 16:27:01.347 22627-23765 DfuImpl                          I  Reading DFU version number...
2025-03-11 16:27:01.373 22627-23765 DfuImpl                          I  Version number read: 0.6 -> Bootloader from SDK 8.0 or newer. Bond sharing supported
2025-03-11 16:27:01.373 22627-23765 DfuImpl                          I  Reading Service Changed CCCD value...
2025-03-11 16:27:01.433 22627-23765 DfuImpl                          I  Service Changed indications enabled
2025-03-11 16:27:01.433 22627-23765 DfuImpl                          W  Legacy DFU bootloader found
2025-03-11 16:27:01.434 22627-23765 DfuImpl                          I  Requesting MTU = 517
2025-03-11 16:27:01.484 22627-23535 DfuImpl                          I  MTU changed to: 23
2025-03-11 16:27:01.485 22627-23765 DfuImpl                          I  Enabling notifications...
2025-03-11 16:27:01.524 22627-23765 DfuImpl                          I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2025-03-11 16:27:01.569 22627-23765 DfuImpl                          I  Sending image size array to DFU Packet (0b, 0b, 58628b)
2025-03-11 16:27:06.357 22627-23219 DfuBaseService                   W  Target device disconnected with status: 8
2025-03-11 16:27:06.358 22627-23765 DfuBaseService                   E  Response not received: device disconnected (error 8)
2025-03-11 16:27:06.359 22627-23765 DfuBaseService                   I  Cleaning up...
2025-03-11 16:27:06.392 22627-23765 DfuBaseService                   I  Restarting the service (1 /3)
2025-03-11 16:27:06.404 22627-22627 DfuBaseService                   I  Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2025-03-11 16:27:06.432 22627-23765 DfuBaseService                   W  Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
                                                                                                    Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
2025-03-11 16:27:06.457 22627-23765 DfuBaseService                   I  Connecting to the device...
2025-03-11 16:27:07.415 22627-23219 DfuBaseService                   I  Connected to GATT server
2025-03-11 16:27:07.415 22627-23219 DfuBaseService                   I  Waiting 4000 ms for a possible Service Changed indication...
2025-03-11 16:27:07.453 22627-22627 DfuBaseService                   I  Action received: android.bluetooth.device.action.ACL_CONNECTED
2025-03-11 16:27:11.442 22627-22627 DfuBaseService                   I  Attempting to start service discovery... succeed
2025-03-11 16:27:11.447 22627-23219 DfuBaseService                   I  Services discovered
2025-03-11 16:27:11.454 22627-22627 GollumBaseFrag                   D  cbOnProgress: title: Démarrage DFU…, message: Initialisation du processus DFU…, progress: 0, exception: null
2025-03-11 16:27:11.463 22627-23765 DfuImpl                          I  Reading DFU version number...
2025-03-11 16:27:11.564 22627-23765 DfuImpl                          I  Version number read: 20.2 -> Unknown version
2025-03-11 16:27:11.565 22627-23765 DfuImpl                          I  Reading Service Changed CCCD value...
2025-03-11 16:27:11.710 22627-23765 DfuImpl                          I  Enabling indications...
2025-03-11 16:27:11.905 22627-23765 DfuImpl                          I  Service Changed indications enabled
2025-03-11 16:27:11.905 22627-23765 DfuImpl                          W  Legacy DFU bootloader found
2025-03-11 16:27:11.906 22627-23765 DfuImpl                          I  Requesting MTU = 517
2025-03-11 16:27:12.003 22627-23219 DfuImpl                          I  MTU changed to: 23
2025-03-11 16:27:12.003 22627-23765 DfuImpl                          I  Enabling notifications...
2025-03-11 16:27:12.100 22627-23765 DfuImpl                          I  Sending Start DFU command (Op Code = 1, Upload Mode = 4)
2025-03-11 16:27:12.196 22627-23219 DfuImpl                          E  Characteristic write error: 3
2025-03-11 16:27:12.198 22627-23765 DfuBaseService                   E  Unable to write Op Code 1 (error 3)
[... cut ...]
2025-03-11 16:27:12.233 22627-23765 DfuBaseService                   I  Cleaning up...
2025-03-11 16:27:12.892 22627-22627 DfuBaseService                   I  DFU service destroyed
```

### How I initialize the DfuServiceInitiator ### 

```
// Kick off DFU
final DfuServiceInitiator starter = new DfuServiceInitiator(targetedBleAddress)
				.setDeviceName(devName)
				.setNumberOfRetries(3)
				.setForceDfu(true)
				.setPacketsReceiptNotificationsEnabled(true)
				.setPacketsReceiptNotificationsValue(5)
				.setMbrSize(4096)
				.setForeground(false)
				.setKeepBond(true);
```

philips77, 2025-03-12:
Hello,
This seems to be an issue with cached services or your DFU bootloader.
The first part looks correct. The library connects to your device in "app mode". One thing that I would point your attention to is the 
```
2025-03-11 16:27:00.931 22627-23535 DfuBaseService                   I  Service Changed indication received
```
line. The Service Change indication should be sent to a bonded peer whenever the attribute table (list of services) changes. I would expect it to be sent after switching to the "bootloader mode" instead. I assume your app is using the services on the device in "app mode" on each connection, so these are the "normal" ones. SC indication should indicate change - that is when jumped to "DFU mode" and again after DFU completed in "app mode".

Anyway, the library sends a command to switch to the bootloader mode, the device restarts and phone reconnects.
Now weird things happen. When reading Version char in bootloader mode, the expected value would be "0.1", not "20.2". This indicates, that the library is reading a value from the same handle number as before (where "0.6" was read), but this time there's something else there, so accidentally, the value is parsed as "20.2". Later, when it tries to write an Op Code to the Control Point characteristic, it gets error 3:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b0d73c655ea336b3cce0265614aac4d7d082e66f/lib/dfu/src/main/java/no/nordicsemi/android/error/GattError.java#L94-L95
Seems again, that the attribute under this handle number is something else, than the library expects.

---
Perhaps I can describe what a handle numbers are. During service discovery, a client reads list of attributes of the connected peer. Attributes are services, characteristics and descriptors. Each characteristic, characteristic's value, a descriptor and its value gets assigned a 2-byte "handle number". These are later used during GATT communication, when you want to read/write a value to a characteristic. In the Andorid API you say "read soem characteristic value", which Android translates to that handle number.

When the list of services changes on the peer, it should notify the client about that fact using Service Change indication. This is especially true for bonded devices, for which the client is free to use cache and never repeat serivce discovery process, unless this indication is received. In your case, seems like the services have changed, but the client (a phone with DFU library) doesn't know about it, so it is using the previous handle numbers. On the DFU target the same handle numbers now point to a different set of attributes, leading to errors.
___
Could you try doing DFU using nRF Connect for Android, or nRF DFU app, both of which are on Play Store?

tudang88, 2025-03-12:
@philips77 
I tried nRF DFU app and the result was the same.
* Succeed with Google Pixel Pro
* Failed with Samsung Galaxy S23
@Rem193 
I finally, ended up the issue, by escalating to firmware development time and ask them to increase the Advertising interval. refer to https://github.com/NordicSemiconductor/Android-DFU-Library/issues/452#issuecomment-2459811944
It's still on-going.

philips77, 2025-03-13:
When you get this versiojn = 20.2, could you read the Version characteristic in nRF Connect using the arrow pointing downm then tap "Refresh services" and read the Version charactersitic again?

Rem193, 2025-03-13:
Maybe an interesting thing:

On nRF connect app after the refresh I tried a DFU, it failed, then I read again the same characteristic (without any refresh) and I got the error message:
"Incorrect data length (2 bytes expected): (0x) 12-14-00-19-2A

philips77, 2025-03-13:
Yes, this seems like caching issue. Check why the device doesn't send Service Change indication when you switch to bootloader mode.

Rem193, 2025-10-01:
When I retrieve version 20.5, the DFU definitely won’t work, so I had the idea to add a call to refreshDeviceCache inside discoverServices() before calling gatt.discoverServices(). I no longer get this incorrect version.

dinesharjani, 2026-08-21:
@rickymohk thanks for the added clarification. Filip is away on much-needed vacation; we'll try to have a look at your submitted PR in ten days.