nordicsemi/Android-DFU-Library #377: Update speed is much slower than iOS

thinhnd-nal, 2023-02-11:
### DFU Bootloader version:
 - nRF5 SDK version: Unknown
 - Are you using bonding: No
 - DFU Library version: 1.11.1

### Device information:
 - Device: Samsung Note 10+
 - OS: Android 12

### Your question
Client complains about my Android app updating firmware more slowly than iOS. With a ~422KB dfu (.zip) file, iOS takes about 30 seconds to complete, while Android takes ~3 minutes, iOS average avgSpeed is about 6Kb/s, while Android's is only about ~1.7Kb/s. Please help me, here is my code (after I wrote the command to launch DFU mode and scan the DFU device)

``` kotlin
DfuServiceListenerHelper.unregisterProgressListener(context, dfuProgressListener)
DfuServiceListenerHelper.registerProgressListener(context, dfuProgressListener, result.device.address)
DfuServiceInitiator(result.device.address)
    .setForeground(false)
    .setDisableNotification(true)
    .setZip(path)
    .setCustomUuidsForSecureDfu(
        Constant.BootloaderService.SERVICE_UUID,
        Constant.BootloaderService.Char.BOOTLOADER_CONTROL_POINT,
        Constant.BootloaderService.Char.BOOTLOADER_PACKET,
    )
    .start(context, CustomDfuService::class.java)
```

### Logs

```
10:56:28.300  D  [DFU Status] Connecting
10:56:28.342  I  DFU service created. Version: 1.11.1
10:56:28.350  W  Foreground service disabled. Android Oreo or newer may kill a background service few moments after user closes the application.
                 Consider enabling foreground service using DfuServiceInitiator#setForeground(boolean)
10:56:28.395  I  Connecting to the device...
10:56:28.517  I  Connected to GATT server
10:56:28.525  I  Action received: android.bluetooth.device.action.ACL_CONNECTED
10:56:28.541  I  Attempting to start service discovery... succeed
10:56:29.387  I  Services discovered
10:56:29.393  W  DFU target does not support (SD/BL)+App update, splitting into 2 parts
10:56:29.393  W  Secure DFU bootloader found
10:56:29.395  D  [DFU Status] Connected
10:56:29.400  D  [DFU Status] Starting
10:56:30.396  I  Requesting MTU = 517
10:56:30.517  I  MTU changed to: 131
10:56:30.519  I  Enabling notifications...
10:56:31.766  I  Setting object to Command (Op Code = 6, Type = 1)
10:56:31.841  I  Command object info received (Max size = 512, Offset = 0, CRC = 00000000)
10:56:31.842  I  Sending the number of packets before notifications (Op Code = 2, Value = 0)
10:56:31.936  I  Creating Init packet object (Op Code = 1, Type = 1, Size = 152)
10:56:32.032  I  Sending 152 bytes of init packet...
10:56:32.032  I  Sending init packet (Value = 12-95-01-0A-4F-08-01-12-4B-08-83-01-10-34-1A-04-A4-02-B7-01-20-03-28-80-8F-09-30-B8-F5-01-38-00-42-24-08-03-12-20-61-8B-C0-14-EC-DB-D2-64-65-75-32-66-FD-E7-13-FF-7E-B2-E8-60-40-C0-CB-39-9B-C7-AE-A9-AE-EB-1F-62-48-00-52-04-08-01-12-00-52-04-08-01-12-00-10-00-1A-40-E3-61-26-EA-EA-F2-6C-BD-CB-B2-98-2A-2C-E8-6D-D1-DC-4E-BC-79-E6-8C-07-93-76-97-C9-04-B2-2A-FD-3C-25-31-6C-A3-EE-34-D4-10)
10:56:32.039  I  Sending init packet (Value = 6C-69-64-D0-0B-E3-57-2E-4E-4B-07-9C-FD-B8-06-C7-C1-B3-6A-16-B0-33-F4-2C)
10:56:32.046  I  Sending Calculate Checksum command (Op Code = 3)
10:56:32.128  I  Checksum received (Offset = 152, CRC = 491C3155)
10:56:32.128  I  Executing init packet (Op Code = 4)
10:56:32.323  W  Sending SD+BL failed. Trying to send App only
10:56:32.323  I  Setting object to Command (Op Code = 6, Type = 1)
10:56:32.423  I  Command object info received (Max size = 512, Offset = 0, CRC = 00000000)
10:56:32.423  I  Sending the number of packets before notifications (Op Code = 2, Value = 0)
10:56:32.517  I  Creating Init packet object (Op Code = 1, Type = 1, Size = 144)
10:56:32.617  I  Sending 144 bytes of init packet...
10:56:32.617  I  Sending init packet (Value = 12-8D-01-0A-47-08-01-12-43-08-86-39-10-34-1A-04-A4-02-B7-01-20-00-28-00-30-00-38-C4-CE-0E-42-24-08-03-12-20-EA-CC-81-17-54-82-16-22-7E-EB-13-5F-C8-AB-A9-0F-B7-A0-BB-2A-4A-C4-84-95-ED-F4-4A-05-8E-0C-8A-04-48-00-52-04-08-01-12-00-10-00-1A-40-C3-1F-D8-26-9C-80-CA-97-53-74-71-9D-37-DE-D3-B8-1A-3F-84-97-D7-F0-BF-21-CF-03-22-A4-53-DD-D7-E3-31-A3-6C-F8-D4-01-00-F7-B3-31-DF-28-5A-8A-CC-83)
10:56:32.622  I  Sending init packet (Value = 00-38-D0-C1-AF-D4-BF-44-5F-BE-E3-24-3E-DC-95-65)
10:56:32.627  I  Sending Calculate Checksum command (Op Code = 3)
10:56:32.713  I  Checksum received (Offset = 144, CRC = 3F1F2E25)
10:56:32.713  I  Executing init packet (Op Code = 4)
10:56:32.907  I  Setting object to Data (Op Code = 6, Type = 2)
10:56:33.005  I  Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
10:56:33.006  I  Creating Data object (Op Code = 1, Type = 2, Size = 4096) (1/59)
10:56:33.007  D  [DFU Status] Started
10:56:33.012  D  [DFU Status] ProgressChanged(percent=0, avgSpeed=0.0)
10:56:33.204  I  Uploading firmware...
10:56:33.279  D  [DFU Status] ProgressChanged(percent=1, avgSpeed=8.941176)
10:56:33.320  I  Sending Calculate Checksum command (Op Code = 3)
10:56:34.131  I  Checksum received (Offset = 4096, CRC = 3B6B0A55)
10:56:34.132  I  Executing data object (Op Code = 4)
10:56:34.228  I  Creating Data object (Op Code = 1, Type = 2, Size = 4096) (2/59)
10:56:34.373  I  Uploading firmware...
10:56:34.397  D  [DFU Status] ProgressChanged(percent=2, avgSpeed=3.496765)
10:56:34.459  D  [DFU Status] ProgressChanged(percent=3, avgSpeed=5.0247936)
10:56:34.483  I  Sending Calculate Checksum command (Op Code = 3)
10:56:37.168  I  Checksum received (Offset = 8192, CRC = 444D6DEB)
10:56:37.168  I  Executing data object (Op Code = 4)
10:56:37.243  I  Creating Data object (Op Code = 1, Type = 2, Size = 4096) (3/59)
10:56:37.313  I  Uploading firmware...
10:56:37.341  D  [DFU Status] ProgressChanged(percent=4, avgSpeed=2.2145329)
10:56:37.384  D  [DFU Status] ProgressChanged(percent=5, avgSpeed=2.7489147)
10:56:37.387  I  Sending Calculate Checksum command (Op Code = 3)
10:56:39.644  I  Checksum received (Offset = 12288, CRC = 2002F576)
10:56:39.644  I  Executing data object (Op Code = 4)
10:56:39.717  I  Creating Data object (Op Code = 1, Type = 2, Size = 4096) (4/59)
10:56:39.792  I  Uploading firmware...
10:56:39.852  D  [DFU Status] ProgressChanged(percent=6, avgSpeed=2.1130753)
...
```

thinhnd-nal, 2023-02-13:
I found that the upload speed dropped suddenly after the `onConnectionUpdated()` call, it was because the Connection Priority was changed for some reason, so I scheduled to set `bluetoothGatt.requestConnectionPriority(BluetoothGatt.CONNECTION_PRIORITY_HIGH)` every time every 3s, so upload speed has increased to ~6kb/s, almost equal to the speed of iOS, but this action sometimes causes GATT_ERROR error while uploading

philips77, 2023-02-13:
You may also set the prepare object delay to 0 to increase the speed  you'll find a method in the initializer.

philips77, 2023-02-13:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/39cd89639ec5080e298a3f5a46355b19fbcce387/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L216-L219

philips77, 2023-02-13:
Try different settings in the nRF DFU, available in play store. Then apply to your app.

thinhnd-nal, 2023-02-13:
I have tried many different settings, but still not much difference

philips77, 2023-10-05:
Hello, did you manage to improve the upload speed?
From what I can tell, this may depend on the device you're using. In my tests we get around 6-7 kB/s on Android. Did you try a different phone? Is it Android which requests long connection interval?

thinhnd-nal, 2023-10-05:
Yes, it depends on the device (maybe bluetooh hardware...), my client uses Umidigi A7 Pro, it's very slow

LukeAveil, 2024-01-26:
Same here. IOS is happening in about 30s, Android 3-4mins. This only started happening since upgrading to 2.3.0. 

Interestingly only seems to happen on Samsung devices, and so far only about 20% of Samsung devices have this problem.

philips77, 2024-01-26:
Hi,
I think this is related to an issue which we found on Samsung A8 Tab, which fails MTU negotiations. During negotiations it claims TX MTU equal to 23, but then sends longer packets, which are correctly rejected by the peripheral.
Have a look at #408.

In the change we're comparing "Build.HARDWARE" to "ums512_25c10". To bypass the check, set MTU to 516 instead of default 517 using https://github.com/NordicSemiconductor/Android-DFU-Library/blob/ee04574dcc438bf067505327eebb37018ee908b8/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L399-L403
Perhaps there's a better condition to be checked, as there may be more devices with that HARDWARE name, which are incorrectly included.

philips77, 2024-01-26:
Note, that the log in the first comment was created before PR 408 was implemented and DFU is requested as normal:
```
10:56:30.396  I  Requesting MTU = 517
10:56:30.517  I  MTU changed to: 131
```
On Samsung A8 Tab the issue manifest with an error after sending just a first packet longer than negotiated 23 bytes.

dinesharjani, 2026-08-20:
Marking as resolved since the issue was pinpointed to Samsung MTU negotiation a while ago.

closed 2026-08-20 by dinesharjani