nordicsemi/Android-DFU-Library #380: error: 133 newState: 0

hjjice, 2023-03-02:
### Information
             I used #385 and it still appears：

       ERROR:
                       Connection state change error: 133 newState: 0
                       Device not reachable. Check if the device with address D6:12:11:00:01:08 is in range, is advertising and is connectable

Brontes, 2023-05-11:
I had similar issue...

My device changes MAC when it goes to DFU mode.

If I wanted to start update immediately after I put device into DFU it was unsuccessful. I received:

"
Connection state change error: 133 newState: 0
Device not reachable. Check if the device with address xxx is in range, is advertising and is connectable
"

I found out that I have to wait the device to start advertising with new MAC. After that I could start update without problems.

philips77, 2023-10-05:
There's an API to delay reconnection to the bootloader.
Have a look at:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/f667c852970e273c86c9c52cef0f3e1663afcd19/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L291-L314

closed 2023-10-05 by philips77