nordicsemi/Android-DFU-Library #384: DfuBaseService-Connection state change error: 255

hjjice, 2023-03-06:
Hello, I have encountered the following problem in DFU library 2.3.0, and I encountered it when testing with Mi 11 and Redmi note9pro：

2023-03-06 15:48:19.907  3293-4740  DfuBaseService          com.xb.ota                      E  Connection state change error: 255 newState: 0
2023-03-06 15:48:19.907  3293-4734  DfuBaseService          com.xb.ota                      E  An error occurred while connecting to the device: 255

hjjice, 2023-03-08:
I found that when the 255 error code appears, it is a gatt connection, in the following code：

private final BluetoothGattCallback mGattCallback = new BluetoothGattCallback() {
		@Override
		public void onConnectionStateChange(final BluetoothGatt gatt, final int status, final int newState) {
			// Check whether an error occurred
			if (status == BluetoothGatt.GATT_SUCCESS) {

Cannot enter BluetoothGatt.GATT_SUCCESS at this time. status is 255, newstate is 0。

But at this time, it is possible to connect with the NRF CONNECT APP，So whether the logic of connecting this piece of DFU library and NRFconnect APP is not the same, what is the specific difference, is there a mechanism for reconnecting or retrying in NRFconnect APP?   Not all devices will appear here, but it will appear when the mobile phone uses a Xiaomi or Redmi mobile phone to connect to this device with the DFU library. Can you do some optimization on the gatt connection?

hjjice, 2023-03-08:
Sometimes try to try, you can connect，

philips77, 2023-10-03:
Try applying delays following this hint: https://github.com/NordicSemiconductor/Android-DFU-Library/issues/387#issuecomment-1744586875

Also, the last PR is supposed to fix Redmi scanning issue, if that's also your case.

closed 2023-10-03 by philips77