nordicsemi/Android-DFU-Library #386: Crashing The App after starting DFU on android 12 or above

sahilshirsat1, 2023-03-09:
FATAL EXCEPTION: IntentService[DfuBaseService]

E/AndroidRuntime(12034): java.lang.IllegalArgumentException: com.atollsolutions.atollbleapp: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
E/AndroidRuntime(12034): Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
E/AndroidRuntime(12034): 	at android.app.PendingIntent.checkFlags(PendingIntent.java:401)
E/AndroidRuntime(12034): 	at android.app.PendingIntent.getActivityAsUser(PendingIntent.java:484)
E/AndroidRuntime(12034): 	at android.app.PendingIntent.getActivity(PendingIntent.java:470)
E/AndroidRuntime(12034): 	at android.app.PendingIntent.getActivity(PendingIntent.java:434)
E/AndroidRuntime(12034): 	at no.nordicsemi.android.dfu.DfuBaseService.startForeground(DfuBaseService.java:1862)
E/AndroidRuntime(12034): 	at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1142)
E/AndroidRuntime(12034): 	at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:78)
E/AndroidRuntime(12034): 	at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime(12034): 	at android.os.Looper.loopOnce(Looper.java:211)
E/AndroidRuntime(12034): 	at android.os.Looper.loop(Looper.java:300)
E/AndroidRuntime(12034): 	at android.os.HandlerThread.run(HandlerThread.java:67)

When i am using   implementation 'no.nordicsemi.android:dfu:1.11.0' in my flutter project app is working fine  but crashing on dfu showing this as a error 

But i am using version 1.12.0 or above it is showing the bluetooth permission is missing in manifest even though i have declared permission well according to android 12 . I have gone through new updates still the permission problem is poping up .

I am using Flutter blue plugin with flutter nordic dfu

philips77, 2023-10-02:
Hi,
Please update to the latest version. The `BLUETOOTH_SCAN` and `BLUETOOTH_CONNECT` permissions have to be requested in runtime, not only declared in the manifest.

philips77, 2023-10-02:
I will release an update tomorrow which should fix some more issues.

closed 2023-10-02 by philips77