nordicsemi/Android-DFU-Library #387: Mi phones cannot be upgraded after installation

hjjice, 2023-03-27:
After newly installing the APP on Xiaomi mobile phones, DFU cannot be upgraded, and the error is reported as follows:

2023-03-27 10:49:55.199 17186-17853 DfuBaseService          com.xb.ota                      E  Connection state change error: 255 newState: 0
2023-03-27 10:49:55.199 17186-18421 DfuBaseService          com.xb.ota                      E  An error occurred while connecting to the device: 255

Test environment:  Android  Target version 33,Phone: XiaoMi 11  And Redmi Note 9 Pro,

When the upgrade is not possible, if you use the NRF connect App to connect to the device in DFU mode, you will find that you can upgrade it with your own APP again in the future. May I ask what the reason may be! Other mobile phones will not have this problem, but Xiaomi mobile phones have encountered it.

voctex, 2023-04-12:
# Some mobile phones have failed to upgrade the DFU library
## Introduction
I have conducted OTA upgrade tests on multiple brands of mobile phones. Among them, the upgrade was successful on Xiaomi 8, Huawei Mate30, Google Pixel4, and Samsung S10. However, there was a problem with upgrading on my Xiaomi 10. The Android version is 12, and the DFU library version used is 2.3.0.

#385 

### DFU Bootloader version (please complete the following information):
- Are you using bonding: no
- DFU Library version: 2.3.0

### Device information (please complete the following information):
- Device: Xiaomi 8
- OS: Android 10
- Result: Success

### Device information (please complete the following information):
- Device: Huawei Mate30
- OS: HarmonyOS 3.0
- Result: Success

### Device information (please complete the following information):
- Device: Google Pixel 4
- OS: Android 13
- Result: Success

### Device information (please complete the following information):
- Device: Samsung S10
- OS: Android 12
- Result: Success

### Device information (please complete the following information):
- Device: Xiaomi 10
- OS: Android 12
- Result: Failed

## Logs(DfuBaseService)
Connection state change error: 255 newState: 0
An error occurred while connecting to the device: 255

Brontes, 2023-05-11:
I had similar issue...

My device changes MAC when it goes to DFU mode.

If I wanted to start update immediately after I put device into DFU it was unsuccessful. I received:
"
Connection state change error: 133 newState: 0
Device not reachable. Check if the device with address xxx is in range, is advertising and is connectable
"
or 
"
Connection state change error: 255 newState: 0
An error occurred while connecting to the device: 255
"
(depends on phone... There was also no problems on some phones)

I found out that I have to wait the device to start advertising with new MAC. After that I could start update without problems.

While searching for solution I also found out something interesting. If I put device to DFU via my app I could connect to it via nRF Connect. And after that I didn't have any problems with updating in my app (Even without waiting for device to start advertising with new MAC). That was until I restarted my phone. After that I wasn't able to update device if I didn't wait for it to start advertising with new MAC.

So if anybody will try to debug this I recommend to restart phone first. 

I had issues on Xiaomi 12 Pro and Pixel 4 - both Android 13. There was also some issues (not always) with OnePlus 10 Pro - also Android 13. 
OTOH I never had any issues on Samsung s10e - Android 12.

philips77, 2023-10-03:
> I found out that I have to wait the device to start advertising with new MAC. After that I could start update without problems.

Some devices, depending on the firmware, require some time to shut down the app before restarting in bootloader mode. If that's also your case, you may customize the delay and scan period using `DfuServiceInitiator`:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/98876948baaef05828a7caba22cad2eb9c143b19/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L291-L314

philips77, 2023-10-03:
The issue with scanning for the device in bootloader mode may also be related to #303.

I'm just working no a fix proposed here: https://github.com/NordicSemiconductor/Android-DFU-Library/issues/303#issuecomment-1729199739

philips77, 2023-10-03:
I hope your issue will be closed. If not, feel free to reopen the issue.

closed 2023-10-03 by philips77

QuentinFarizon, 2023-10-03:
@philips77 Could you do a release ?

philips77, 2023-10-03:
Will do tomorrow. I want to close an issue with Samsung Tab A8 before.