nordicsemi/Android-DFU-Library #390: Importing source code showing error

Ashwin-Sqt, 2023-05-23:
### Your question
Importing the source code getting this error . An help will be appreciated

### Logs
No matching variant of no.nordicsemi.android.gradle:plugins:1.4.2 was found. The consumer was configured to find a library for use during runtime, compatible with Java 11, packaged as a jar, and its dependencies declared externally, as well as attribute 'org.gradle.plugin.api-version' with value '8.1.1' 

[image]

Ashwin-Sqt, 2023-05-23:
Is there any sample app to refer ?

rodrigosimoesrosa, 2023-05-30:
I've solved it with the new android studio flamingo version and changing via Jenv the Java version to 17.

philips77, 2023-10-02:
Fixed with #402. Should work on Android Studio Giraffe.

closed 2023-10-02 by philips77