nordicsemi/Android-DFU-Library #391: Crash on Android14(targetSDK=34)

TadashiSatoo, 2023-06-16:
### Information
Hi, I'm Tadashi Sato.

I'm using your library.
Thank you.

I'm changing my application for targetSdk 34.
But my application crashed with DFU library.

### DFU Bootloader version (please complete the following information):
 - DFU Library version: 2.3.0
    > implementation 'no.nordicsemi.android:dfu:2.3.0'

### Device information (please complete the following information):
 - Device: Pixel7 Android14 beta

### Your question
Your application must be change for below.
https://developer.android.com/about/versions/14/behavior-changes-14#runtime-receivers-exported

Could you tell me a modification plan.

Thank you for your support.

### Logs
point:
> at no.nordicsemi.android.dfu.DfuBaseService.onCreate(DfuBaseService.java:1038)

all:
> FATAL EXCEPTION: main
                                                                                                    Process: xxxxxxxx, PID: 9831
                                                                                                    java.lang.RuntimeException: Unable to create service me.qrio.smartlock2.sdk.common.dfu.DfuService: java.lang.SecurityException: xxxxxxx: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
                                                                                                    	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4663)
                                                                                                    	at android.app.ActivityThread.-$$Nest$mhandleCreateService(Unknown Source:0)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2264)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205)
                                                                                                    	at android.os.Looper.loop(Looper.java:294)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8176)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
                                                                                                    Caused by: java.lang.SecurityException: xxxxxxxx: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
                                                                                                    	at android.os.Parcel.createExceptionOrNull(Parcel.java:3057)
                                                                                                    	at android.os.Parcel.createException(Parcel.java:3041)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:3024)
                                                                                                    	at android.os.Parcel.readException(Parcel.java:2966)
                                                                                                    	at android.app.IActivityManager$Stub$Proxy.registerReceiverWithFeature(IActivityManager.java:5668)
                                                                                                    	at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1852)
                                                                                                    	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1792)
                                                                                                    	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1780)
                                                                                                    	at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:755)
                                                                                                    	at no.nordicsemi.android.dfu.DfuBaseService.onCreate(DfuBaseService.java:1038)
                                                                                                    	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4650)
                                                                                                    	at android.app.ActivityThread.-$$Nest$mhandleCreateService(Unknown Source:0) 
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2264) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106) 
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205) 
                                                                                                    	at android.os.Looper.loop(Looper.java:294) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8176) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971) 
                                                                                                    Caused by: android.os.RemoteException: Remote stack trace:
                                                                                                    	at com.android.server.am.ActivityManagerService.registerReceiverWithFeature(ActivityManagerService.java:13895)
                                                                                                    	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:2563)
                                                                                                    	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2708)
                                                                                                    	at android.os.Binder.execTransactInternal(Binder.java:1339)
                                                                                                    	at android.os.Binder.execTransact(Binder.java:1275)

TadashiSatoo, 2023-06-30:
I change code below.
Does not crash for DFU.(targetSDK34)

DfuBaseService.java 
```
if (Build.VERSION.SDK_INT >= 34) {
    registerReceiver(mDfuActionReceiver, actionFilter, RECEIVER_EXPORTED);
} else {
    registerReceiver(mDfuActionReceiver, actionFilter); // Additionally we must register this receiver as a non-local to get broadcasts from the notification actions
}
```

Thank you for confirmation.

Andrey-Kinsa, 2023-09-05:
Hi Nordic team. Are there any plans to include this fix into the next library release?

philips77, 2023-09-08:
Hi, yes, we have it the backlog. Sorry that we keep you waiting.

marcbaldwin, 2023-09-25:
Hi @philips77 would it be helpful if we created a PR for fixing the issue?

marcbaldwin, 2023-10-02:
@philips77 I've made a PR for Android14 support -> https://github.com/NordicSemiconductor/Android-DFU-Library/pull/401

marcbaldwin, 2023-10-02:
@sylwester-zielinski @philips77 could you please look at my PR with some urgency? We're getting a lot of crashes on Android 14 devices now and we're getting some very upset customers. At the very least could you create a snapshot release of the PR so me and others here have a workaround we can build against and deploy. Thanks

philips77, 2023-10-02:
Fixed with #402.

closed 2023-10-02 by philips77

philips77, 2023-10-05:
Released as 2.3.1.