nordicsemi/Android-DFU-Library #392: 1.11.0 not published anymore ?

QuentinFarizon, 2023-06-22:
It was working yesterday, now all our build fail with error :

* What went wrong:
Execution failed for task ':app:mergeReleaseResources'.
> Could not resolve all files for configuration ':app:releaseRuntimeClasspath'.
   > Could not find no.nordicsemi.android:dfu:1.11.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - https://repo.maven.apache.org/maven2/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - https://maven.google.com/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - file:/Users/vagrant/git/platforms/android/app/libs/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - https://www.jitpack.io/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - https://jcenter.bintray.com/no/nordicsemi/android/dfu/1.11.0/dfu-1.11.0.pom
       - file:/Users/vagrant/git/platforms/android/app/libs/dfu-1.11.0.jar
       - file:/Users/vagrant/git/platforms/android/app/libs/dfu.jar
     Required by:
         project :app

philips77, 2023-06-30:
Have a look at released versions:
https://central.sonatype.com/artifact/no.nordicsemi.android/dfu/2.3.0/versions
The version 1.11.0 isn't on maven central for some reason. Update to 1.11.1 or to the latest one.

QuentinFarizon, 2023-07-01:
@philips77 well we're stuck to  < 1.11.1 until this bug is fixed : https://github.com/NordicSemiconductor/Android-DFU-Library/issues/303#issuecomment-938719414
It's really not cool to unpublish versions used by customers in production

philips77, 2023-07-02:
It's not possible to unpublish. I guess the JCenter repo, which was scheduled to be removed finally got removed. For some reason that version didn't end up on Maven Central.
Anyway, you may clone 1.11.0 tag and add to your project as a module dependency.

QuentinFarizon, 2023-09-04:
> Anyway, you may clone 1.11.0 tag and add to your project as a module dependency.

@philips77 do you have more info on how to do that ? I've followed https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/documentation/README.md#integration along with several configuration changes but without success

closed 2023-10-04 by philips77