nordicsemi/Android-DFU-Library #394: `PlatformException` -> Code: 133, DFU FAILED: GATT ERROR

shubhamsinghmutualmobile, 2023-07-27:
Hi @philips77, I am trying to use the [nordic_dfu](https://pub.dev/packages/nordic_dfu) flutter library to perform DFU for my nrf51/52 devices. I understand this might not be the right place to raise such an issue but please know that I already tried raising it on the Flutter library repo (https://github.com/juliansteenbakker/nordic_dfu/issues/100) but I did not getting any response over there hence I'm raising an issue over here in a hope that I get some direction.

I'm using the latest version of the library i.e. `6.0.1` which internally uses version `2.3.0` of this Android library.

Here are the details of the issue:

I am getting this error every time I try to start DFU for my BLE device. Could you please help?

```
I/DfuBaseService(12367): DFU service destroyed
E/flutter (12367): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(133, DFU FAILED: GATT ERROR, Address: C3:CE:FE:CF:F5:E9, Error Type: 1, null)
E/flutter (12367): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (12367): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (12367): <asynchronous suspension>
E/flutter (12367):
```

## Version info
1. Library version: `6.0.1`
2. Flutter version: `3.13.0-0.2.pre`
3. Android Version: `13 (API 33)`
4. Android Kotlin Version: `1.8.22`
5. Android BLE DFU Library Version (used by the flutter library internally): `2.3.0`

### Complete Logs
```
D/BluetoothAdapter(12367): isLeEnabled(): ON
D/BluetoothLeScanner(12367): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/BluetoothAdapter(12367): isLeEnabled(): ON
I/DfuBaseService(12367): DFU service created. Version: 2.3.0
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
D/BluetoothGatt(12367): registerApp() - UUID=832eaa21-d03a-411a-b24d-471893d93208
D/BluetoothGatt(12367): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Connected to GATT server
I/DfuBaseService(12367): Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt(12367): discoverServices() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Attempting to start service discovery... succeed
D/BluetoothGatt(12367): onSearchComplete() = Device=C3:CE:FE:CF:F5:E9 Status=0
I/DfuBaseService(12367): Services discovered
I/DfuImpl (12367): Buttonless service without bond sharing found -> SDK 13 or newer
W/DfuImpl (12367): Device is paired! Use Buttonless DFU with Bond Sharing instead (SDK 14 or newer)
I/DfuImpl (12367): Enabling indications...
D/BluetoothGatt(12367): setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
I/DfuImpl (12367): Sending Enter Bootloader (Op Code = 1)
I/DfuImpl (12367): Response received (Op Code = 1, Status = 1)
I/DfuBaseService(12367): Disconnecting from the device...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Disconnected from GATT server
I/DfuImpl (12367): Device disconnected
W/m.emberbaby.stg(12367): Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (unsupported, reflection, allowed)
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Refreshing result: true
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuImpl (12367): Restarting to bootloader mode
D/BluetoothAdapter(12367): isLeEnabled(): ON
D/BluetoothLeScanner(12367): onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/BluetoothAdapter(12367): isLeEnabled(): ON
I/DfuImpl (12367): Scanning for new address finished with: C3:CE:FE:CF:F5:E9
I/DfuBaseService(12367): Starting DFU service in foreground
I/DfuBaseService(12367): Connecting to the device...
D/BluetoothGatt(12367): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(12367): registerApp()
[... 55 lines cut ...]
I/DfuBaseService(12367): Attempt: 3
I/DfuBaseService(12367): Cleaning up...
D/BluetoothGatt(12367): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): close()
D/BluetoothGatt(12367): unregisterApp() - mClientIf=7
I/DfuBaseService(12367): DFU service destroyed
E/flutter (12367): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(133, DFU FAILED: GATT ERROR, Address: C3:CE:FE:CF:F5:E9, Error Type: 1, null)
E/flutter (12367): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (12367): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (12367): <asynchronous suspension>
E/flutter (12367): 
D/BluetoothGatt(12367): refresh() - device: C3:CE:FE:CF:F5:E9
D/clearGattCache(12367): CLEAR GATT CACHE: true
D/BluetoothGatt(12367): onClientConnectionState() - status=0 clientIf=6 device=C3:CE:FE:CF:F5:E9
D/FlutterBluePlugin(12367): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt(12367): discoverServices() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(12367): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(12367): onSearchComplete() = Device=C3:CE:FE:CF:F5:E9 Status=0
D/FlutterBluePlugin(12367): [onServicesDiscovered] count: 5 status: 0
```

## Additional info
In the logs, it shows that `Device not reachable. Check if the device with address C3:CE:FE:CF:F5:E9 is in range, is advertising and is connectable` but I can check in the nRFConnect app that while the device is in bootloader state, the MAC address is incremented by 1 i.e. it is `C3:CE:FE:CF:F5:EA` now and the library is not able to search for it on its own, it's still searching for the non-bootloader MAC address.

Any ideas about this? What can we do here? Should the library automatically search for a +1 MAC address?

Erko256, 2023-07-27:
Same here on Android 13, version 2.3.0, but I'm using it under React Native .

philips77, 2023-07-28:
Hello,

In the log I found the following:
```
I/DfuImpl (12367): Buttonless service without bond sharing found -> SDK 13 or newer
W/DfuImpl (12367): Device is paired! Use Buttonless DFU with Bond Sharing instead (SDK 14 or newer)
```
What nRF5 SDK version is the fw on your device based on?
There was no nRF5 SDK v 13 for nRF51 devices. As far as I remember version 12.1 was the last one and it didn't have Buttonless Service, but similar called "Experimental Buttonless Service" with notification instead of indication. Have you ported it manually?
If you're using bonding, why aren't you using Buttonless DFU with Bond Sharing instead?

I also found
```
I/DfuImpl (12367): Scanning for new address finished with: C3:CE:FE:CF:F5:E9
```
which means the service tries to scan for ...EA, but fails and defaults the the old address. Does your device require some additional time to restart? In some cases implementations require time to shut down and restart. This can be customized with these two methods:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b37200033f5d3fe412994e05dd20dd57e14acfe5/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L291-L314
I don't know if they are implemented in your wrappers, as they were added recently.
If your device restarts to bootloader mode immediately that should not be required. 

Do you request for Fine Location Permission (Android 6-11), or Bluetooth Scan permission (Android 12+), or do you have `never_for_location` flag  in Android manifest for that permission? If not, the scanner may not get any results even if your device is advertising. The DFU library won't do that for you.

shubhamsinghmutualmobile, 2023-07-28:
Thanks for the response @philips77 
I do ask for all appropriate permissions, and the device does go into the Bootloader mode instantly.

As for this:
> If you're using bonding, why aren't you using Buttonless DFU with Bond Sharing instead?

I may either need to find a way to contact @juliansteenbakker as he is the one who is working on the Flutter wrapper and request him to ask for this feature, otherwise, I'll have to try to implement it on my own.

Also, I am easily able to perform DFU using the `nRF DFU` app without any hiccups. Does that app use the `Experimental Buttonless Service` by default (or maybe as a fallback)?

philips77, 2023-07-29:
It does enabled use of this service, yes. But looking at your logs you don't need it, as you're using Buttonless service from SDK 13+.

If the nRF DFU app doesn't trigger automatic switch to bootloader, perhaps the issue it related to how your device starts advertising in that mode. Your address ends with E9 and bootloader with EA. In decimal they may be negative, so I can suspect a major issue in the library. Could you change the Mac address to end with 00 and try?

shubhamsinghmutualmobile, 2023-07-29:
Just to confirm, if until now I was running `_firmwareUpdater.startDfu("C3:CE:FE:CF:F5:E9" ...)`, then now I should try running this: `_firmwareUpdater.startDfu("C3:CE:FE:CF:F5:00" ...)`?

philips77, 2023-07-29:
You also need to modify your firmware so that it actually advertise with these MAC addresses.

shubhamsinghmutualmobile, 2023-07-29:
I'm not sure if I'll be able to contact the firmware team and ask them to do it since this is an internal project/experimentation 😅 Extremely sorry about that.

Is there any other thing that I can do to help resolve this issue?

What I can tell you is that they already do have a working app that I'm maintaining but that app is using a legacy version of the library. The details of the version they've been using are:
```
Version: 1.0.4
HEAD checksum: 9cd78d010a38cf13743d9c9160d7e2336c8f9bd4
```

This is why I've been experimenting with some small POCs that use the latest versions of all dependencies and if I'm able to show them that we're able to get DFU to work with their BLE devices with the latest dependency versions, I'll surely be able to convince them to upgrade their existing codebase as well.

philips77, 2023-07-29:
I double checked the code responsible for calculating incremented address:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b37200033f5d3fe412994e05dd20dd57e14acfe5/lib/dfu/src/main/java/no/nordicsemi/android/dfu/internal/scanner/BootloaderScannerFactory.java#L44-L49
and it's correct. For suffix E9 it correctly returns EA, so that was a false lead.

After the library triggers jump to bootloader mode can you scan the device using nRF Connect app?

shubhamsinghmutualmobile, 2023-07-29:
Yes, I am able to see the device in the nRF Connect App with the new (incremented) MAC address while it's in bootloader mode ✅

philips77, 2023-07-29:
And are you able to do DFU using nRF Connect with automatic jump to bootloader?

shubhamsinghmutualmobile, 2023-07-29:
No, I'm not able to do that. It's stuck in the `Connecting...` state while the device is in bootloader mode.

philips77, 2023-07-29:
Could you share logs from nRF Connect?

philips77, 2023-07-29:
> Seems like it did complete the update after a while, because I can see the device automatically got out of the bootloader mode and restarted, but it didn't show any progress.

During the update it switches to another tab, but you may switch back to the original device's tab and progress should be there.

shubhamsinghmutualmobile, 2023-07-29:
> Could you share logs from nRF Connect?

```
nRF Connect, 2023-07-29
Ember Baby Bottle (C3:CE:FE:CF:F5:E9)
V	23:50:45.908	Connecting to C3:CE:FE:CF:F5:E9...
D	23:50:45.908	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	23:50:45.994	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	23:50:45.994	Connected to C3:CE:FE:CF:F5:E9
D	23:50:45.998	wait(1600ms)
D	23:50:46.026	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	23:50:47.599	Discovering services...
D	23:50:47.599	gatt.discoverServices()
D	23:50:47.605	[Callback] Services discovered with status: 0
I	23:50:47.605	Services discovered
V	23:50:47.614	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
Secure DFU Service (0xFE59)
- Buttonless DFU [I W] (8ec90003-f315-4f60-9fb8-838830daea50)
   Client Characteristic Configuration (0x2902)
Unknown Service (fc5421a0-236c-4c94-8fa9-944a3e5353fa)
Unknown Service (fc5421a3-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [R] (fc540201-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [N R] (fc540202-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (fc540203-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [N R] (fc540204-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc540205-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc540206-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (fc540207-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [N R W] (fc540208-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R] (fc540213-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc5402db-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
D	23:50:47.615	gatt.setCharacteristicNotification(fc540202-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.619	gatt.setCharacteristicNotification(fc540204-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.625	gatt.setCharacteristicNotification(fc540205-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.633	gatt.setCharacteristicNotification(fc540206-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.638	gatt.setCharacteristicNotification(fc540208-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.640	gatt.setCharacteristicNotification(fc540213-236c-4c94-8fa9-944a3e5353fa, true)
D	23:50:47.641	gatt.setCharacteristicNotification(fc5402db-236c-4c94-8fa9-944a3e5353fa, true)
I	23:50:51.551	Connection parameters updated (interval: 195.0ms, latency: 0, timeout: 4000ms)
V	23:51:06.276	[DFU] DFU service started
V	23:51:06.276	[DFU] Opening file...
I	23:51:06.311	[DFU] Firmware file opened successfully
V	23:51:06.311	[DFU] Connecting to DFU target...
D	23:51:06.311	[DFU] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferredPhy = LE_1M | LE_2M)
I	23:51:06.327	[DFU] Connected to C3:CE:FE:CF:F5:E9
D	23:51:06.327	[DFU] wait(1600)
V	23:51:07.916	[DFU] Discovering services...
D	23:51:07.916	[DFU] gatt.discoverServices()
I	23:51:07.921	[DFU] Services discovered
D	23:51:07.943	[DFU] wait(1000)
W	23:51:08.967	[DFU] Application with buttonless update found
V	23:51:08.967	[DFU] Jumping to the DFU Bootloader...
V	23:51:08.967	[DFU] Enabling indications for 8ec90003-f315-4f60-9fb8-838830daea50
D	23:51:08.967	[DFU] gatt.setCharacteristicNotification(8ec90003-f315-4f60-9fb8-838830daea50, true)
D	23:51:08.967	[DFU] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x02-00)
I	23:51:09.295	[DFU] Data written to descr.8ec90003-f315-4f60-9fb8-838830daea50, value (0x): 02-00
V	23:51:09.295	[DFU] Notifications enabled for 8ec90003-f315-4f60-9fb8-838830daea50
A	23:51:09.296	[DFU] Indications enabled
[... 90 lines cut ...]
- Unknown Characteristic [N R] (fc540202-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (fc540203-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [N R] (fc540204-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc540205-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc540206-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R W] (fc540207-236c-4c94-8fa9-944a3e5353fa)
- Unknown Characteristic [N R W] (fc540208-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R] (fc540213-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W] (fc5402db-236c-4c94-8fa9-944a3e5353fa)
   Client Characteristic Configuration (0x2902)
D	23:53:13.302	gatt.setCharacteristicNotification(fc540202-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.305	gatt.setCharacteristicNotification(fc540204-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.308	gatt.setCharacteristicNotification(fc540205-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.310	gatt.setCharacteristicNotification(fc540206-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.311	gatt.setCharacteristicNotification(fc540208-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.313	gatt.setCharacteristicNotification(fc540213-236c-4c94-8fa9-944a3e5353fa, true)
D	23:53:13.314	gatt.setCharacteristicNotification(fc5402db-236c-4c94-8fa9-944a3e5353fa, true)
I	23:53:17.409	Connection parameters updated (interval: 195.0ms, latency: 0, timeout: 4000ms)
```

philips77, 2023-07-29:
From the log it looks like nRF Connect also was not able to find the device in bootloader mode. Can you share a screenshot with how it advertises? Expanded row from the Scanner tab.

shubhamsinghmutualmobile, 2023-07-29:
> From the log it looks like nRF Connect also was not able to find the device in bootloader mode. Can you share a screenshot with how it advertises? Expanded row from the Scanner tab.

Sure, here it is in the bootloader mode:

[image]

shubhamsinghmutualmobile, 2023-07-29:
> From the log it looks like nRF Connect also was not able to find the device in bootloader mode. Can you share a screenshot with how it advertises? Expanded row from the Scanner tab.

Sure, here it is in the bootloader mode:

[image]

Do you need an image from the normal mode as well?

philips77, 2023-07-30:
Ok, looking at the logs again I found this:
```
V	23:51:10.325	[DFU] Writing to characteristic 8ec90003-f315-4f60-9fb8-838830daea50
[...]
A	23:51:10.661	[DFU] Enter bootloader sent (Op Code = 1)
I	23:51:10.662	[DFU] Notification received from 8ec90003-f315-4f60-9fb8-838830daea50, value (0x): 20-01-01
A	23:51:10.663	[DFU] Response received (Op Code = 1, Status = 1)
[...]
V	23:51:10.669	[DFU] Scanning for the DFU Bootloader... (timeout 5000 ms)
I	23:51:10.752	[DFU] DFU Bootloader found with address C3:CE:FE:CF:F5:E9
```
After sending 'jump to bootloader' command the library started scanning and almost 100 ms after that it found the device still advertising with the original device. It did not timeout to the old address, but rather such address was scanned.
Consider using `setRebootTime(long)` to delay scan. Perhaps the device indeed needs some more time to restart?
If you don't have such API in your Flutter or ReactNative wrappers, you'll have to add this functionality to the them.

But before you do that, you may test using nRF DFU app. In Settings you may set the Reboot Time and check whether with 500 ms or 1 sec it works for you.

shubhamsinghmutualmobile, 2023-07-30:
You are correct! In the nRF DFU app, if I set the reboot time to 0ms, DFU doesn't work but it starts working perfectly fine for the duration of 1000ms

Thank you very much @philips77 
You saved a lot of time and effort :)

I'm closing this ticket now. Thank you for your time and effort 🙌

closed 2023-07-30 by shubhamsinghmutualmobile