nordicsemi/Android-DFU-Library #395: FW version failure, Error Type: 3

shubhamsinghmutualmobile, 2023-08-01:
Hi @philips77, apologies for bothering you again, first of all, thank you for the help in https://github.com/NordicSemiconductor/Android-DFU-Library/issues/394, the updates work perfectly fine. 

Now, after finishing that implementation, my app's logic is to recursively check for additional updates, and if an update is found, then re-initiate the DFU process, so in my case, I have 3 pending updates, out of which, 2 recursive updates work absolutely fine, no issues there, but just with the last update, I'm facing an issue where the library throws the `FW version failure` error (more info in logs below). 

I tried going through a few other issues in this repo (like #124, #176, #327, etc) but none of those solutions seem to be viable in my case, that's why creating this issue.

According to your answer [here](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/124#issuecomment-421979757), I can change the scope of the update to either target just BL+SD or just the App, so I tried it too. In my case, the hardware version stays the same, just the application version updates (but still I tried the system components scope as well just in case, logs below).

Adding logs below for the following cases:
1. Without using any scope
2. With Application scope (Constant: 2) only
3. With System Components scope (Constant: 1) only

## Without using any scope
```
I/DfuBaseService(13868): DFU service created. Version: 2.3.0
I/DfuBaseService(13868): Starting DFU service in foreground
I/DfuBaseService(13868): Connecting to the device...
D/BluetoothGatt(13868): connect() - device: C3:CE:FE:CF:F5:E9, auto: false
D/BluetoothGatt(13868): registerApp()
D/BluetoothGatt(13868): registerApp() - UUID=2fd8ea4e-ec2f-4996-8094-8380df2e0237
D/BluetoothGatt(13868): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt(13868): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(13868): Connected to GATT server
I/DfuBaseService(13868): Waiting 1600 ms for a possible Service Changed indication...
D/BluetoothGatt(13868): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=156 latency=0 timeout=400 status=0
D/BluetoothGatt(13868): discoverServices() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(13868): Attempting to start service discovery... succeed
D/BluetoothGatt(13868): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=156 latency=0 timeout=400 status=0
D/BluetoothGatt(13868): onSearchComplete() = Device=C3:CE:FE:CF:F5:E9 Status=0
I/DfuBaseService(13868): Services discovered
I/DfuImpl (13868): Buttonless service without bond sharing found -> SDK 13 or newer
W/DfuImpl (13868): Device is paired! Use Buttonless DFU with Bond Sharing instead (SDK 14 or newer)
I/DfuImpl (13868): Enabling indications...
D/BluetoothGatt(13868): setCharacteristicNotification() - uuid: 8ec90003-f315-4f60-9fb8-838830daea50 enable: true
I/flutter (13868): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (13868): │ #0   ScannedBluetoothDevice.checkForDFU.<anonymous closure> (package:example_app_flutter/features/home/viewmodel/baby_bottle_providers.dart:331:24)
I/flutter (13868): │ #1   NordicDfu.startDfu.<anonymous closure> (package:nordic_dfu/src/nordic_dfu.dart:137:32)
I/flutter (13868): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13868): │ 💡 DFU: onEnablingDfuMode called
I/flutter (13868): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/DfuImpl (13868): Sending Enter Bootloader (Op Code = 1)
I/DfuImpl (13868): Response received (Op Code = 1, Status = 1)
I/flutter (13868): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (13868): │ #0   ScannedBluetoothDevice.checkForDFU.<anonymous closure> (package:example_app_flutter/features/home/viewmodel/baby_bottle_providers.dart:338:24)
I/DfuBaseService(13868): Disconnecting from the device...
D/BluetoothGatt(13868): cancelOpen() - device: C3:CE:FE:CF:F5:E9
I/flutter (13868): │ #1   NordicDfu.startDfu.<anonymous closure> (package:nordic_dfu/src/nordic_dfu.dart:120:36)
I/flutter (13868): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (13868): │ 💡 DFU: onDeviceDisconnecting called
I/flutter (13868): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
D/BluetoothGatt(13868): onClientConnectionState() - status=0 clientIf=7 device=C3:CE:FE:CF:F5:E9
I/DfuBaseService(13868): Disconnected from GATT server
I/DfuImpl (13868): Device disconnected
D/BluetoothGatt(13868): refresh() - device: C3:CE:FE:CF:F5:E9
I/DfuBaseService(13868): Refreshing result: true
I/DfuBaseService(13868): Cleaning up...
D/BluetoothGatt(13868): cancelOpen() - device: C3:CE:FE:CF:F5:E9
D/BluetoothGatt(13868): close()
D/BluetoothGatt(13868): unregisterApp() - mClientIf=7
I/DfuImpl (13868): Restarting to bootloader mode
D/BluetoothGatt(13868): onConnectionUpdated() - Device=C3:CE:FE:CF:F5:E9 interval=6 latency=0 timeout=500 status=0
D/BluetoothAdapter(13868): isLeEnabled(): ON
D/BluetoothLeScanner(13868): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
D/BluetoothAdapter(13868): isLeEnabled(): ON
I/DfuImpl (13868): Scanning for new address finished with: C3:CE:FE:CF:F5:EA
I/DfuBaseService(13868): Starting DFU service in foreground
I/DfuBaseService(13868): Connecting to the device...
D/BluetoothGatt(13868): connect() - device: C3:CE:FE:CF:F5:EA, auto: false
D/BluetoothGatt(13868): registerApp()
[... 114 lines cut ...]
E/DfuBaseService(27251): 	at android.os.Looper.loop(Looper.java:288)
E/DfuBaseService(27251): 	at android.os.HandlerThread.run(HandlerThread.java:67)
I/flutter (27251): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (27251): │ 4098
I/flutter (27251): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (27251): │ #0   ScannedBluetoothDevice.checkForDFU.<anonymous closure> (package:ember_baby_flutter/features/home/viewmodel/baby_bottle_providers.dart:358:24)
I/flutter (27251): │ #1   NordicDfu.startDfu.<anonymous closure> (package:nordic_dfu/src/nordic_dfu.dart:145:22)
I/flutter (27251): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (27251): │ ⛔ DFU: onError called. Error type: 0, Reason: DFU FILE ERROR
I/flutter (27251): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/DfuBaseService(27251): DFU service destroyed
I/flutter (27251): ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (27251): │ #0   BabyBottles.observeDevicePreviews (package:ember_baby_flutter/features/home/viewmodel/baby_bottle_providers.dart:429:14)
I/flutter (27251): │ #1   ScannedBluetoothDevice.checkForDFU.invalidateBottlesProvider (package:ember_baby_flutter/features/home/viewmodel/baby_bottle_providers.dart:269:46)
I/flutter (27251): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (27251): │ 💡 Observing device previews...
I/flutter (27251): └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
E/flutter (27251): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(4098, DFU FAILED: DFU FILE ERROR, Address: C3:CE:FE:CF:F5:E9, Error Type: 0, null)
E/flutter (27251): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (27251): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (27251): <asynchronous suspension>
E/flutter (27251): 
```

In the nordic_dfu library, I can see that the zip file is always set first before setting any other configuration option(s), here's the [code](https://github.com/juliansteenbakker/nordic_dfu/blob/master/android/src/main/kotlin/dev/steenbakker/nordicdfu/NordicDfuPlugin.kt#L154).

philips77, 2023-08-02:
You shouldn't need to use Scopes in this case.

This error seems to be expected. The bootloader verifies the new image based on the init packet. It contains the version of the new image. There is a default behavior, that the new bootlaoder has to have version number higher than the current one (if you're updating SD+BL or BL), and app needs to have at least the same version or higher. If the version number in the init packet is <= or < to the version currently running you'll get this error.

The version number you set in the `nrf util` command when you generate ZIP file.

shubhamsinghmutualmobile, 2023-08-02:
> There is a default behavior

Is there no way to override this behavior?

shubhamsinghmutualmobile, 2023-08-02:
If it is possible and not a deal breaker, would it please be possible to give a simple switch in the library (maybe named `Boolean forceUpdate`) that by default is `false` (thereby not affecting any of the existing users) and on `true`, overrides the behavior mentioned above?

philips77, 2023-08-02:
I used `nrfutil pkg display <zipfile>` to list details of your init packet:
[image]

It contains the bootloader, which means you need to recreate the zip with incremented version number to be able to update your device.

shubhamsinghmutualmobile, 2023-08-02:
Understood. Thanks a lot, @philips77 

Just for my understanding, could you please explain to me what would happen if the version-checking logic was not there? Let's say someone flashes the same update twice, would that surely brick the device? Or there's a chance that the device might not be bricked but since Nordic had to cater to a much larger user base, it had to disable FW downgrade?

shubhamsinghmutualmobile, 2023-08-02:
Understood. Much thanks for the explanation and for your time and effort. Closing this issue now :)

closed 2023-08-02 by shubhamsinghmutualmobile