nordicsemi/Android-DFU-Library #396: Translation of notification

DomiR, 2023-08-08:
Hey, we'd love to change the translation of the key and also would like to use dynamic values here. I can open a PR for this if you tell me how you would like to have this implemented. e.g. overload with name and description?

https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b37200033f5d3fe412994e05dd20dd57e14acfe5/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L919

philips77, 2023-10-02:
Can't you just redefine the text in the app's module? Doesn't it work anymore?

philips77, 2023-10-04:
All notification strings are defined here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/845cf5aadf4091184a6cbba367f67112836d0e1a/lib/dfu/src/main/res/values/strings.xml#L1-L35

You may redefine them in your app's module, just like that:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/845cf5aadf4091184a6cbba367f67112836d0e1a/app/src/main/res/values/strings.xml#L36-L37

You may also provide translations in other languages.

PR #397 (2023-10-05): feat(notification): allow passing of name and description
fix #396
closed 2023-10-04 by philips77