nordicsemi/Android-DFU-Library #400: Batch/Automatic DFU update mode?

gfwilliams, 2023-09-20:
### DFU Bootloader version (please complete the following information):
 - nRF5 SDK version: SDK 12
 - Are you using bonding: no
 - DFU Library version: 2.3

### Device information (please complete the following information):
 - Device: Redmi Note 8 Pro
 - OS: Android 11

### Your question
I have a bunch of devices that I would like to automatically update, but right now I have to manually tap `Device -> Select`, `DfuTarg` then `Start`

Would it be possible to add  a mode (maybe even just accessible from Settings) where as soon as the DFU update completes, it immediately starts scanning for a device with the same name as last time, and then flashes that too, and so on.

While I've developed some Android apps before I'm totally new to Kotlin and I'm struggling to really find my way around the source. Even if I did manage to hack it together I don't think it would be in a state for me to contribute back, so any help with this would be gratefully appreciated!

ZIZI51, 2023-11-10:
> ### DFU Bootloader version (please complete the following information):
>  - nRF5 SDK version: SDK 12
>  - Are you using bonding: no
>  - DFU Library version: 2.3
> 
> ### Device information (please complete the following information):
>  - Device: Redmi Note 8 Pro
>  - OS: Android 11
> 
> ### Your question
> I have a bunch of devices that I would like to automatically update, but right now I have to manually tap `Device -> Select`, `DfuTarg` then `Start` for every device
> 
> Would it be possible to add  a mode (maybe even just accessible from Settings) where as soon as the DFU update completes, it immediately starts scanning for a device with the same name as last time, and then flashes that too, and so on.
> 
> While I've developed some Android apps before I'm totally new to Kotlin and I'm struggling to really find my way around the source. Even if I did manage to hack it together I don't think it would be in a state for me to contribute back, so any help with this would be gratefully appreciated!
> 
>

dinesharjani, 2026-08-21:
Hi @gfwilliams - we're mostly working on fixing bugs & issues regarding nRF5 SDK's DFU library. We appreciate your feature request, but we're not sure we can acommodate at the moment. 

Since you mentioned working on your own solution, you can always just keep your own fork of the library with your own solution and run it for your own use case, and just rebase or update when we release fixes. This way you don't have to submit a PR and you get to keep your solution :)