nordicsemi/Android-DFU-Library #412: [Version Android - 14] - java.lang.SecurityException:  One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts

vasim1422, 2023-11-02:
### Where do you suspect the issue?

Issue related to Android version or specific device

### Version

2.3.1 (Latest)

### Describe the issue

App crashing in **Android 14** due to exception of **java.lang.SecurityException**  One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts. more log attached 

`implementation 'no.nordicsemi.android:dfu:2.3.1'
`

Please provide quick help on this as this issue occurred in our production product in which used your services.

Thanks

### Relevant log output

```text
java.lang.RuntimeException: Unable to create service no.nordicsemi.android.dfu
.DfuBaseService: java.lang.SecurityException: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts 	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4664)
                                                                                                    	at android.app.ActivityThread.-$$Nest$mhandleCreateService(Unknown Source:0)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2264)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:205)
                                                                                                    	at android.os.Looper.loop(Looper.java:294)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8177)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)

Caused by: java.lang.SecurityException: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
```

PavlosTze, 2023-11-03:
We also have the exact same issue. Based on this, the fix should be really simple. We need a fix asap for this as its affecting our production apps.

closed 2023-11-03 by sylwester-zielinski

PavlosTze, 2023-11-03:
@sylwester-zielinski When can we expect a new version of the library to use in our apps?

sylwester-zielinski, 2023-11-03:
I need to release dependency libraries. Maybe today if there were no issues.

sylwester-zielinski, 2023-11-03:
@PavlosTze Version 2.3.2 should be already available.

PavlosTze, 2023-11-03:
@sylwester-zielinski I get the following:
```
   > Could not find no.nordicsemi.android:dfu:2.3.2.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/no/nordicsemi/android/dfu/2.3.2/dfu-2.3.2.pom
       - https://repo.maven.apache.org/maven2/no/nordicsemi/android/dfu/2.3.2/dfu-2.3.2.pom
       - https://api.mapbox.com/downloads/v2/releases/maven/no/nordicsemi/android/dfu/2.3.2/dfu-2.3.2.pom
       - https://jitpack.io/no/nordicsemi/android/dfu/2.3.2/dfu-2.3.2.pom
```

Maybe it's not uploaded yet?

roshanrajaratnam, 2023-11-03:
@PavlosTze it might take a couple of hours till its available via mavenCentral. You can include the library directly in your source code so that you can get going.

PavlosTze, 2023-11-04:
@roshanrajaratnam @sylwester-zielinski  Still not there one day later. We want to include it directly via mavenCentral.

roshanrajaratnam, 2023-11-04:
@PavlosTze just noticed that the staging hasn't completed properly. Give it a an hour or so and it should be available. Sorry for any inconvenience.

vasim1422, 2023-11-06:
@PavlosTze Still not working for Device Android 14 please help one this

philips77, 2023-11-07:
Have you followed [this](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/414#issuecomment-1794417727) and [this](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/414#issuecomment-1794460206) comments?