nordicsemi/Android-DFU-Library #414: Crash on Android 14 because of no type on startForeground

PavlosTze, 2023-11-05:
### Where do you suspect the issue?

Issue related to Android version or specific device

### Version

2.3.1 (Latest)

### Describe the issue

I'm getting this error on the latest version which is `2.3.2`.
```
Fatal Exception: android.app.MissingForegroundServiceTypeException: Starting FGS without a type  callerApp=ProcessRecord{921e68e 8405:com.weatherxm.app/u0a505} targetSDK=34
       at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:53)
       at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:49)
       at android.os.Parcel.readParcelableInternal(Parcel.java:4870)
       at android.os.Parcel.readParcelable(Parcel.java:4852)
       at android.os.Parcel.createExceptionOrNull(Parcel.java:3052)
       at android.os.Parcel.createException(Parcel.java:3041)
       at android.os.Parcel.readException(Parcel.java:3024)
       at android.os.Parcel.readException(Parcel.java:2966)
       at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6761)
       at android.app.Service.startForeground(Service.java:775)
       at no.nordicsemi.android.dfu.DfuBaseService.startForeground(DfuBaseService.java:1910)
       at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1154)
       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:77)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:205)
       at android.os.Looper.loop(Looper.java:294)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```
On `DfuBaseService.java` you need to specify a type on Android 14 on line 1910:
`startForeground(NOTIFICATION_ID, builder.build());`

Source: https://developer.android.com/about/versions/14/changes/fgs-types-required

### Relevant log output

```text
Fatal Exception: android.app.MissingForegroundServiceTypeException: Starting FGS without a type  callerApp=ProcessRecord{921e68e 8405:com.weatherxm.app/u0a505} targetSDK=34
       at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:53)
       at android.app.MissingForegroundServiceTypeException$1.createFromParcel(MissingForegroundServiceTypeException.java:49)
       at android.os.Parcel.readParcelableInternal(Parcel.java:4870)
       at android.os.Parcel.readParcelable(Parcel.java:4852)
       at android.os.Parcel.createExceptionOrNull(Parcel.java:3052)
       at android.os.Parcel.createException(Parcel.java:3041)
       at android.os.Parcel.readException(Parcel.java:3024)
       at android.os.Parcel.readException(Parcel.java:2966)
       at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6761)
       at android.app.Service.startForeground(Service.java:775)
       at no.nordicsemi.android.dfu.DfuBaseService.startForeground(DfuBaseService.java:1910)
       at no.nordicsemi.android.dfu.DfuBaseService.onHandleIntent(DfuBaseService.java:1154)
       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:77)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loopOnce(Looper.java:205)
       at android.os.Looper.loop(Looper.java:294)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

philips77, 2023-11-06:
Are you getting this error in your app, which is using the DFU library?
The nRF DFU app already specifies the service type here:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/34515ab33627a4282079663299a07f2e5290a0f8/profile/main/src/main/AndroidManifest.xml#L43

PavlosTze, 2023-11-06:
Yes @philips77 we get this error on an app using the DFU Library on Android 14 devices.

philips77, 2023-11-06:
If it's your app, you need to fix it on your side. The library doesn't register the service in its Android Manifest, as it doen't know the final type. Have a look at the link I posted above. You'll also need this:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/34515ab33627a4282079663299a07f2e5290a0f8/profile/main/src/main/AndroidManifest.xml#L37

PavlosTze, 2023-11-06:
That worked, thank you @philips77 . I'm closing this issue.