nordicsemi/Android-DFU-Library #428: Problems with binding for xamarin/maui: string with multiple placeholders

tompi, 2023-12-22:
### Version

2.4.1 (Latest)

### Ask the question

Hi, am tagging this as a question, since I dont think this is your bug....

I had some problems binding the latest version, specifically when using produced binding library in a Maui application, I get this compile time error:

```
res/values/values.xml(2): error APT2000: multiple substitutions specified in non-positional format; did you mean to add the formatted="false" attribute?
```

This line seems to be causing the problem:
```xml
<string name="dfu_status_uploading_part">Uploading part %d/%d&#8230;</string>
```

I tried introducing positional format like "Uploading part %1$d/%2$d", but that didnt help.
When I just used "Uploading part %d", it worked...

I see you removed the  formatted=false on October 2 in commit 4a5c18d 
(I assume since formatted=false would not interpolate, so you probably just got the percent signs?)

Anyway, I know this is not your problem, but MAYBE you could change it to just have one substitution?
(just build the x/y in code and substitute for an %s in the xml)

Its a bit dirty, so I totally understand if you wont do it...

andersonvieiragomeslopes, 2024-02-16:
Hi @tompi  

Have you made any progress with this issue? I have the same problem on another SDK and it is also related to android library binding for maui.

Silverdark, 2024-02-16:
> I tried introducing positional format like "Uploading part %1$d/%2$d", but that didnt help.

I tried it myself by build an `aar` file and used it to build a NuGet with bindings. For me this worked so hopefully the PR can solve this.

tompi, 2024-02-17:
Yes, i just unzipped the aar, changed the xml file, rezipped and then bound the library. Cumbersome, but it works...

ksidirop-laerdal, 2024-05-22:
Having this exact same issue for the exact same cause when trying to build MAUI sample-apps. The build fails with the same kind of error:

```
0>values.xml(2): Error APT2000 : multiple substitutions specified in non-positional format; did you mean to add the formatted="false" attribute?.

0>values.xml: Error APT2261 : file failed to compile.
```

PR #435 (2024-05-23): Use positional format specifier for uploading resource
This PR fixes the following issue:

- When creating bindings of the latest version for a MAUI application, the build fails with the following error. By using positional format specifiers for the string resource with two format specifiers, it worked for me.

```
values.xml(2): Error APT2000 : multiple substitutions specified in non-positional format; did you mean to add the formatted="false" attribute?.
values.xml: Error APT2261 : file failed to compile.
```

Fixes #428
closed 2024-05-23 by philips77