nordicsemi/Android-DFU-Library #429: Application stuck after Discovering services.

xiaozhao1995, 2024-01-05:
### Where do you suspect the issue?

Issue in DFU library, e.g. upload stops in the middle

### Version

2.3.x

### Describe the issue

When I do a device upgrade（Dfu SDK 2.3.2），The log stops after printing the ”Discovering services...“  "gatt.discoverServices()"

DfuBaseService#onConnectionStateChange 
Should we do a 100ms delay before discovering service？

philips77, 2024-01-05:
Hello,
What phone are you testing on? We used to have some delays in multiple places, but they were not required on phones that we test on anymore.

xiaozhao1995, 2024-01-08:
It's Redmi 8A

philips77, 2024-10-25:
I'll release 2.7.0 soon. Please check if something has changed.

dinesharjani, 2026-08-20:
Marking as closed since an update was released and no further updates regarding the issue have been made.

closed 2026-08-20 by dinesharjani