nordicsemi/Android-DFU-Library #433: `startForeground` doesn't specify a `foregroundServiceType`

fabiendem, 2024-01-29:
### Where do you suspect the issue?

Other

### Version

2.4.1 (Latest)

### Describe the issue

At this line https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1926

the `startForeground` call to start a Foreground service doesn't specify a `foregroundServiceType` as per the doc at https://developer.android.com/reference/androidx/core/app/ServiceCompat#startForeground(android.app.Service,int,android.app.Notification,int)

Apps which set their `targetSdkVersion` to 34 must now specify the Foreground Service type, as per the doc at https://developer.android.com/about/versions/14/changes/fgs-types-required

See the requirements for `connectedDevice`: https://developer.android.com/about/versions/14/changes/fgs-types-required#connected-device

philips77, 2024-02-06:
You may specify the foreground service type in the Android Manifest file.
> If the foreground service type is not specified in the call, the type defaults to the values defined in the manifest. If you didn't specify the service type in the manifest, the system throws [MissingForegroundServiceTypeException](https://developer.android.com/reference/android/app/MissingForegroundServiceTypeException).

This is what we do in the nRF DFU app:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/main/profile/main/src/main/AndroidManifest.xml

fabiendem, 2024-02-06:
Hey @philips77, thanks, I missed that.

Link: https://developer.android.com/about/versions/14/changes/fgs-types-required#include-fgs-type-runtime

[image]

Closing this!