nordicsemi/Android-DFU-Library #440: Automatic Reconnection not available on Android 14

LucaOnesto, 2024-04-04:
Hi, 

in our mobile application we implemented the automatic reconnection to our devices, so that if a disconnection occurs (e.g. due to a reboot of the system), the system will automatic reconnect. 

With the application running on Android 14, e noticed that if our device is not available for more than around 10 seconds, the following error occurs:  

onClientConnectionState() - status=133 clientIf=6 device=XX:XX:XX:XX:29:42

This does not happen with previous version of Android. 

Tests were done with version 2.2.2 and 2.4.2 of lib-dfu.

philips77, 2024-04-04:
Hi,
But.. what does it have to do with the DFU library? The lib does not hold connection.

Error 133 after 10ish or 30 sec may mean a timeout. This usually happens when you try to reconnect with `autoConnect = false` when the device is not available. But with automatic reconnections there should be no timeout.
I didn't try that on Android 14.

LucaOnesto, 2024-04-05:
Hi philips77,

Thanks for your suggestion. 
I checked the latest version of the ble-library and it works also with Android 14.

closed 2024-04-05 by LucaOnesto