nordicsemi/Android-DFU-Library #441: Android 14 GATT 133 1

MarioTheGreatest, 2024-04-05:
### Where do you suspect the issue?

Issue related to Android version or specific device

### Version

2.4.1 (Latest)

### Describe the issue

[image]

On Android bellow 14 the DFU library uploads the firmware as expected, but on Android 14 for some reason it will make 3 attempts while stuck on onDeviceConnecting and then return GATT Error 133 1. I have tried disableMtuRequest() and setPacketsReceiptNotificationsValue(1) but it doesn't fix the issue.

sumitemizentech, 2024-04-05:
I am having similar issue on android 14, Samsung phone. After 3% it throws an error - GATT INVALID PDU.

SEAbdulbasit, 2024-07-15:
Same issue
`Connection state change error: 133 newState: 0`
` An error occurred while connecting to the device: 133`

SEAbdulbasit, 2024-07-24:
When i pair first and then try to update the firmware then it works fine.

christian-zs, 2024-08-26:
I have the same problem here. I need to apply for the dynamic permission ACCESS_FINE_LOCATION. After applying, it's done.

christian-zs, 2024-08-26:
> ### Where do you suspect the issue?
> Issue related to Android version or specific device
>
> ### Version
> 2.4.1 (Latest)
>
> ### Describe the issue [...]
I have the same problem here. I need to apply for the dynamic permission ACCESS_FINE_LOCATION. After applying, it's done.

philips77, 2024-09-18:
Hello,
Last week we released version 2.6, which should fix scanning in background. Did you try it?

As it appears only in background, I suspect it's related to scanning without a `ScanFilter` (or actually, with an empty one). Recent Android versions began to check if the filter isn't empty. In that case the phone doesn't scan and in result cannot connect to the device advertising with address +1.

Since 2.6 the filters are no longer empty and you may customize them using:
https://github.com/NordicSemiconductor/Android-DFU-Library/blob/4123bdd5f20c5f8a1a51cb61546b4fc132513dec/lib/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1987-L1990

closed 2024-10-22 by philips77