nordicsemi/Android-DFU-Library #442: Error 8(0x8): GATT CONN TIMEOUT

hariavantari, 2024-04-25:
### Where do you suspect the issue?

Issue related to Android version or specific device

### Version

2.4.1 (Latest)

### Describe the issue

We are working on a health-based project that works with a smart ring using Bluetooth. The problem is app and ring are working fine on 95% of Android phones. But 5% of the phones get disconnection problems in the following scenario.

The app sends a command to the ring to fetch data from a characteristic. The ring is receiving commands properly from the app(We can see lights glowing), but after 15 seconds(which is supervisor timeout) ring disconnects from the app. On the same mobiles, the NRF Connect app also has the same problem. This problem mainly occurs on OnePlus phones.

I tried changing the connection interval, slave latency, and supervisor timeout on the ringside. But nothing worked.

Can someone
[image]
 suggest a fix for this whether the fix should happen on the ring side or the app side?

hariavantari, 2024-04-29:
Hi
Can anyone help me to resolve this?

philips77, 2024-10-22:
Hi,
First of all, sorry for the long delay. I just noticed this question.
This seems not to be related to the DFU library, rather Bluetooth on Android in general. Are you using the BLE library from Nordic, or the Android API directly?

It's hard to say what's the root cause of the disconnection without deeper analisys and sniffing the connection. For me it looks like the device stopped responding to the phone for some reason. Usually, it happens when there's a bug on the phone side which doesn't reply to a request from the peripheral.

We will try to buy a OnePlus phone and add it to our library for testing. I can't help you much without it.

philips77, 2024-10-25:
I'm closing, as this isn't related to the DFU library.

closed 2024-10-25 by philips77