nordicsemi/Android-DFU-Library #444: DfuServiceInitiator requires a new method

xiaozhao1995, 2024-06-11:
### Target

Library

### Describe the feature

	/**
	 * Sets the time required by the device to reboot. The library will wait for this time to disconnect
	 * for the device to make sure device has received BLE_GATTC_EVT_HVX
	 *
	 * @param disconnectTime the disconnect time in milliseconds,default 0.
	 * @return the builder
	 */
	public DfuServiceInitiator setDisconnectTimeout(final long disconnectTime) {
		this.disconnectTime = disconnectTime;
		return this;
	}
	
	We found that on some phones, such as Huawei mate30pro, Huawei pura70, if you call （ButtonlessDfuImpl#performDfu）disconnect immediately after receiving 200101, the device may not receive the BLE_GATTC_EVT_HVX event, resulting in an interrupt into boot.

philips77, 2024-10-22:
Hello,
Sorry for teh very long delay. I had a lot on my plate. 

Usually, it's the device that initiates disconnection and the lib calls `disconnect()` so that it doesn't have to wait Supervision Timeout. Wouldn't it be sufficient to just wait ~100 ms before calling this method? What would be the value of `disconnectTime` that you'd intend to use?

Or did you fix your issue already in some other way?

xiaozhao1995, 2024-10-25:
We currently have a delay of 500 milliseconds enabled. I'm not sure if this value is optimal, but it resolves the issues on all devices where problems have occurred.

xiaozhao1995, 2024-10-25:
[image]

philips77, 2024-10-25:
I'll add 500 ms delay in upcoming version 2.7.0.

PR #471 (2024-10-25): Adding a 500 ms delay before disconnecting
This PR fixes #444.
closed 2024-10-25 by philips77