nordicsemi/Android-DFU-Library #454: DFU CHARACTERISTICS NOT FOUND

faisalm70033, 2024-09-04:
### Version

None

### Ask the question

const result = await NordicDFU.startDFU({
      deviceAddress: deviceId,
       deviceName: deviceName,
      filePath: actualFilePath,
    }).then((afterResult)=>{
      console.log("---->>> result", afterResult);
      
     
      
    })
    .catch((error) =>{
        console.log("show errorr ->>>",error);
        
    })
    
    
     when we start erro show 
   DFU State: DEVICE_DISCONNECTING
   DFU State: DFU_FAILED
   show errorr ->>> [Error: DFU CHARACTERISTICS NOT FOUND]

philips77, 2024-09-07:
This library supports Legacy and Secure DFU from Nordic nRF5 SDK.

Make sure your device supports it and the service UUID and characteristic UUIDs are the same. If the protocol is the same, but the UUIDs have been modified, you can use `UuidHelper` to assign the right ones.

If your device does not support any of these, for example uses nRF Connect SDK's McuMgr, use a different library, which is design for that DFU protocol.

closed 2024-09-07 by philips77

faisalm70033, 2024-09-09:
I am detect my device have SMP service and Nordic DFU not support SMP service for DFU update in react native .
how can we do DFU update with SMP service  ? In react native.

philips77, 2024-09-09:
Hi,
I've never tried, but you may try this library: https://github.com/PlayerData/react-native-mcu-manager

faisalm70033, 2024-09-09:
Thanks,
Butt this Library use in expo only but my project create in cli ?

philips77, 2024-09-09:
Sorry, but I have no idea what it means. You may search for another wrapper or write your own. We only provide native implementations and a Flutter wrapper, which we needed for one project.

faisalm70033, 2024-09-09:
I am create a project in react native with BLE Manager scan and connect device work perfectly write and read data perfectly but when i am try to update DFU with the help of Nordic DFU show error characteristic not found when we search on it my device have SMP service which is not work with Nordic DFU .
Then you share another library (library: https://github.com/PlayerData/react-native-mcu-manager)
but this library work with expo because two platform in react native one is expo and other is CLI i am work with CLI that is why i am not work with your shared library any other idea you have please share with me ?