nordicsemi/Android-DFU-Library #459: DFU becomes significantly slower on Android 15

v-nakamuray, 2024-09-25:
### Version

2.5.0 (Latest)

### Ask the question

When performing DFU with the same Android device + nRF Connect on the same target device, the speed becomes significantly slower on Android 15.

The attached image captures the graph of DFU performed on the same device using Pixel 6a + Android 14 or 15, and the transfer speed is clearly different.

The device in use is an nRF52840. However, we have confirmed that the DFU speed decreases on other devices as well.
Both tests used nRF Connect version 4.28.1.

Is it possible to improve the speed by changing the settings in nRF Connect or the usage of the DFU library?

[image]

philips77, 2024-09-25:
Hello,
I have not yet tried the DFU on Android 15. The only difference I can imagine would be different connection parameters.
You may try disabling PRN in _Settings / nRF5 DFU options_ or setting it to some relatively small number. This will make DFU even slower, but perhaps the speed will be more consistent among different devices.

Do you know which nRF5 SDK version you have on nRF52840?

v-nakamuray, 2024-09-26:
Hello,
I tried on Android15 with the following settings..
- PRN(packet receipt notification) OFF
- PRN ON, number of packets = 1
- PRN ON, number of packets = 5
- PRN ON, number of packets = 8
- PRN ON, number of packets = 32

A smaller value for the "number of packets" resulted in a bit more slowness.
Overall, there is not much change.

[image]

> Do you know which nRF5 SDK version you have on nRF52840?

SDK is v15.2.
I am also having the same problem with devices using v17.

philips77, 2024-10-22:
Hi,
I had my phone updated to Android 15 and will start doing throughput tests with a sniffer. I wonder what the reason is.

philips77, 2024-10-22:
Hello,
Let me share you my findings. I was using the nRF DFU app and set _Prepare object delay_ to 0 ms in Settings (default is 400 ms)[^1].

In this post I'll show a graph showing the upload from Pixel 5 with Android 14 (not 15) onto a nRF52840 DK running a Buttonless Sample from nRF5 SDK 15.2. On average I was getting 9.5 kB/s, which is an OK speed. The MTU is 247 (244 bytes of data per write operation). Connection interval is 15 ms.
[image]
This is how an average packet being sent. As you can see - it's sent twice!!! Every packet is sent at least 2 times. The phone tries to fit more than one packet in a single connection interval (there's space for 6 packets), but the target discards it, as it's not ready to save it in the flash. The second packet in a connection interval isn't confirmed, which indicates that it needs to be repeated in the next interval. The actual upload speed is then twice as fast!!! It's just that 50% of the data sent is discarded due to flash implementation. Still, 9.5 kB/s is OKish.

[^1]: This is, actually, a hardcoded value in nRF Connect for Android 4.28.1.

philips77, 2024-10-22:
I tried using Pixel 7 with Android 15 and got similar speed: 9.1 kB/s.

Here's a graph showing the operations in-between sending data objects: CRC calculation, Executing and Creating a new objects:
[image]
As you can see, there's some time spent on this as well. Executing takes 135 ms and in this time the bootloader is copying the received 4096 bytes to the slot. It does it every time it receives an object (4096 bytes).

However, I did not observe any significant upload speed decease between Android 14 and 15. They seem to behave the same way. Perhaps you were using a beta version?

philips77, 2024-10-22:
I'll now test it with bonding enabled.

v-nakamuray, 2024-10-22:
Thank you for your research.

Following the advice of NordicSemiconductor, I tried DFU with the following configuration.

nRF52 DK + nRF5_SDK_17.1.0_ddde560/examples/dfu/secure_dfu_test_images/secure_dfu_test_images/ble/nrf52832/
  sd_s132_bootloader_buttonless_with_setting_page_dfu_secure_ble_debug_without_bonds.hex
  sd_s132_bootloader_buttonless_with_setting_page_dfu_secure_ble_debug_with_bonds.hex

 "without_bonds" -> did not change speed.
 "with_bonds" -> slowed down on Android 15.

philips77, 2024-10-22:
I'm actually having hard time here, as the DK seems not sending a Service Change indication to refresh the service cache. Or I'm missing something. I'll try again to sniff it.

philips77, 2024-10-22:
I've noticed that when bonding, the connection interval gets set to whooping 192 ms and still the phone is able to send just one packet per interval.
[image]
Compare this with the 1st image I sent.

philips77, 2024-10-22:
I'm requesting high connection priority before upload, and it seems to fix the slow upload speed. I also found another issue related with bond.

philips77, 2024-10-22:
Have a look at 466. I believe I fixed the issue. I'll do some more testing on other phones tomorrow.

closed 2024-10-22 by philips77

v-nakamuray, 2024-10-23:
@philips77 
Thank you for your investigation and improvement!
I have used the DFU Library that incorporates the changes in #466
and have confirmed that it improves speed on Android 15 + (Pixel6a, 7).

philips77, 2024-10-23:
I'll go through other reported issues and release it as 2.7.0, perhaps later today.

philips77, 2024-10-24:
I did some research and got answers, that you may be interested in.

Android 15 started reading _Peripheral Preferred Connection Parameters_ (PPCP) characteristic after bonding and using this value for requesting connection parameters in following connections.
However, often an app may have very different preferred parameters than the DFU bootloader. For example, a HRM or Buttonless sample from nRF5 SDK returns 400-650 ms or 100-200 ms. 
In the meantime, the DFU Bootloader returns 15-15ms from PPCP.
The issue is that Android reads it just once, after bonding. So, when it switches to DFU bootloader mode, it immediately requests whatever value the app preferred, even when it's asked to clear the cache (Service Changed indication).

In the PR I simply request HIGH connection priority (that is connection interval 15 ms), but the change requires ~6-8 packets before it's applied. With a long interval in force this will take few seconds. After receiving SD indication Android will initiate a fresh service discovery, which initially will be performed using the old, long interval.

What's worse, is that Android 15 seems to perform service discovery on each connection, even when bonded. This is what I observe when there's no _Database Hash_ characteristic in _Generic Access_ service. After switching to DFU bootloader mode the SC indication is sent, so the discovery is expected anyway. For the discovery Android switches to 7.5 ms interval. But, again, teh new params are applied after few packets, so with an interval of 645 ms takes few seconds. After the discovery is complete Android again requests this long "preferred" interval.

philips77, 2024-10-25:
Version 2.7.0 is now released.

PR #466 (2024-10-22): DFU speed improved on bonded devices
Android 15 extends connection interval on bonded devices to ~190 ms.
This is significantly longer that on previous versions, where it was kept set to 15 ms.

This PR requests high connection priority, which effectively means requesting short connection interval from the Android side. The upload speed increases in effect from 0.9 to 9.1 kb/s.

This PR fixes #459.