nordicsemi/Android-DFU-Library #481: Execution failed for task ':profile:welcome:kaptDebugKotlin'.

hu03ping, 2024-12-17:
### Version

2.7.0 or 2.8.0 (Latest)

### Ask the question

step:
1、pull lastest code
2、android studio 2024.2.1 Patch 3
3、when i build, the error come.
What should I do? 

### Relevant log output

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':profile:welcome:kaptDebugKotlin'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.lambda$executeIfValid$1(ExecuteActionsTaskExecuter.java:130)
	at org.gradle.internal.Try$Failure.ifSuccessfulOrElse(Try.java:293)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeIfValid(ExecuteActionsTaskExecuter.java:128)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:116)
	at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:46)
	at org.gradle.api.internal.tasks.execution.ResolveTaskExecutionModeExecuter.execute(ResolveTaskExecutionModeExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:57)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:74)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:77)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:55)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:209)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:204)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:166)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:52)
	at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:42)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:331)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:318)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.lambda$execute$0(DefaultTaskExecutionGraph.java:314)
	at org.gradle.internal.operations.CurrentBuildOperationRef.with(CurrentBuildOperationRef.java:85)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:314)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:303)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:459)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:376)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
	at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:48)
Caused by: org.gradle.workers.internal.DefaultWorkerExecutor$WorkExecutionException: A failure occurred while executing org.jetbrains.kotlin.gradle.internal.KaptWithoutKotlincTask$KaptExecutionWorkAction
	at org.gradle.workers.internal.DefaultWorkerExecutor$WorkItemExecution.waitForCompletion(DefaultWorkerExecutor.java:287)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.lambda$waitForItemsAndGatherFailures$2(DefaultAsyncWorkTracker.java:130)
	at org.gradle.internal.Factories$1.create(Factories.java:31)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLocks(DefaultWorkerLeaseService.java:339)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLocks(DefaultWorkerLeaseService.java:322)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withoutLock(DefaultWorkerLeaseService.java:327)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:126)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForItemsAndGatherFailures(DefaultAsyncWorkTracker.java:92)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForAll(DefaultAsyncWorkTracker.java:78)
	at org.gradle.internal.work.DefaultAsyncWorkTracker.waitForCompletion(DefaultAsyncWorkTracker.java:66)
	at org.gradle.api.internal.tasks.execution.TaskExecution$3.run(TaskExecution.java:252)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:166)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.api.internal.tasks.execution.TaskExecution.executeAction(TaskExecution.java:229)
	at org.gradle.api.internal.tasks.execution.TaskExecution.executeActions(TaskExecution.java:212)
	at org.gradle.api.internal.tasks.execution.TaskExecution.executeWithPreviousOutputFiles(TaskExecution.java:195)
	at org.gradle.api.internal.tasks.execution.TaskExecution.execute(TaskExecution.java:162)
[... 134 lines cut ...]
	at org.jetbrains.kotlin.kapt3.base.AnnotationProcessingKt.doAnnotationProcessing(annotationProcessing.kt:97)
	at org.jetbrains.kotlin.kapt3.base.AnnotationProcessingKt.doAnnotationProcessing(annotationProcessing.kt:36)
	at org.jetbrains.kotlin.kapt3.base.Kapt.kapt(Kapt.kt:46)
	... 31 more
Caused by: com.sun.tools.javac.processing.AnnotationProcessingError: java.lang.IllegalArgumentException: Invalid relative name: META-INF\proguard\no_nordicsemi_android_dfu_profile_welcome_viewmodelWelcomeViewModel_HiltModules_BindsModule_LazyClassKeys.pro
	at jdk.compiler/com.sun.tools.javac.processing.JavacProcessingEnvironment.callProcessor(Unknown Source)
	at jdk.compiler/com.sun.tools.javac.processing.JavacProcessingEnvironment$DiscoveredProcessors$ProcessorStateIterator.runContributingProcs(Unknown Source)
	at jdk.compiler/com.sun.tools.javac.processing.JavacProcessingEnvironment$Round.run(Unknown Source)
	at jdk.compiler/com.sun.tools.javac.processing.JavacProcessingEnvironment.doProcessing(Unknown Source)
	at jdk.compiler/com.sun.tools.javac.main.JavaCompiler.processAnnotations(Unknown Source)
	... 35 more
Caused by: java.lang.IllegalArgumentException: Invalid relative name: META-INF\proguard\no_nordicsemi_android_dfu_profile_welcome_viewmodelWelcomeViewModel_HiltModules_BindsModule_LazyClassKeys.pro
	at jdk.compiler/com.sun.tools.javac.file.JavacFileManager.getFileForOutput(Unknown Source)
	at org.jetbrains.kotlin.kapt3.base.javac.KaptJavaFileManager.getFileForOutput(KaptJavaFileManager.kt:82)
	at jdk.compiler/com.sun.tools.javac.processing.JavacFiler.createResource(Unknown Source)
	at org.jetbrains.kotlin.kapt3.base.incremental.IncrementalFiler.createResource(incrementalProcessors.kt:121)
	at dagger.spi.internal.shaded.androidx.room.compiler.processing.javac.JavacFiler.writeResource(JavacFiler.kt:94)
	at dagger.internal.codegen.processingstep.LazyClassKeyProcessingStep.writeProguardFile(LazyClassKeyProcessingStep.java:112)
	at dagger.internal.codegen.processingstep.LazyClassKeyProcessingStep.processOver(LazyClassKeyProcessingStep.java:106)
[... cut ...] t:54)
	at dagger.spi.internal.shaded.androidx.room.compiler.processing.CommonProcessorDelegate.processLastRound(XBasicAnnotationProcessor.kt:168)
	at dagger.spi.internal.shaded.androidx.room.compiler.processing.javac.JavacBasicAnnotationProcessor.process(JavacBasicAnnotationProcessor.kt:65)
	at org.jetbrains.kotlin.kapt3.base.incremental.IncrementalProcessor.process(incrementalProcessors.kt:90)
	at org.jetbrains.kotlin.kapt3.base.ProcessorWrapper.process(annotationProcessing.kt:216)
	... 40 more

philips77, 2024-12-17:
This looks like some issue with Android Studio. Try invalidating caches and syncing project again.

hu03ping, 2024-12-17:
> You may also change JDK to the one by JetBrains, if not set already: _Settings > Build, Execution, Deployment > Build Tools > Gradle_ - select JetBrains Runtime 21.0.3+.

yes, I use version 21.03

hu03ping, 2024-12-17:
> This looks like some issue with Android Studio. Try invalidating caches and syncing project again.

still have issue.
[image]

[image]

philips77, 2024-12-17:
I think it's related to: https://stackoverflow.com/a/79257730/2115352

I'll try updating Dagger and test locally.

philips77, 2024-12-17:
I released new version catalog: https://github.com/NordicSemiconductor/Android-Gradle-Plugins/releases/tag/2.5-5

In 10 min please update the version in _settings.gradle.kts_ file, sync and try again.

hu03ping, 2024-12-17:
> I released new version catalog: https://github.com/NordicSemiconductor/Android-Gradle-Plugins/releases/tag/2.5-5
> 
> In 10 min please update the version in _settings.gradle.kts_ file, sync and try again.

Hi philips77, after update to 2.5-5, also have issue.
[image]

philips77, 2024-12-17:
We managed to replicate your issue on Windows. For some reason the project builds without any issues on Mac.

philips77, 2024-12-17:
This may be related to too long file name. Could you try renaming `WelcomeViewModel` to `W` and rebuild?

philips77, 2024-12-17:
I think this is this one: https://github.com/google/dagger/issues/4533

philips77, 2024-12-18:
For now you may try checking out tag `2.7.0`. It was using Dagger 2.52 and perhaps this issue isn't there.

hu03ping, 2024-12-18:
> I think this is this one: [google/dagger#4533](https://github.com/google/dagger/issues/4533)

yes, @philips77 . It's working,  catalog v2.5-1.  thanks very much.

philips77, 2024-12-30:
I released version 2.9.0 with updated dependencies.