nordicsemi/Android-DFU-Library #482: [Core] Can I delete 'DfuBaseService'?

sieunju, 2024-12-17:
### Target

Library

### Describe the feature

## Why?
- The advantage of using the 'DfuBaseService' inherited service is that you can safely upload DFU.
However, if you upload several DFUs at the same time, you have to add several of the relevant 'DfuBase Service' and declare it in 'Android Manifest'. 

## Solution
- How about controlling DFU uploads to be processed within one class?

philips77, 2024-12-17:
Yes, true. This is a very old library, modern approach would be very different, with split for transport and logic, tests, etc.
`IntentService` besides that is deprecated and it's the user that should decide how and where to run it, e.g. in `WorkManager`.

Anyway, you may cut the relevant part from the `DfuBaseService` and implement it as a method.

We have a (very) long plan to finish [Version 2 of Kotlin BLE Library](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/tree/version/2.0) and rewrite DFU using it.

sieunju, 2024-12-18:
Aha, does the [Version 2 of Kotlin BLE Library](https://github.com/NordicSemiconductor/Kotlin-BLE-Library/tree/version/2.0) have a Device Firmware Update (DFU) feature?? 🙇‍♂️

@philips77

philips77, 2024-12-18:
No, it's lower layer. Handles GATT and GAP operations (connection, GATT, advertising) and allows peripheral emulation for test, kinds like [CoreBluetooth Mock](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock) for iOS, but better. And it may be KMM at some point also.

The new DFU would be built on top of it as version 3 of DFU library.

dinesharjani, 2026-08-20:
Marking as closed since an acceptable response was provided as to our future plans.

closed 2026-08-20 by dinesharjani