nordicsemi/Android-DFU-Library #484: Scanning for the DFU Bootloader... (timeout 5000 ms) Scanning for new address finished with: null Connection state change error: 255 newState: 0  How did it happen?

w-sso, 2024-12-20:
### Where do you suspect the issue?

Issue in DFU library, e.g. upload stops in the middle

### Version

2.7.0 or 2.8.0 (Latest)

### Describe the issue

It appeared while trying to connect

### Relevant log output

2024-12-20 08:46:59.418 23358-24048 DfuImpl                 com.kw.discalclaw                    I  Restarting to bootloader mode
2024-12-20 08:47:04.030 23358-23358 DfuBaseService          com.kw.discalclaw                    I  Action received: android.bluetooth.device.action.ACL_DISCONNECTED
2024-12-20 08:47:04.419 23358-24048 DfuImpl                 com.kw.discalclaw                    I  Scanning for the DFU Bootloader... (timeout 5000 ms)
2024-12-20 08:47:04.423 23358-24048 BluetoothAdapter        com.kw.discalclaw                    D  35070036: getState(). Returning ON
2024-12-20 08:47:04.425 23358-24048 BluetoothAdapter        com.kw.discalclaw                    D  isLeEnabled(): ON
2024-12-20 08:47:04.433 23358-23947 BluetoothLeScanner      com.kw.discalclaw                    D  onScannerRegistered() - status=0 scannerId=4 mScannerId=0
2024-12-20 08:47:07.833 10682-12934 SocketReaderNewtrue     pid-10682                            I  SocketReader(732) continue
2024-12-20 08:47:07.840 10682-12934 MSF.C.NetConnTag.true   pid-10682                            E  closeSocket for reason:readError with socket null, socketEngineId:1
2024-12-20 08:47:07.846 10682-12934 SocketReaderNewtrue     pid-10682                            E  socketEngine: 1socket hashCode: 0SocketReader(732) errorjava.net.SocketException: Socket is closed
2024-12-20 08:47:07.847 10682-12934 SocketReaderNewtrue     pid-10682                            I  SocketReader(732) wait
2024-12-20 08:47:09.425 23358-24048 BluetoothAdapter        com.kw.discalclaw                    D  isLeEnabled(): ON
2024-12-20 08:47:09.440 23358-24048 DfuImpl                 com.kw.discalclaw                    I  Scanning for new address finished with: null
2024-12-20 08:47:09.467 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Starting DFU service in foreground
2024-12-20 08:47:09.485 23358-24048 BluetoothAdapter        com.kw.discalclaw                    D  267216065: getState(). Returning ON
2024-12-20 08:47:09.485 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Connecting to the device...
2024-12-20 08:47:09.486 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  connect() - device: xx:xx:xx:xx:DB:4A, auto: false
2024-12-20 08:47:09.486 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  registerApp()
2024-12-20 08:47:09.486 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  registerApp() - UUID=0df6145c-e6be-4551-b685-8a67bee4321e
2024-12-20 08:47:09.489 23358-23947 BluetoothGatt           com.kw.discalclaw                    D  onClientRegistered() - status=0 clientIf=7
2024-12-20 08:47:22.199 12472-12796 mars::stn               pid-12472                            I  [longlink_connect_monitor.cc:536, OnHeartbeatAlarmReceived]:longlink_progress on_heartbeat_alarm, time: 20801989, is_noop_timeout:false
2024-12-20 08:47:22.260 12472-28636 mars::stn               pid-12472                            I  [longlink_connect_monitor.cc:530, OnHeartbeatAlarmSet]:longlink_progress on_heartbeat_set, time: 20802049, time interval:210000
2024-12-20 08:47:22.269 12472-28636 mars::stn               pid-12472                            I  [longlink_connect_monitor.cc:530, OnHeartbeatAlarmSet]:longlink_progress on_heartbeat_set, time: 20802058, time interval:210000
2024-12-20 08:47:22.537 12456-12578 com.tencent.mm          pid-12456                            W  Profile data size exceeds 500000 bytes. It has 507135 bytes.
2024-12-20 08:47:28.993  1539-3651  BatteryStatsService     pid-1539                             W  Unknown sensor handle 0 received in noteWakeupSensorEvent
2024-12-20 08:47:30.856 10682-24120 UnusedCodeCheck.Check   pid-10682                            I  init end result:-1002
2024-12-20 08:47:37.259 10682-11441 WorkerThreadMonitor     pid-10682                            D  Kill MSF check result[ senderSize:0,sendInterval:139960,receiveInterval=15800,addCmdCount=221
2024-12-20 08:47:37.854 10682-12934 SocketReaderNewtrue     pid-10682                            I  SocketReader(732) continue
2024-12-20 08:47:37.855 10682-12934 MSF.C.NetConnTag.true   pid-10682                            E  closeSocket for reason:readError with socket null, socketEngineId:1
2024-12-20 08:47:37.856 10682-12934 SocketReaderNewtrue     pid-10682                            E  socketEngine: 1socket hashCode: 0SocketReader(732) errorjava.net.SocketException: Socket is closed
2024-12-20 08:47:37.857 10682-12934 SocketReaderNewtrue     pid-10682                            I  SocketReader(732) wait
2024-12-20 08:47:37.885 10682-11457 MSF.C.DeepSleepDetector pid-10682                            D  [run] count after++: 81
2024-12-20 08:47:39.033  1539-2694  BatteryStatsService     pid-1539                             W  Unknown sensor handle 0 received in noteWakeupSensorEvent
2024-12-20 08:47:39.504 23358-23950 BluetoothGatt           com.kw.discalclaw                    D  onClientConnectionState() - status=255 clientIf=7 device=xx:xx:xx:xx:DB:4A
2024-12-20 08:47:39.505 23358-23950 DfuBaseService          com.kw.discalclaw                    E  Connection state change error: 255 newState: 0
2024-12-20 08:47:39.505 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Connection error after: 30021 ms
2024-12-20 08:47:39.506 23358-24048 DfuBaseService          com.kw.discalclaw                    E  An error occurred while connecting to the device: 255
2024-12-20 08:47:39.507 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Retrying... (attempt 2 / 3)
2024-12-20 08:47:39.508 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  refresh() - device: 
2024-12-20 08:47:39.516 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Refreshing result: true
2024-12-20 08:47:39.516 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Cleaning up...
2024-12-20 08:47:39.517 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  cancelOpen() - device: 
2024-12-20 08:47:39.521 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  close()
2024-12-20 08:47:39.522 23358-24048 BluetoothGatt           com.kw.discalclaw                    D  unregisterApp() - mClientIf=7
2024-12-20 08:47:39.528 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Restarting the service
2024-12-20 08:47:39.547 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Starting DFU service in foreground
2024-12-20 08:47:39.555 23358-24048 BluetoothAdapter        com.kw.discalclaw                    D  267216065: getState(). Returning ON
2024-12-20 08:47:39.555 23358-24048 DfuBaseService          com.kw.discalclaw                    I  Connecting to the device...

w-sso, 2024-12-23:
public void onScanResult(final int callbackType, final ScanResult result) {
        final String address = result.getDevice().getAddress();
        Log.i("TAG", "onScanResult: "+
                result.getDevice().getAddress()+" mDeviceAddress "+mDeviceAddress
        +"  mDeviceAddressIncremented " );
        if (!mFound && mSelector.matches(
                result.getDevice(), result.getRssi(),
                result.getScanRecord().getBytes(),
                mDeviceAddress, mDeviceAddressIncremented
        )) {
            mBootloaderAddress = address;
            mFound = true;

            // Notify the waiting thread
            synchronized (mLock) {
                mLock.notifyAll();
            }
        }
    }
Why does the address increment problem exist

philips77, 2024-12-30:
Please provide full log. The snippet above isn't enough.