nordicsemi/Android-DFU-Library #491: DFU 2.9.0 breaks LegacyDFU update

marcbaldwin, 2025-02-28:
### Where do you suspect the issue?

Issue in DFU library, e.g. upload stops in the middle

### Version

2.7.0 - 2.9.0 (Latest)

### Describe the issue

Hi there,

On version 2.9.0 we are unable to update devices using LegacyDFU, 2.8.0 works fine.

I did a git compare between 2.9.0 and 2.8.0 and what's strange is there doesn't seem to be any code changes other than doc updates?

Any help on this would be greatly appreciated

philips77, 2025-03-03:
Should we restore the documentation? ;)

marcbaldwin, 2025-03-05:
Thanks for the clarification on the java version.

It is very confusing as there really doesn't seem to be any significant changes between these versions; however `2.8.0` works every time and `2.9.0` fails every time for legacy DFU (secure DFU is fine).

I've attached the full logs from the two different versions. The 2.9.0 version fails when `Sending firmware to characteristic` with a status of 6. The logs prior to this point appear to be identical.

[DFU-2.9.0.txt](https://github.com/user-attachments/files/19090817/DFU-2.9.0.txt)
[DFU-2.8.0.txt](https://github.com/user-attachments/files/19090816/DFU-2.8.0.txt)

Any help would be greatly appreciated, thank you

DFU 2.9.0 Fails
```
14:15:44.021  D  DFU Log: [DFU] Receive Firmware Image request sent
14:15:44.027  D  DFU Log: [DFU] Uploading firmware...
14:15:44.028  D  DFU Log: [DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...
14:15:44.440  D  DFU Log: [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-03-06
14:15:44.440  D  DFU Log: [DFU] Response received (Op Code = 3, Status = 6)
14:15:44.441  D  DFU Log: [DFU] Remote DFU error: OPERATION FAILED
14:15:44.441  D  DFU Log: [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123, value (0x): 06
14:15:44.441  D  DFU Log: [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123, value=0x06, WRITE_TYPE_DEFAULT)
```

DFU 2.8.0 Success
```
14:12:29.750  D  DFU Log: [DFU] Receive Firmware Image request sent
14:12:29.755  D  DFU Log: [DFU] Uploading firmware...
14:12:29.755  D  DFU Log: [DFU] Sending firmware to characteristic 00001532-1212-efde-1523-785feabcd123...
14:13:34.087  D  DFU Log: [DFU] Notification received from 00001531-1212-efde-1523-785feabcd123, value (0x): 10-03-01
14:13:34.087  D  DFU Log: [DFU] Response received (Op Code = 3, Status = 1)
14:13:34.087  D  DFU Log: [DFU] Upload completed in 64314 ms
14:13:34.088  D  DFU Log: [DFU] Writing to characteristic 00001531-1212-efde-1523-785feabcd123, value (0x): 04
14:13:34.088  D  DFU Log: [DFU] gatt.writeCharacteristic(00001531-1212-efde-1523-785feabcd123, value=0x04, WRITE_TYPE_DEFAULT)
```

marcbaldwin, 2025-03-05:
Some more logs that could be useful during the DFU
```
15:36:04.412  E  packages/modules/Bluetooth/system/stack/btu/btu_hcif.cc:204 btu_hcif_log_event_metrics: Unexpectedly received event_code:0xff that should not be handled here
15:36:04.412  W  [WARNING:btif_bqr.cc(431)] Appro LSTO , Handle: 0x0014, NULL, Central, PwLv: 9, RSSI: -49, SNR: 0, UnusedCh: 59, UnidealCh: 1, ReTx: 0, NoRX: 0, NAK: 0, FlowOff: 0, OverFlow: 0, UndFlow: 0
15:36:04.412  E  packages/modules/Bluetooth/system/stack/btm/btm_sco.cc:950 btm_sco_on_disconnected: Unable to find sco connection
15:36:04.412  I  packages/modules/Bluetooth/system/stack/acl/btm_acl.cc:2569 btm_acl_iso_disconnected: ISO disconnection from GD, handle: 0x14, reason: 0x08
15:36:04.412  I  [INFO:gatt_attr.cc(343)] gatt_connect_cback: remove untrusted client status, bda=d7:7f:c6:6f:b6:4c
15:36:04.413  I  packages/modules/Bluetooth/system/stack/gap/gap_ble.cc:303 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:b6:4c callback:Disconnected
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:3 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:4 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:5 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:6 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:8 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:9 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:10 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:11 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  I  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_act.cc:1212 bta_gattc_conn_cback: Disconnected att_id:12 addr:xx:xx:xx:xx:b6:4c, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
15:36:04.413  W  packages/modules/Bluetooth/system/stack/btm/btm_sec.cc:3747 btm_sec_disconnected: Got uncommon disconnection reason:Connection Timeout handle:0x0014 comment:stack::acl::btm_acl::btm_acl_disconnected
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 771
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 772
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 773
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 774
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 776
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 777
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 778
15:36:04.413  E  packages/modules/Bluetooth/system/bta/gatt/bta_gattc_main.cc:395 bta_gattc_hdl_event: Ignore unknown conn ID: 779
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_av.cc:3409 btif_av_acl_disconnected: btif_av_acl_disconnected: Peer d7:7f:c6:6f:b6:4c : ACL Disconnected
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_av.cc:1086 FindOrCreatePeer: BtifAvPeer *BtifAvSource::FindOrCreatePeer(const RawAddress &, tBTA_AV_HNDL): Create peer: peer_address=d7:7f:c6:6f:b6:4c bta_handle=0x41 peer_id=0
15:36:04.413  D  onDisconnected() - removing permit for address=D7:7F:C6:6F:B6:4C
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_av.cc:3336 btif_av_get_peer_sep: No active sink or source peer found
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_a2dp_source.cc:708 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_av.cc:3336 btif_av_get_peer_sep: No active sink or source peer found
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_a2dp_source.cc:708 btif_a2dp_source_on_idle: btif_a2dp_source_on_idle: state=STATE_OFF
15:36:04.413  I  packages/modules/Bluetooth/system/btif/src/btif_av.cc:1131 DeleteIdlePeers: DeleteIdlePeers: Deleting idle peer: d7:7f:c6:6f:b6:4c bta_handle=0x41
```