nordicsemi/Android-DFU-Library #493: SecureDFUImpl fails when used with bin/hex and init file

cjstoltzfus, 2025-06-06:
### Where do you suspect the issue?

Issue in DFU library, e.g. upload stops in the middle

### Version

2.7.0 - 2.9.0 (Latest)

### Describe the issue

Hello,
I am attempting to initiate a DFU using a BIN application file and an associated DAT init file, but it is consistently failing with the stack trace in "relevant log output" and then eventually disconnects with a GATT timeout. My initiator code is as follows: 
```
// binFile and datFile point to files in the app's temp file directory
val starter = DfuServiceInitiator(dfuDevice.id).apply {
    setBinOrHex(DfuBaseService.TYPE_APPLICATION, binFile.toUri())
    setInitFile(datFile.toUri())
    setForeground(true)
    setDisableNotification(false)
}
starter.start(context, DFUService::class.java)
```
Looking at the code for `SecureDFUImpl`, it seems to assume that `mFirmwareStream` is always an `ArchiveInputStream`, when that does not seem like it will be the case based on `DfuBaseService.openInputStream()` as the MIME type there will not be zip. As a result the cast fails, the exception is thrown, and the DFU fails. Is there a way to work around this? I don't see a way to force the service to use a specific `BaseDFUImpl`.

Full log file attached from start of DFU service to failure.

[dfu-failure.log](https://github.com/user-attachments/files/20631358/dfu-failure.log)

### Relevant log output

java.lang.ClassCastException: java.io.FileInputStream cannot be cast to no.nordicsemi.android.dfu.internal.ArchiveInputStream
                                                                                                    	at no.nordicsemi.android.dfu.SecureDfuImpl$SecureBluetoothCallback.onCharacteristicChanged(SecureDfuImpl.java:113)
                                                                                                    	at no.nordicsemi.android.dfu.BaseDfuImpl$BaseBluetoothGattCallback.onCharacteristicChanged(BaseDfuImpl.java:240)
                                                                                                    	at no.nordicsemi.android.dfu.DfuBaseService$5.onCharacteristicChanged(DfuBaseService.java:1025)
                                                                                                    	at android.bluetooth.BluetoothGattCallback.onCharacteristicChanged(BluetoothGattCallback.java:159)
                                                                                                    	at android.bluetooth.BluetoothGatt$1$8.run(BluetoothGatt.java:525)
                                                                                                    	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:864)
                                                                                                    	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
                                                                                                    	at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:519)
                                                                                                    	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:246)
                                                                                                    	at android.os.Binder.execTransactInternal(Binder.java:1285)
                                                                                                    	at android.os.Binder.execTransact(Binder.java:1244)

philips77, 2025-06-07:
Hi, 
Thank you creating the issue. We will investigate next week.

philips77, 2025-06-07:
I recommend putting the files in a zip file together with a *manifest.json* file and use `setZip(..)` instead.

dinesharjani, 2026-08-20:
Hello,

Marking this issue as closed since Filip provided a solution that received a thumbs up. Please reopen if necessary.

closed 2026-08-20 by dinesharjani