nordicsemi/Android-DFU-Library #501: Your NRF Android app is faster updating the FW than my app

mobilekosmos, 2025-09-22:
### Version

None

### Ask the question

I noticed that  your NRF Android app is updating the FW in about 1min, while my app using the Jetpack Bluetooth Alpha Lib is taking aroung 1:50m, with the latest DFU lib.
I was using an older version of the DFU version where it was taking 2:55, after updating it went down to 1:50 but it's still a big difference to your app, and I'm not saying anything I can configure while establishing a connection to the device via the Google lib. Any ideas?

mobilekosmos, 2025-09-24:
Currently I'm using these:
"androidx.bluetooth:bluetooth:1.0.0-alpha02"
"no.nordicsemi.android:dfu:2.9.0"

dinesharjani, 2026-08-20:
@mobilekosmos this issue looks resolved. Please reply if you'd like to reopen it. I'm marking it as closed for now.

closed 2026-08-20 by dinesharjani